Koenkk/zigbee2mqtt #2930: Ikea bulb sometimes starts turning on and off after being powered on

nikolaykasyanov, 2020-02-10:
# Bug Report

I have the following bulb: TRADFRI LED bulb E12/E14 400 lumen, dimmable, white spectrum, opal white (LED1536G5).

I'm using it in a setup where it's turned on and off physically (TL;DR I'm can't yet go full smart in this room but I appreciate being able to tune light temperature at night with a HASS automation that starts when the lamp goes from unavailable to on state, hence a smart bulb).

## What happened

Sometimes (I'd say once-twice a month) when the lamp is turned on, it starts alternating between on and off states with ~0.5 sec interval, also light temperature goes warm. Turning it off and on again physically fixes it.

I had been thinking it was a firmware issue but recently I was been able to update it thanks to amazing work in #2921 and today it's happened again.

It's hard to say whether it's related to zigbee2mqtt or just a hardware quirk, although I haven't been able to find similar reports mentioning Trådfri bulbs online.

Here's what I see in the logs around the time it happened (log level `warn`):

```
zigbee2mqtt_1  | zigbee2mqtt:error 2020-02-10 14:46:25: Failed to read state of '0x90fd9ffffe8287ab' after reconnect
zigbee2mqtt_1  | zigbee2mqtt:error 2020-02-10 14:46:25: Publish 'set' 'brightness' to '0x90fd9ffffe8287ab' failed: 'Error: Command 0x90fd9ffffe8287ab/1 genLevelCtrl.moveToLevelWithOnOff({"level":255,"transtime":0}, {"timeout":10000,"defaultResponseTimeout":15000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Data request failed with error: 'MAC no ack' (233))'
zigbee2mqtt_1  | zigbee2mqtt:error 2020-02-10 14:46:25: Publish 'set' 'color_temp' to '0x90fd9ffffe8287ab' failed: 'Error: Command 0x90fd9ffffe8287ab/1 lightingColorCtrl.moveToColorTemp({"colortemp":250,"transtime":0}, {"timeout":10000,"defaultResponseTimeout":15000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Data request failed with error: 'MAC no ack' (233))'
```

`0x90fd9ffffe8287ab` is the bulb.

## What did you expect to happen

I expected the normal operation of the bulb after turning it off.

## How to reproduce it (minimal and precise)

Use the bulb as a regular bulb, wait couple of weeks.

## Debug Info
zigbee2mqtt version: 1.10.0-dev (984bad4)
CC253X firmware version: 20190619

way2many, 2020-02-11:
I experience the same behaviour with a GU10 of Ikea. I bought 4 on the same day and on same pallet.
Out of the 4, 1 was very difficult to pair, is often seen as unavailable (due to no ping) and has the same behaviour as you describe. Never checked the logs as I use it a group of lights so even if it is unavailable it responds to the commands send to the group.

It is not the first time I had some "bad" lamps with Ikea.

nikolaykasyanov, 2020-02-11:
@way2many that makes sense, it never occurred to me that it might just be a defective unit 😅

closed 2020-04-18 by stale[bot]