Koenkk/zigbee2mqtt #2931: Please include:  IKEA LED1903C5, TRADFRI bulb E14 WS 470lm

SargonofAssyria, 2020-02-11:
Please include the new IKEA light bulb:  LED1903C5 - IKEA TRADFRI LED bulb E14 470 lumen, dimmable, white spectrum, opal white (Router)
It presents it self with: TRADFRI bulb E14 WS 470lm

sjorge, 2020-02-12:
```
    {
        zigbeeModel: ['TRADFRI bulb E14 WS 470lm'],
        model: 'LED1903C5',
        vendor: 'IKEA',
        description: 'TRADFRI bulb E14 WS 470 lumen, dimmable, white spectrum, opal white',
        extend: generic.light_onoff_brightness_colortemp,
        ota: ota.tradfri,
    },
```

I think the following entry should work, not tested as I do not have the bulb.

sjorge, 2020-02-12:
You probably need to remove the `ota: ota.trodfri,` line if you are not on 1.10+dev

SargonofAssyria, 2020-02-12:
Yes, that did work. This has now only to be included in the next beta.
And now I have to wait for the zigbee2mqtt-plugin, because it does not show up in Domoticz yet.

closed 2020-02-12 by SargonofAssyria

sjorge, 2020-02-12:
I'll do a PR against converter repos to get this included.

blakadder, 2020-02-13:
I'm only finding LEDARE bulbs under that model number... Is there a link to a Tradfri version?

sjorge, 2020-02-13:
@blakadder I'm not sure, I based it off the database entry posted by @SargonofAssyria

SargonofAssyria, 2020-02-19:
Its a new bulb, link in the first post. And it is working.

blakadder, 2020-02-19:
do you have an SKU for the bulb  or some other item code?

SargonofAssyria, 2020-02-19:
Yes I have, but currently I am on ski holliday, so after next saterday.

SargonofAssyria, 2020-02-23:
On the box it states: Tradfri LED 470 lm 5.2W E14 LED1835C6
And under the barcode 704.243.12 19972