Koenkk/zigbee2mqtt #2932: MQTTS: no such file or directory, open '/app/data/mqtt-ca.crt'

ochstobi, 2020-02-11:
I started the zigbee2mqtt docker container with this command: **sudo docker run -itd --name="Zigbee2MQTT" --net=host -e TZ=Europe/Berlin --device=/dev/ttyACM0 --restart=always -v /volume1/docker/zigbee2mqtt/data:/app/data koenkk/zigbee2mqtt:latest**.
It worked fine when the mqtt broker was unsecured. But now I have updated the broker to use SSL.
The new configuration for MQTT is the following: 

```
mqtt:
  # MQTT base topic for zigbee2mqtt MQTT messages
  base_topic: zigbee2mqtt
  # MQTT server URL
  server: 'mqtt://myserver:8883'
  ca: '/app/data/mqtt-ca.crt'
```

Since I added the ca file to the configuration.yaml I get this error: **ENOENT: no such file or directory, open '/app/data/mqtt-ca.crt'**

The CA file is located in /volume1/docker/zigbee2mqtt/data/mqtt-ca.crt. Since this path is mapped to /app/data I thought the ca file must be accessible for zigb2mqtt. What am I missing?

Thanks in advance
Tobias

closed 2020-02-12 by ochstobi

clemenlg, 2025-05-29:
Hello,
I have same issue as you, can you explain please how you solve it?
I saw you closed the issue, but not the reason. I'm stuck in the same point, and I will be very, very appreciate if you respond.
Regards

ochstobi, 2025-05-29:
Hello,
I didn't resove the issue. I just go unsecured in my home network and have a reverse proxy that handles connection to the internet.
Many small applications dont even offer handling of certificates, so I dropped the idea for my home network

clemenlg, 2025-05-29:
Hello @ochstobi ,
Sorry, I just delete my comment before see your comment.
My mistake was the path, because I really think that the Proxmox LXC container (https://community-scripts.github.io/ProxmoxVE/scripts?id=zigbee2mqtt) uses docker, although I can't find the binary (neither the PDI). Finally I check the installation script (https://raw.githubusercontent.com/community-scripts/ProxmoxVE/main/install/zigbee2mqtt-install.sh) and confirmed that there aren't dropbox files, it runs with pnpm, then the files path aren't inside _/data_, the real path is _/opt/zigbee2mqtt/data/_. If I change this path, this _ENOENT_ error (_no such file or directory, open '/path-to-data/ca.crt'_) disappears, and only appears a new one:
```
error: z2m: MQTT failed to connect, exiting... (error:1C800064:Provider routines::bad decrypt)
info: z2m: Stopping zigbee-herdsman...
```
I'm stuck now in this point, but I think this error may be because the ca.crt file is encrypted with a passphrase (-sha256). I must do some tests generating a new ca key without this passphrase.
I'm now at the work, when I arrive to house, I will check it.
Regards

clemenlg, 2025-05-29:
Hello @ochstobi ,
It works well with the certificates, but can't be encrypted with a passphrase.
Steps:
Create a CA:
```
openssl genrsa -out rootCA.key 2048
openssl req -new -x509 -days 3650 -key rootCA.key -out rootCA.crt -config <( cat losan.csr.cnf )
```
Create the server certificates:
```
openssl genrsa -out mqtt/mqtt_server.key 2048
openssl req -new -key mqtt/mqtt_server.key -out mqtt/mqtt_server.csr -config <( cat mqtt/mqtt_server.csr.cnf )
openssl x509 -req -in mqtt/mqtt_server.csr -extfile mqtt/mqtt_server_v3.ext -CA rootCA.crt -CAkey rootCA.key -CAcreateserial -out mqtt/mqtt_server.crt -days 3650
```
Example for .csr.cnf file:
```
# mqtt_server.csr.cnf
[req]
default_bits = 2048
prompt = no
default_md = sha256
distinguished_name = dn

[dn]
C=ES
ST=Valencia
L=Country
O=SURNAME HOME
OU=Home
emailAddress=mymail@gmail.com
CN=mqtt
```
Example for .v3.ext file:
```
authorityKeyIdentifier=keyid,issuer
basicConstraints=CA:FALSE
keyUsage = digitalSignature, nonRepudiation, keyEncipherment, dataEncipherment
subjectAltName = @alt_names
[alt_names]
DNS.1 = mqtt
IP = WRITE.YOUR.IP.ADDRESS
```
Create the client certificates:
```
openssl genrsa -out mqtt/mqtt_client_zigbee2mqtt.key 2048
openssl req -new -key mqtt/mqtt_client_zigbee2mqtt.key -out mqtt/mqtt_client_zigbee2mqtt.csr -config <( cat mqtt/mqtt_client_zigbee2mqtt.csr.cnf )
openssl x509 -req -in mqtt/mqtt_client_zigbee2mqtt.csr -CA rootCA.crt -CAkey rootCA.key -CAcreateserial -out mqtt/mqtt_client_zigbee2mqtt.crt -days 3650
```
Adjust permissions:
```
chown -R mosquitto:mosquitto /etc/mosquitto/certs /etc/mosquitto/ca_certificates
chmod 0600 /etc/mosquitto/ca_certificates/* /etc/mosquitto/certs/*
chmod 0755 /etc/mosquitto/ca_certificates /etc/mosquitto/certs
```
Edit default.conf file (mosquitto), in my system is : /etc/mosquitto/conf.d/default.conf:
```
allow_anonymous false
persistence true
password_file /etc/mosquitto/passwd
#per_listener_settings true
listener 1883 localhost

listener 8883 0.0.0.0
cafile /etc/mosquitto/ca_certificates/rootCA.crt
certfile /etc/mosquitto/certs/mqtt_server.crt
keyfile /etc/mosquitto/certs/mqtt_server.key
require_certificate true

listener 8884 0.0.0.0
protocol websockets
cafile /etc/mosquitto/ca_certificates/rootCA.crt
certfile /etc/mosquitto/certs/mqtt_server.crt
keyfile /etc/mosquitto/certs/mqtt_server.key
require_certificate true
```
Restart service:
```
systemctl restart mosquitto
systemctl status mosquitto
```
Check it:
```
mosquitto_pub -h localhost -p 1883 -t test -u "mosquitto" -P "M0squ1tt0" -m "hello tls" -d
mosquitto_pub -h mqtt -p 8883 --cafile ca_certificates/rootCA.crt --cert certs/mqtt_client_zigbee2mqtt.crt --key certs/mqtt_client_zigbee2mqtt.key -t test -u "mosquitto" -P "M0squ1tt0" -m "hello tls" -d
```
Edit zigbee2mqtt file, in my system /opt/zigbee2mqtt/data/configuration.yaml:
```
homeassistant:
  enabled: true
frontend:
  enabled: true
  port: 9442
mqtt:
  base_topic: zigbee2mqtt
  server: mqtts://mqtt:8883
  ca: /opt/zigbee2mqtt/data/ssl/rootCA.crt
  key: /opt/zigbee2mqtt/data/ssl/mqtt_client_zigbee2mqtt.key
  cert: /opt/zigbee2mqtt/data/ssl/mqtt_client_zigbee2mqtt.crt
  user: mosquittoUser
  password: mosquitoPassword
  client_id: zigbee2mqtt
  keepalive: 60
  reject_unauthorized: true
  version: 4
advanced:
  network_key: WHATEVER
  pan_id: 25675
  ext_pan_id: WHATEVER
  transmit_power: 20
serial:
  port: tcp://slzbmr1:6638 (in my case)
  baudrate: 115200 (in my case)
  adapter: zstack (in my case)
  disable_led: false (in my case)
version: 4
devices:
```
Restart service:
```
systemctl stop zigbee2mqtt
systemctl start zigbee2mqtt
systemctl status zigbee2mqtt
```
Check logs:
`cat /opt/zigbee2mqtt/data/log/path`