Koenkk/zigbee2mqtt #2933: Aqara Opple Switch (WXCJKG13LM) problem

jacekpaszkowski, 2020-02-11:
**zigbee2mqtt add-on version (if edge, please report commit hash):**
version 1.9.0-dev (commit #cf1bb386ead98a78b6b3bfed01e8d151cc7e9c6c) and version 1.9.0 stable

**Operating environment (HassOS, Virtual Machine, Device/platform running Hass.io):**
Raspberry Pi3 Hass.io

**Description of problem:**
I'm not able to get fully working my Aqara Opple Switch (WXCJKG13LM). For some button there is only single click, some also has hold/relese, none has double click.

- button 1: only single click (doble click reported as single)
- button 2: only single click (doble click reported as single)
- button 3-4: only single click (doble click reported as single), hold/release works ok

Second question is that each restart addon seems to loose all the data. I have to pair and unpair devices after each addon restart.
EDIT: i've manged to get it working

**Your entire configuration from the frontend (with sensitive fields redacted)**:
```json
{
  "data_path": "/share/zigbee2mqtt",
  "devices": "devices.yaml",
  "groups": "groups.yaml",
  "homeassistant": false,
  "permit_join": true,
  "mqtt": {
    "base_topic": "zigbee2mqtt",
    "server": "",
    "user": "",
    "password": ""
  },
  "serial": {
    "port": "/dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B0018DE81C2-if00"
  },
  "advanced": {
    "pan_id": 6754,
    "channel": 11,
    "network_key": [
      1,
      3,
      5,
      7,
      9,
      11,
      13,
      15,
      0,
      2,
      4,
      6,
      8,
      10,
      12,
      13
    ],
    "availability_blacklist": []
  },
  "ban": [],
  "whitelist": [],
  "queue": {},
  "socat": {
    "enabled": false,
    "master": "pty,raw,echo=0,link=/dev/ttyZ2M,mode=777",
    "slave": "tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5",
    "restartdelay": 1.0,
    "initialdelay": 1.0,
    "options": "-d -d",
    "log": false
  }
```

**Your logs from Hass.io**
```
zigbee2mqtt:info  2020-01-27 22:59:09: Device '0x04cf8cdf3c75b85a' is supported, identified as: Xiaomi Aqara Opple switch 3 bands (WXCJKG13LM)
zigbee2mqtt:info  2020-01-27 22:59:09: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0x04cf8cdf3c75b85a","model":"WXCJKG13LM","vendor":"Xiaomi","description":"Aqara Opple switch 3 bands","supported":true}}'
zigbee2mqtt:info  2020-01-27 22:59:09: Configuring '0x04cf8cdf3c75b85a'
zigbee2mqtt:info  2020-01-27 22:59:14: Successfully configured '0x04cf8cdf3c75b85a'
zigbee2mqtt:info  2020-01-27 22:59:21: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c75b85a', payload '{"linkquality":23,"battery":"100.00","voltage":3000}'
zigbee2mqtt:info  2020-01-27 22:59:21: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c75b85a', payload '{"linkquality":21,"battery":"100.00","voltage":3000,"action":"button_1_single"}'
zigbee2mqtt:info  2020-01-27 22:59:28: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c75b85a', payload '{"linkquality":65,"battery":"100.00","voltage":3000,"action":"button_1_single"}'
zigbee2mqtt:info  2020-01-27 23:00:23: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c75b85a', payload '{"linkquality":63,"battery":"100.00","voltage":3000}'
zigbee2mqtt:info  2020-01-27 23:03:19: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c75b85a', payload '{"linkquality":44,"battery":"100.00","voltage":3000,"action":"button_1_single"}'
zigbee2mqtt:info  2020-01-27 23:03:21: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c75b85a', payload '{"linkquality":39,"battery":"100.00","voltage":3000,"action":"button_2_single"}'
zigbee2mqtt:info  2020-01-27 23:03:22: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c75b85a', payload '{"linkquality":47,"battery":"100.00","voltage":3000,"action":"button_2_single"}'
zigbee2mqtt:info  2020-01-27 23:03:23: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c75b85a', payload '{"linkquality":39,"battery":"100.00","voltage":3000,"action":"button_3_single"}'
zigbee2mqtt:info  2020-01-27 23:03:24: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c75b85a', payload '{"linkquality":36,"battery":"100.00","voltage":3000,"action":"button_3_hold"}'
zigbee2mqtt:info  2020-01-27 23:03:26: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c75b85a', payload '{"linkquality":39,"battery":"100.00","voltage":3000,"action":"button_3_release","duration":1779}'
zigbee2mqtt:info  2020-01-27 23:03:27: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c75b85a', payload '{"linkquality":0,"battery":"100.00","voltage":3000,"action":"button_4_single"}'
zigbee2mqtt:info  2020-01-27 23:03:30: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c75b85a', payload '{"linkquality":0,"battery":"100.00","voltage":3000,"action":"button_4_hold"}'
zigbee2mqtt:info  2020-01-27 23:03:32: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c75b85a', payload '{"linkquality":0,"battery":"100.00","voltage":3000,"action":"button_4_release","duration":2257}'
zigbee2mqtt:info  2020-01-27 23:03:35: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c75b85a', payload '{"linkquality":15,"battery":"100.00","voltage":3000,"action":"button_5_single"}'
zigbee2mqtt:info  2020-01-27 23:03:36: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c75b85a', payload '{"linkquality":15,"battery":"100.00","voltage":3000,"action":"button_5_single"}'
zigbee2mqtt:info  2020-01-27 23:03:39: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c75b85a', payload '{"linkquality":34,"battery":"100.00","voltage":3000,"action":"button_5_hold"}'
zigbee2mqtt:info  2020-01-27 23:03:43: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c75b85a', payload '{"linkquality":15,"battery":"100.00","voltage":3000,"action":"button_5_release","duration":3397}'
zigbee2mqtt:info  2020-01-27 23:03:45: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c75b85a', payload '{"linkquality":31,"battery":"100.00","voltage":3000,"action":"button_6_single"}'
zigbee2mqtt:info  2020-01-27 23:03:47: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c75b85a', payload '{"linkquality":34,"battery":"100.00","voltage":3000,"action":"button_6_single"}'
zigbee2mqtt:info  2020-01-27 23:03:53: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c75b85a', payload '{"linkquality":68,"battery":"100.00","voltage":3000,"action":"button_5_release","duration":13855}'
```

Please see my original post at danielwelch/hassio-zigbee2mqtt:
https://github.com/danielwelch/hassio-zigbee2mqtt/issues/290

Koenkk, 2020-02-11:
Should work with zigbee2mqtt 1.10

jacekpaszkowski, 2020-02-11:
I've upgraded to 1.10. But nothing hanged, no double clicks and no hold/release for button 1 & 2. :(

kirovilya, 2020-02-11:
you must re-configure or unpair and pair the switch again

jacekpaszkowski, 2020-02-11:
@kirovilya you are right! great, issue solved!
THANK YOU! :)

closed 2020-02-11 by jacekpaszkowski

drschnalli, 2020-02-19:
Hello i also got my issue solved with the help of this issue and its answers. Anyways the long press function now shows the same hold value every time. Does not matter when you hold the button 1-2-3 seconds its showing the same value. I also repaired the device already after i updatet to 1.10 

Got the Z-Stack_Home_1.2 CC2531_DEFAULT_20190608 Firmware installed (coordinator)