Koenkk/zigbee2mqtt #2937: Transition of 0 gets ignored when turning off a light group

nmaggioni, 2020-02-12:
# Bug Report
## What happened

The `transition` value for turning off a light is ignored if it's `0` or `0.0`. Very small values such as `0.001` get recognized and can be practically used in place of `0`.

## What did you expect to happen

When turning off a light with a `transition` value of `0`, I expected it to shut down immediately. Instead it seems to be following the default value set in the `groups.yaml` file.

## How to reproduce it (minimal and precise)

I'm toggling a group of two [Tradfri bulbs](https://www.zigbee2mqtt.io/devices/LED1537R6.html).

### Logs
https://pastebin.com/kybzf0TM

### `devices.yaml`
```yml
#...
'0x000b57fffe3054bb':
  friendly_name: tradfri_bulb_1
  retain: false
'0x000b57fffe0e3c73':
  friendly_name: tradfri_bulb_2
  retain: false
#...
```

### `groups.yaml`
```yml
'1':
  friendly_name: tradfri_bulbs_1_2
  devices:
    - '0x000b57fffe0e3c73'
    - '0x000b57fffe3054bb'
  transition: 1
```

## Debug Info
zigbee2mqtt version: 1.10.0
CC253X firmware version: 20190608

Koenkk, 2020-02-13:
Fixed, will be available in latest dev in a few hours, thanks!

closed 2020-02-13 by Koenkk

nmaggioni, 2020-02-13:
Thanks for the super quick patch 👍