Koenkk/zigbee2mqtt #2939: Sonoff BasicZBR3 - state not updated

hossamnagy, 2020-02-12:
# Bug Report
## What happened
When turning on BASICZBR3 switch through Lovelace card, The switch turn-on momentarily and then return to the OFF state. The device it self is turned ON 

## What did you expect to happen
The Switch should remain on the ON state until turned OFF . 

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version: 1.10.0-dev
CC253X firmware version:{'type': 'zStack12', 'meta': {'transportrev': 2, 'product': 0, 'majorrel': 2, 'minorrel': 6, 'maintrel': 3, 'revision': 20190608}}

Koenkk, 2020-02-13:
> From log files it appears that zigbee2mqtt is unable to parse the state reported by the device .

Please post the logs of these errors.

hossamnagy, 2020-02-15:
Closing this issue as the problem is now resolved ! not sure what was the error .  the main change I did was to upgrade to the latest home assistant version this morning.

closed 2020-02-15 by hossamnagy