Koenkk/zigbee2mqtt #2941: Bug report: Failed: 'Error: Command device_id/1 genOnOff.off', 'Error: Command device_id/1 genOnOff.on'

dzungpv, 2020-02-13:
## Bug Report
Switch can update it state but could not control (send ON/OFF command).
I have this issue very long time, both with Z-stack 1.2, Ztack 3.0.x and 3.x.0.
The device firmware base on Z-stack 1.2 SampleLight/SampleSwitch of TI.
There are 19 devices (End devices) in my network, it is random, it can work fine for about 1 week and then suddenly occur.  I also ask on TI forum, they say it may relate to Child Aging so 1 try update Coordinator to 3.x.x, but problem still here.
My Coordinator firmware base on https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.x.0 on CC1352p-2, except with Child Aging enable.
There is a router running on CC2652R1, Z-Stack_3.x.0 base on SampleLight, with Child Aging enable too, it also occurs with all known router firmware.

This may relate to this bug when running 1.2 base firmware: https://github.com/Koenkk/zigbee2mqtt/issues/1090

## Debug Info
zigbee2mqtt version: 1.10-dev
CC2652/CC1352 firmware version: Z-Stack_3.x.0 with Child Aging enable.
Below is the logs:

It work when update on/off state from device

```
2020-02-13T04:28:42.922Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-02-13T04:28:46.829Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,27,68,129,0,0,6,0]
2020-02-13T04:28:46.830Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,27,68,129,0,0,6,0]
2020-02-13T04:28:46.831Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [200,118,1,1,0,105,0,68,121,198,0,0,7,24,22,10,0,0,16,1,191,81,27,19]
2020-02-13T04:28:46.832Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,27,68,129,0,0,6,0,200,118,1,1,0,105,0,68,121,198,0,0,7,24,22,10,0,0,16,1,191,81,27,19]
2020-02-13T04:28:46.832Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 27 - 2 - 4 - 129 - [0,0,6,0,200,118,1,1,0,105,0,68,121,198,0,0,7,24,22,10,0,0,16,1,191,81,27] - 19
2020-02-13T04:28:46.835Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":30408,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":105,"securityuse":0,"timestamp":13007172,"transseqnumber":0,"len":7,"data":{"type":"Buffer","data":[24,22,10,0,0,16,1]}}
2020-02-13T04:28:46.838Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true},"transactionSequenceNumber":22,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":0,"dataType":16,"attrData":1}]},"address":30408,"endpoint":1,"linkquality":105,"groupID":0}'
zigbee2mqtt:debug 2020-02-13 11:28:46: Received Zigbee message from '0x00124b0012111fff', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-02-13 11:28:46: MQTT publish: topic 'zigbee2mqtt/0x00124b0012111fff', payload '{"state":"ON","linkquality":105}'
2020-02-13T04:28:46.846Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
```

When it not working with ON/OFF command:
```
2020-02-13T04:27:08.469Z zigbee-herdsman:controller:endpoint Command 0x00124b0012111fff/1 genOnOff.on({}, {"timeout":10000,"defaultResponseTimeout":15000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Timeout - 30408 - 1 - 11 - 6 - 11 after 15000ms)
zigbee2mqtt:error 2020-02-13 11:27:08: Publish 'set' 'state' to '0x00124b0012111fff' failed: 'Error: Command 0x00124b0012111fff/1 genOnOff.on({}, {"timeout":10000,"defaultResponseTimeout":15000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Timeout - 30408 - 1 - 11 - 6 - 11 after 15000ms)'
zigbee2mqtt:debug 2020-02-13 11:27:08: Error: Command 0x00124b0012111fff/1 genOnOff.on({}, {"timeout":10000,"defaultResponseTimeout":15000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Timeout - 30408 - 1 - 11 - 6 - 11 after 15000ms)
    at Endpoint.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:347:23)
    at Generator.throw (<anonymous>)
    at rejected (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65)
zigbee2mqtt:info  2020-02-13 11:27:08: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to '0x00124b0012111fff' failed: 'Error: Command 0x00124b0012111fff/1 genOnOff.on({}, {\"timeout\":10000,\"defaultResponseTimeout\":15000,\"manufacturerCode\":null,\"disableDefaultResponse\":false}) failed (Error: Timeout - 30408 - 1 - 11 - 6 - 11 after 15000ms)'","meta":{"friendly_name":"0x00124b0012111fff"}}'
```

```
zigbee2mqtt:error 2020-02-13 10:55:08: Publish 'set' 'state' to '0x00124b0012111fff' failed: 'Error: Command 0x00124b0012111fff/1 genOnOff.off({}, {"timeout":10000,"defaultResponseTimeout":15000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Timeout - 30408 - 1 - 76 - 6 - 11 after 15000ms)'
zigbee2mqtt:debug 2020-02-13 10:55:08: Error: Command 0x00124b0012111fff/1 genOnOff.off({}, {"timeout":10000,"defaultResponseTimeout":15000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Timeout - 30408 - 1 - 76 - 6 - 11 after 15000ms)
    at Endpoint.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:347:23)
    at Generator.throw (<anonymous>)
    at rejected (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65)
zigbee2mqtt:info  2020-02-13 10:55:08: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to '0x00124b0012111fff' failed: 'Error: Command 0x00124b0012111fff/1 genOnOff.off({}, {\"timeout\":10000,\"defaultResponseTimeout\":15000,\"manufacturerCode\":null,\"disableDefaultResponse\":false}) failed (Error: Timeout - 30408 - 1 - 76 - 6 - 11 after 15000ms)'","meta":{"friendly_name":"0x00124b0012111fff"}}'
```

dzungpv, 2020-02-13:
Updated: restart zigbee2mqtt could not solved the problem, but after soft reset the device can be control again.

verjus, 2020-02-19:
I have the almost identical error with CC2538-CC2592 running 3.0.x on the latest modkaru_v3 firmware with child aging disabled:

```
+/**
+ * Disable child aging, otherwise Xiaomi devices are being kicked off the netwo
+ */
+uint8 zgChildAgingEnable = FALSE;
```

` {"version":"1.10.0","commit":"8df6be8","coordinator":{"type":"zStack30x","meta":{"transportrev":2,"product":2,"majorrel":2,"minorrel":7,"maintrel":2,"revision":20200128}},"log_level":"error","permit_join":true
`

`zigbee2mqtt/bridge/log {"type":"zigbee_publish_error","message":"Publish 'set' 'brightness' to 'Hue_4' failed: 'Error: Command 0x00xxxxxxxxxxxxxxxa/11 genLevelCtrl.moveToLevelWithOnOff({\"level\":250,\"transtime\":10}, {\"timeout\":10000,\"defaultResponseTimeout\":15000,\"manufacturerCode\":null,\"disableDefaultResponse\":false}) failed (Error: Timeout - 51912 - 11 - 235 - 8 - 11 after 15000ms)'","meta":{"friendly_name":"Hue_4"}}`

Not sure where the root cause lies, firmware or z2m?

dzungpv, 2020-02-19:
@verjus i have this error with both zgChildAgingEnable  enable and disable, may be it relate to this error https://github.com/Koenkk/zigbee2mqtt/issues/2958#issuecomment-587355388

timdonovanuk, 2020-02-19:
How are you enabling child aging? I cannot find any references to child aging being supported or enabled in the z2m firmware, only an open ticket to support it here https://github.com/Koenkk/zigbee2mqtt/issues/2313. 

I got this error last night but afaik do not have child aging enabled, as I'm running the default firmware for CC1352P-2 (afaik there is only one firmware available for this presently - CC1352P_2_20191106).

dzungpv, 2020-02-19:
> How are you enabling child aging? I cannot find any references to child aging being supported or enabled in the z2m firmware, only an open ticket to support it here #2313.
>
> I got this error last [...]

You just change false to true and rebuild the firmware, the is guide in the firmware project

timdonovanuk, 2020-02-19:
Ah right, not seen that. Looking at the code its not relevant for me, as I have Xiaomi devices in my network and it says "Disable child agaging, otherwise Xiaomi devices are being kicked off the network.". Anyway, all irrelevant to this bug, as its seems the issue surfaces regardless of child aging setting! :)

closed 2020-04-26 by stale[bot]