Koenkk/zigbee2mqtt #2942: Lamps and Spots Ikea and power cut

Ricardo544, 2020-02-13:
Hello,

I have a lot of lamps and Ikea spots controlled with zigbee2mqtt and it works wonderfully. Would it be possible to see if in the clusters linked to the bulb or to the spot there would not be the possibility of making sure to keep the state of the lamp before power cut. In fact, after each power outage, the Ikea lamps reignite full after the power returns, even if they were turned off before.

It seems that with Deconz I quote: In deconz the attribute ‘PowerOn OnOff’ of the OnOff cluster is not supported by the philips Hue (nor the aqara bulb), while it is for the ikea bulb; this is what allows you to configure the behavior at power-up from deconz.

The same goes for the ‘Power On Level’ attribute of the ‘Level Control’ cluster, which allows you to adjust the light intensity at power up. With ikea, for the ‘Power On Level’: if we set this attribute to the value 255, the bulb retains the state of before cut.

Thank you

tricoos, 2020-02-14:
I think they have to turn on, because a power cut also happens if you turn them off and on again to reset them. Otherwise they would never light up at all if you turn off the switch and then on again. 

The current setting is helpful if there is an issue with the lamp or Zigbee2mqtt. Just imagine that the lamp would not turn on because of an issue and you still cannot turn it on even by power-cycling.

Ricardo544, 2020-02-14:
Theoretically, by resetting the bulb, the parameter to keep the state before power failure is reset to false by default. Finally it is like that with the connected Xiaomi bulbs and it is normal that is what is used for a reset, we reset the default settings.
Good weekend to all

Koenkk, 2020-02-17:
I cannot find which attribute they mean, it at least doesn't seem to be defined in the ZCL:

[image]

Ricardo544, 2020-02-18:
Yes I see that on the documentation that you copied. I think if this option was available you would have implemented the option.
To see if this option will be available in a possible firmware update

closed 2020-04-25 by stale[bot]