Koenkk/zigbee2mqtt #2944: New device Paulmann 500.48

nagi121, 2020-02-14:
# Not a Bug, it's a feature request
## What happened
The Paulmann Controller 500.48 is not supported

## What did you expect to happen

## How to reproduce it (minimal and precise)
Add a Paulmann Controller 500.48.

## Debug Info
zigbee2mqtt version: 1.7.1
CC253X firmware version: {"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}

This is the entry of database.db:
{"id":9,"type":"Router","ieeeAddr":"0x00158d00044d8506","nwkAddr":26660,"manufId":4151,"manufName":"Paulmann Licht","powerSource":"DC Source","modelId":"500.48","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,4,3,6,8,5,4096,25],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"500.48","manufacturerName":"Paulmann Licht","powerSource":4,"zclVersion":3,"appVersion":12,"stackVersion":1,"hwVersion":1,"dateCode":"20190515","swBuildId":"1400-0001"}}},"binds":[]},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[]}},"appVersion":12,"stackVersion":1,"hwVersion":1,"dateCode":"20190515","swBuildId":"1400-0001","zclVersion":3,"interviewCompleted":true,"meta":{}}

This is the web page from manufacturer describing the device:
https://de.paulmann.com/innenleuchten/smart-home/zigbee/smarthome-zigbee-yourled-dimm-schalt-controller-max.-60w/50048

I am using Hassio on Pi4 with Raspbian and I am a newbie to linux and docker. I kindly request to add this device to the supported devices.

Best regards,
Andreas

nagi121, 2020-02-14:
This are the log entries regarding this device:

[paulmann_50048.log](https://github.com/Koenkk/zigbee2mqtt/files/4204979/paulmann_50048.log)

Koenkk, 2020-02-14:
Added, will be supported in the dev branch in a few hours.

closed 2020-02-14 by Koenkk