Koenkk/zigbee2mqtt #2946: Heiman HS3SA, cannot pair

federic0, 2020-02-14:
Hi,

I cannot pair a Heiman HS3SA smoke sensor. There are two buttons (test and a button new the battery), but pressing either one does not produce any thing in the logs, Alabel in the box says HS3SA*1.

Any help is appreciated.

xaziwb, 2020-02-14:
It's not a button - it is a small hole. If the test button is in front of you the hole is on the right. Push it wirh an thin pin for about 2 secs.
To finish pairing it may be necessary to push it (very short) again.

federic0, 2020-02-15:
> It's not a button - it is a small hole. If the test button is in front of you the hole is on the right. Push it wirh an thin pin for about 2 secs.
> To finish pairing it may be necessary to push it (very short) again.

Thanks, but there doesn't seem to be a button in the hole either, I wonder if this not a Zigbee version...

Koenkk, 2020-02-16:
Probably only the `-M` version is Zigbee (https://www.zigbee2mqtt.io/devices/HS1SA-M.html#important)

closed 2020-04-25 by stale[bot]

ovizii, 2022-05-30:
> Probably only the `-M` version is Zigbee (https://www.zigbee2mqtt.io/devices/HS1SA-M.html#important)

that is the HS1SA, the HS3SA page does not say anything about different versions: https://www.zigbee2mqtt.io/devices/HS3SA.html