Koenkk/zigbee2mqtt #2947: Docker image v1.10 (latest) as of 14-Feb 2020

01kbpatel, 2020-02-14:
Just tried to upgrade to the latest docker image and get the following error:

```
- /app/index.js
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:793:17)
    at Function.Module._load (internal/modules/cjs/loader.js:686:27)
    at Module.require (internal/modules/cjs/loader.js:848:19)
    at require (internal/modules/cjs/helpers.js:74:18)
    at Object.<anonymous> (/app/node_modules/zigbee-herdsman-converters/ota/tradfri.js:1:15)
    at Module._compile (internal/modules/cjs/loader.js:955:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:991:10)
    at Module.load (internal/modules/cjs/loader.js:811:32)
    at Function.Module._load (internal/modules/cjs/loader.js:723:14)
    at Module.require (internal/modules/cjs/loader.js:848:19) {
  code: 'MODULE_NOT_FOUND',
  requireStack: [
    '/app/node_modules/zigbee-herdsman-converters/ota/tradfri.js',
    '/app/node_modules/zigbee-herdsman-converters/ota/index.js',
    '/app/node_modules/zigbee-herdsman-converters/devices.js',
    '/app/node_modules/zigbee-herdsman-converters/index.js',
    '/app/lib/util/utils.js',
    '/app/lib/util/settings.js',
    '/app/lib/util/logger.js',
    '/app/lib/mqtt.js',
    '/app/lib/controller.js',
    '/app/index.js'
  ]
}
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.10.0 start: `node index.js`
npm ERR! Exit status 1
```

Koenkk, 2020-02-14:
You probably have a custom zigbee-herdsman-converters mapped via a docker volume, can you try removing it?

01kbpatel, 2020-02-14:
Complete new install on dev machine, pulled zigbee-herdsman-converters from github into /opt mapped directory. Also tried delteing everything from /opt directory and deleteing container and restarting. Using docker-compose file on raspberry 3

Koenkk, 2020-02-15:
>  pulled zigbee-herdsman-converters from github into /opt mapped directory. 

Can you try without this? This isn't needed, only when adding support for new devices.

01kbpatel, 2020-02-15:
I have already tried this as well, but have same error!!

Koenkk, 2020-02-16:
What is the output of `docker inspect [CONTAINER_ID]`? (replace CONTAINER_ID with the actual one).

PedroLamas, 2020-02-17:
I also see this error if I map a version of the zigbee-herdsman-converters to the docker container.

The first line (missing from the report above) says it all:

```
Error: Cannot find module 'axios'
```

Would it be enough to just do `npm install` on the converters root?

Koenkk, 2020-02-17:
Yes just do npm install in the converters root

closed 2020-04-25 by stale[bot]