Koenkk/zigbee2mqtt #2948: OTA Update Available Sensor

kiall, 2020-02-14:
Now that OTA exists and works it would be awesome to add a sensor as part of the HA discovery info which shows the update status (up to date vs update available).

This would probably involve:

1) Periodically checking for updates for each device (to refresh the data)
2) Exposing over MQTT if an update is known to be available or not
3) Generating a sensor config for HA auto discovery to show this data

Ideally, this HA sensor discovery config would only show up for devices which have an OTA implementation (today, that's just IKEA Tradfri) and wouldn't show for devices that don't.

Related to #2921 - but I've filed it separately as it's probably a bigger change that isn't core to OTA support.

Koenkk, 2020-02-16:
Implemented, updated docs: https://github.com/Koenkk/zigbee2mqtt.io/blob/develop/docs/information/ota_updates.md

closed 2020-02-16 by Koenkk

kiall, 2020-02-16:
:+1: 

I've pulled + restarted. New sensors are showing populating with data :)

So far, 10 showing as update available.