Koenkk/zigbee2mqtt #2950: Osram Flex LED Strip 233 Error

irakic89, 2020-02-15:
After pairing Osram LED strip doesn't respond to commands (on/off, color change...).
I'm getting:
_zigbee2mqtt:error 2020-02-15 01:14:13: Publish 'set' 'state' to '0x7cb03eaa00a9ee71' failed: 'Error: Command 0x7cb03eaa00a9ee71/3 genOnOff.on({}, {"timeout":10000,"defaultResponseTimeout":15000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Data request failed with error: 'MAC no ack' (233))'_

I understood from earlier posts that this should be a distance issue, but signal strength to coordinator via multiple routers is more than strong, only direct signal to the coordinator is 0.

Tried to repair it close to the coordinator and then move it 5m away, but after some time, it stops responding. I read somewhere that coordinators had some issue with router to router connection, but that seems to be resolved in previous firmware for older devices. 

I'm using CC1352P_2, don't know if this is an issue within the latest firmware for this device?

Thanks

Koenkk, 2020-02-15:
I have an idea were this issue is coming from and can probably provide a fix for it. But before I can do this I need to make sure this is the issue.

Can you try sniffing the traffic when controlling the device fails? https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

irakic89, 2020-02-15:
Will do that tomorrow.
Thanks
;)

irakic89, 2020-02-16:
> I have an idea were this issue is coming from and can probably provide a fix for it. But before I can do this I need to make sure this is the issue.
> 
> Can you try sniffing the traffic when controlling the device fails? https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

[sniff_osram.zip](https://github.com/Koenkk/zigbee2mqtt/files/4210935/sniff_osram.zip)

I tried to turn on Osram Flex couple of times and you can find sniffed traffic in the attached zip file.
Let me know if you need anything else.

irakic89, 2020-02-18:
@Koenkk  Have you managed to open sniff network file?

Koenkk, 2020-02-20:
@irakic89 could you share your network key with me? (on telegram @koenkk)

closed 2020-03-04 by irakic89