Koenkk/zigbee2mqtt #2951: Problem binding Ikea E1524/E1810 to control Ikea LED1736G9

mlava, 2020-02-15:
Hi all,
I hope this isn't a noob question.

I have the listed remote and bulb paired in Z2M and getting appropriate messages in log for button clicks, bulb reporting is on as well.

I can control the bulb over mqtt.

I have bound the remote to the bulb. See log entries below:

```
Feb 15 07:41:25 raspberrypi npm[4081]: zigbee2mqtt:info  2020-02-15 07:41:25: Successfully bound cluster 'genScenes' from 'TradfriRemote1' to 'TradfriBulb1'
Feb 15 07:41:25 raspberrypi npm[4081]: zigbee2mqtt:info  2020-02-15 07:41:25: MQTT publish: topic '/zigbee2mqtt/bridge/log', payload '{"type":"device_bind","message":{"from":"TradfriRemote1","to":"TradfriBulb1","cluster":"genScenes"}}'
Feb 15 07:41:26 raspberrypi npm[4081]: zigbee2mqtt:info  2020-02-15 07:41:26: Successfully bound cluster 'genOnOff' from 'TradfriRemote1' to 'TradfriBulb1'
Feb 15 07:41:26 raspberrypi npm[4081]: zigbee2mqtt:info  2020-02-15 07:41:26: MQTT publish: topic '/zigbee2mqtt/bridge/log', payload '{"type":"device_bind","message":{"from":"TradfriRemote1","to":"TradfriBulb1","cluster":"genOnOff"}}'
Feb 15 07:41:27 raspberrypi npm[4081]: zigbee2mqtt:info  2020-02-15 07:41:27: Successfully bound cluster 'genLevelCtrl' from 'TradfriRemote1' to 'TradfriBulb1'
Feb 15 07:41:27 raspberrypi npm[4081]: zigbee2mqtt:info  2020-02-15 07:41:27: MQTT publish: topic '/zigbee2mqtt/bridge/log', payload '{"type":"device_bind","message":{"from":"TradfriRemote1","to":"TradfriBulb1","cluster":"genLevelCtrl"}}'
```

Despite what appears to be successful binding, the bulb is not responding to button pushes.

Am on:
Version of Zigbee2Mqtt: 1.9.0
Coordinator version: 20190608

Did I miss a step??
Thanks in advance!!

Koenkk, 2020-02-15:
Make sure to follow all steps from: https://www.zigbee2mqtt.io/devices/E1524_E1810.html#binding

mlava, 2020-02-16:
@Koenkk 
I apologise.
I followed the steps again to define a group for the bulb and then bind the remote and it works now. I thought I had already tried that way, but I must have been mistaken.
Sorry for wasting your time!

closed 2020-02-16 by mlava