Koenkk/zigbee2mqtt #2952: Updated the zigbee2mqtt 1.10.0 from 1.8.0 and have problem with IKEA bulb

spider7611, 2020-02-15:
Have problem after the upgrade. Can't switch on and off from domoticz. Only the IKEA remote switch is working. Sometimes can switch on with domoticz but after only the Ikea remote control is working.

In the log show this error:

`info  2020-02-15 08:45:35: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to 'IKEAbulb' failed: 'Error: Command 0x14b457fffe733d3a/1 genOnOff.off({}, {\"timeout\":10000,\"defaultResponseTimeout\":15000,\"manufacturerCode\":null,\"disableDefaultResponse\":false}) failed (Error: Data request failed with error: 'MAC no ack' (233))'","meta":{"friendly_name":"IKEAbulb"}}'
error 2020-02-15 08:45:54: Publish 'set' 'brightness' to 'IKEAbulb' failed: 'Error: Command 0x14b457fffe733d3a/1 genLevelCtrl.moveToLevelWithOnOff({"level":45,"transtime":0}, {"timeout":10000,"defaultResponseTimeout":15000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Data request failed with error: 'MAC no ack' (233))'
info  2020-02-15 08:45:54: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'brightness' to 'IKEAbulb' failed: 'Error: Command 0x14b457fffe733d3a/1 genLevelCtrl.moveToLevelWithOnOff({\"level\":45,\"transtime\":0}, {\"timeout\":10000,\"defaultResponseTimeout\":15000,\"manufacturerCode\":null,\"disableDefaultResponse\":false}) failed (Error: Data request failed with error: 'MAC no ack' (233))'","meta":{"friendly_name":"IKEAbulb"}}'
error 2020-02-15 08:45:59: Publish 'set' 'state' to 'IKEAbulb' failed: 'Error: Command 0x14b457fffe733d3a/1 genOnOff.off({}, {"timeout":10000,"defaultResponseTimeout":15000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Data request failed with error: 'MAC no ack' (233))'
info  2020-02-15 08:45:59: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to 'IKEAbulb' failed: 'Error: Command 0x14b457fffe733d3a/1 genOnOff.off({}, {\"timeout\":10000,\"defaultResponseTimeout\":15000,\"manufacturerCode\":null,\"disableDefaultResponse\":false}) failed (Error: Data request failed with error: 'MAC no ack' (233))'","meta":{"friendly_name":"IKEAbulb"}}'
error 2020-02-15 08:46:02: Publish 'set' 'brightness' to 'IKEAbulb' failed: 'Error: Command 0x14b457fffe733d3a/1 genLevelCtrl.moveToLevelWithOnOff({"level":255,"transtime":0}, {"timeout":10000,"defaultResponseTimeout":15000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Data request failed with error: 'MAC no ack' (233))'
info  2020-02-15 08:46:02: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'brightness' to 'IKEAbulb' failed: 'Error: Command 0x14b457fffe733d3a/1 genLevelCtrl.moveToLevelWithOnOff({\"level\":255,\"transtime\":0}, {\"timeout\":10000,\"defaultResponseTimeout\":15000,\"manufacturerCode\":null,\"disableDefaultResponse\":false}) failed (Error: Data request failed with error: 'MAC no ack' (233))'","meta":{"friendly_name":"IKEAbulb"}}'
error 2020-02-15 08:46:03: Publish 'set' 'state' to 'IKEAbulb' failed: 'Error: Command 0x14b457fffe733d3a/1 genOnOff.off({}, {"timeout":10000,"defaultResponseTimeout":15000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Data request failed with error: 'MAC no ack' (233))'`

Koenkk, 2020-02-15:
I have an idea were this issue is coming from and can probably provide a fix for it. But before I can do this I need to make sure this is the issue.

Can you try sniffing the traffic when controlling the device fails? https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

spider7611, 2020-02-15:
Done a clean install and now the Bulb and the remote control and domoticz working again together. But now have another problem with my aquara motion sensor. Can't pair it. Not recognized at all. 
Push the pairing button for 5 second, and the blue led blink 3 times after blink quickly about 2 seconds like everything is ok, but in the log nothing happend. Before it was working good. What happend with it, what you think?
Restart zigbee2mqtt lot of time and turn on off my  CC2530 router but no success... :(

Koenkk, 2020-02-16:
Check https://www.zigbee2mqtt.io/information/FAQ.html#no-logging-is-shown-at-all especially "If it’s a battery powered device, try replacing the battery."

If it doesn't help, try reflashing your CC2531.

spider7611, 2020-02-17:
Try take out the battery and wait more than one minutes. About 3x times make this procedure and finally catch the sign. Now working ok everything.

closed 2020-02-17 by spider7611