Koenkk/zigbee2mqtt #2954: Ikea E1743: failed to unbind default_bind_group

miknuc, 2020-02-16:
# Bug Report
## What happened
I tried to unbind my E1743 according to the instructions:
https://www.zigbee2mqtt.io/devices/E1743.html 

It failed (output from log):
```
  zigbee2mqtt:error 2/16/2020, 11:11:21 AM Failed to find device with entity ID 'default_bind_group'

  zigbee2mqtt:warn 2/16/2020, 11:11:21 AM Cannot handle MQTT message on 'zuhause/zigbee2mqtt/bridge/unbind/Ikea1' with data 'default_bind_group'
```

This is a new device, freshly paired.
After reading  https://github.com/Koenkk/zigbee2mqtt/issues/2924 I also tried to unbind from groups 0 and 901, with same result (failed to find device with entity id ...)
My only existing groups in configuration.yaml are 1 and 25014.

## What did you expect to happen

Device is unbound
## How to reproduce it (minimal and precise)
`mosquitto_pub -u XXX -P XXX -h mqtt -t "zuhause/zigbee2mqtt/bridge/unbind/Ikea1" -m "default_bind_group"`

## Debug Info
I am running zigbee2mqtt as docker container on a raspberry. Output from log file:
zigbee2mqtt version:
CC253X firmware version:
```
2/15/2020, 10:14:35 AM - info: Logging to directory: '/app/data/log/2020-02-15.10-14-33'
2/15/2020, 10:14:35 AM - info: Starting zigbee2mqtt version 1.5.1 (commit #unknown)
2/15/2020, 10:14:35 AM - info: Starting zigbee-shepherd
2/15/2020, 10:14:38 AM - info: zigbee-shepherd started
2/15/2020, 10:14:38 AM - info: Coordinator firmware version: '20190109'
```

Koenkk, 2020-02-16:
You are running an outdated zigbee2mqtt version, updating to 1.10 fixes it.

closed 2020-02-16 by Koenkk

tsprivate, 2020-07-13:
> You are running an outdated zigbee2mqtt version, updating to 1.10 fixes it.

I found the solution as it was not working with the latest zigbee2mqtt version:

1) Factory reset the E1743 (keep pairing pressed >10sec)
2) Pair E1743 (press pairing 4 x quickly till led flashes.... than ... keep pressing pairing slowly ..until pairing is confirmed
3) use zigbee2mqtt/bridge/unbind   and use ieeeAdr 0xeaaaaaaa4  as payload !! (will not work with friendly_Name (e.g. Switch.Bedroom)

Missing any of these steps and unbind was not working ... just bought a bew E1743 and was successful on first run with these steps

maz3max, 2021-05-05:
> I found the solution as it was not working with the latest zigbee2mqtt version:
>
>     1. Factory reset the E1743 (keep pairing pressed >10sec)
>
>     2. Pair E1743 (press pairing 4 x quickly till led [...]

For me, following the current documentation works. No need to tinker with this method.

BleesTFE, 2021-11-07:
I am having this issue now with the latest version of z2m gfor home assistant @Koenkk

Tuschz, 2021-11-08:
Hi,

I am having the same problem. But only with the devices with updated FW.

[image]