Koenkk/zigbee2mqtt #2956: NEO Sirene / Temp / Humi sensors

ersiensi, 2020-02-17:
Trigger alarm, Make sound and light, Temperature/humidity sensor with anti-dismantle function
https://szneo.com/en/products/show.php?id=241

Hi. 2 days Iam learning how to add this sirene, but its beyond my powers. I cant make converter. Can you help me please?

**Log:**

zigbee2mqtt:debug 2020-02-17 03:22:44: Received Zigbee message from 'NEO Sirene', type 'raw', cluster 'manuSpecificTuyaDimmer', data '{"type":"Buffer","data":[9,87,2,0,2,105,2,0,4,0,0,0,194]}' from endpoint 1 with groupID 0
zigbee2mqtt:warn  2020-02-17 03:22:44: Received message from unsupported device with Zigbee model 'neo_sirene'
zigbee2mqtt:warn  2020-02-17 03:22:44: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:debug 2020-02-17 03:22:45: Received Zigbee message from 'NEO Sirene', type 'raw', cluster 'manuSpecificTuyaDimmer', data '{"type":"Buffer","data":[9,88,2,0,2,106,2,0,4,0,0,0,68]}' from endpoint 1 with groupID 0
zigbee2mqtt:warn  2020-02-17 03:22:45: Received message from unsupported device with Zigbee model 'neo_sirene'
zigbee2mqtt:warn  2020-02-17 03:22:45: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:debug 2020-02-17 03:22:46: Received Zigbee message from 'NEO Sirene', type 'raw', cluster 'manuSpecificTuyaDimmer', data '{"type":"Buffer","data":[9,89,2,0,2,106,2,0,4,0,0,0,69]}' from endpoint 1 with groupID 0
zigbee2mqtt:warn  2020-02-17 03:22:46: Received message from unsupported device with Zigbee model 'neo_sirene'
zigbee2mqtt:warn  2020-02-17 03:22:46: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:debug 2020-02-17 03:23:42: Received Zigbee message from 'NEO Sirene', type 'raw', cluster 'manuSpecificTuyaDimmer', data '{"type":"Buffer","data":[9,90,2,0,2,106,2,0,4,0,0,0,70]}' from endpoint 1 with groupID 0
zigbee2mqtt:warn  2020-02-17 03:23:42: Received message from unsupported device with Zigbee model 'neo_sirene'
zigbee2mqtt:warn  2020-02-17 03:23:42: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:debug 2020-02-17 03:23:43: Received Zigbee message from 'NEO Sirene', type 'raw', cluster 'manuSpecificTuyaDimmer', data '{"type":"Buffer","data":[9,91,2,0,2,105,2,0,4,0,0,0,197]}' from endpoint 1 with groupID 0
zigbee2mqtt:warn  2020-02-17 03:23:43: Received message from unsupported device with Zigbee model 'neo_sirene'
zigbee2mqtt:warn  2020-02-17 03:23:43: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:debug 2020-02-17 03:23:44: Received Zigbee message from 'NEO Sirene', type 'raw', cluster 'manuSpecificTuyaDimmer', data '{"type":"Buffer","data":[9,92,2,0,2,106,2,0,4,0,0,0,65]}' from endpoint 1 with groupID 0
zigbee2mqtt:warn  2020-02-17 03:23:44: Received message from unsupported device with Zigbee model 'neo_sirene'
zigbee2mqtt:warn  2020-02-17 03:23:44: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:debug 2020-02-17 03:23:45: Received Zigbee message from 'NEO Sirene', type 'raw', cluster 'manuSpecificTuyaDimmer', data '{"type":"Buffer","data":[9,93,2,0,2,106,2,0,4,0,0,0,64]}' from endpoint 1 with groupID 0
zigbee2mqtt:warn  2020-02-17 03:23:45: Received message from unsupported device with Zigbee model 'neo_sirene'
zigbee2mqtt:warn  2020-02-17 03:23:45: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:debug 2020-02-17 03:23:46: Received Zigbee message from 'NEO Sirene', type 'raw', cluster 'manuSpecificTuyaDimmer', data '{"type":"Buffer","data":[9,94,2,0,2,106,2,0,4,0,0,0,63]}' from endpoint 1 with groupID 0
zigbee2mqtt:warn  2020-02-17 03:23:46: Received message from unsupported device with Zigbee model 'neo_sirene'
zigbee2mqtt:warn  2020-02-17 03:23:46: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:debug 2020-02-17 03:23:48: Received Zigbee message from 'NEO Sirene', type 'raw', cluster 'manuSpecificTuyaDimmer', data '{"type":"Buffer","data":[9,95,2,0,2,105,2,0,4,0,0,0,198]}' from endpoint 1 with groupID 0
zigbee2mqtt:warn  2020-02-17 03:23:48: Received message from unsupported device with Zigbee model 'neo_sirene'
zigbee2mqtt:warn  2020-02-17 03:23:48: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.

**from.Zigbee.js**

    neo_temperature: {
        cluster: 'manuSpecificTuyaDimmer',
        type: 'raw',
        convert: (model, msg, publish, options) => {
          const temperature = parseFloat(msg.data['measuredValue']) / 100.0;
          return {temperature: temperature};
        },
    },
    neo_battery: {
        cluster: 'manuSpecificTuyaDimmer',
        type: ['raw'],
        convert: (model, msg, publish, options, meta) => {
            if (msg.data.hasOwnProperty('batteryPercentageRemaining')) {
                return {battery: msg.data['batteryPercentageRemaining']};
            }
        },
    },

**devices.js**

    {
        zigbeeModel: ['neo_sirene'],
        model: 'neo_sirene',
        vendor: 'neo',
        description: 'Temperature & humidity sensor',
        supports: 'temperature and humidity',
        fromZigbee: [fz.battery_percentage_remaining, fz.neo_temperature],
        toZigbee: [],
    },

Thank you.

ersiensi, 2020-02-21:
Can someone help me please?

Koenkk, 2020-02-23:
What is currently working and what now?

I think you can get the sound/light working by using

```js
{
    zigbeeModel: ['neo_sirene'],
    model: 'neo_sirene',
    vendor: 'neo',
    description: 'Temperature & humidity sensor',
    supports: 'temperature and humidity',
    fromZigbee: [fz.neo_battery, fz.neo_temperature],
    toZigbee: [tz.warning],
},
```

and now send to `zigbee2mqtt/NEO Sirene/set` with payload `{"warning": {"mode": "emergency"}}`

ersiensi, 2020-02-24:
Hi Koen, thank you for your answer. 
I cant send payload, because I dont have this sirene in MQTT integrations. Or can I?

**These are raw data from log without any intervention in my settings:**
zigbee2mqtt:debug 2020-02-24 12:07:46: Received Zigbee message from 'Sirena', type 'raw', cluster 'manuSpecificTuyaDimmer', data '{"type":"Buffer","data":[9,120,2,0,2,105,2,0,4,0,0,0,217]}' from endpoint 1 with groupID 0
zigbee2mqtt:warn  2020-02-24 12:07:46: Received message from unsupported device with Zigbee model '0yu2xgi'
zigbee2mqtt:warn  2020-02-24 12:07:46: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:debug 2020-02-24 12:07:47: Received Zigbee message from 'Sirena', type 'raw', cluster 'manuSpecificTuyaDimmer', data '{"type":"Buffer","data":[9,121,2,0,2,106,2,0,4,0,0,0,46]}' from endpoint 1 with groupID 0
zigbee2mqtt:warn  2020-02-24 12:07:47: Received message from unsupported device with Zigbee model '0yu2xgi'
zigbee2mqtt:warn  2020-02-24 12:07:47: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:debug 2020-02-24 12:07:50: Received Zigbee message from 'Sirena', type 'raw', cluster 'manuSpecificTuyaDimmer', data '{"type":"Buffer","data":[9,122,2,0,2,105,2,0,4,0,0,0,218]}' from endpoint 1 with groupID 0
zigbee2mqtt:warn  2020-02-24 12:07:50: Received message from unsupported device with Zigbee model '0yu2xgi'
zigbee2mqtt:warn  2020-02-24 12:07:50: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.

**database.db**
{"id":4,"type":"EndDevice","ieeeAddr":"0xec1bbdfffe6c1f45","nwkAddr":24610,"manufId":0,"manufName":"_TYST11_d0yu2xgi","powerSource":"Battery","modelId":"0yu2xgi\u0000","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":0,"inClusterList":[0,3],"outClusterList":[3,25],"clusters":{"genBasic":{"attributes":{"modelId":"0yu2xgi\u0000","manufacturerName":"_TYST11_d0yu2xgi","powerSource":3,"zclVersion":3,"appVersion":73,"stackVersion":0,"hwVersion":1,"dateCode":"20180727"}}},"binds":[]}},"appVersion":73,"stackVersion":0,"hwVersion":1,"dateCode":"20180727","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1582541805739}

Original modelId from output is :"0yu2xgi\u0000" and manufacturerName:"_TYST11_d0yu2xgi"

**devices.js**
	{
		zigbeeModel: ['_TYST11_d0yu2xgi'],
		model: '0yu2xgi\u0000',
		vendor: 'neo',
		description: 'Temperature & humidity sensor',
		supports: 'temperature and humidity',
		fromZigbee: [fz.neo_battery, fz.neo_temperature],
		toZigbee: [tz.warning],
	},

**fromZigbee.js**
	0yu2xgi\u0000_temperature: {
		cluster: 'manuSpecificTuyaDimmer',
		type: 'raw',
		convert: (model, msg, publish, options) => {
			const temperature = parseFloat(msg.data['measuredValue']) / 100.0;
			return {temperature: temperature};
		},
	},
	0yu2xgi\u0000_battery: {
		cluster: 'manuSpecificTuyaDimmer',
		type: ['raw'],
		convert: (model, msg, publish, options, meta) => {
			if (msg.data.hasOwnProperty('batteryPercentageRemaining')) {
				return {battery: msg.data['batteryPercentageRemaining']};
			}
		},
	},	

I've tried everything but it is still unsupported device, I still don't see sirene in MQTT, I don't know what to do next. :/
Can this 0yu2xgi\u0000 "\u0000" like space or blank character cause integration problems? Or what Iam doing bad?
Thank you Koen for your time and help.

Koenkk, 2020-02-25:
Try with:

```js
{
    zigbeeModel: ['0yu2xgi\u0000'],
    model: 'neo_sirene',
    vendor: 'neo',
    description: 'Temperature & humidity sensor',
    supports: 'temperature and humidity',
    fromZigbee: [fz.neo_battery, fz.neo_temperature],
    toZigbee: [tz.warning],
},
```

You can send the MQTT commands with e.g. MQTT.fx (https://mqttfx.jensd.de/)

tlokaj, 2020-02-29:
Hi,

I have the same problem. I'm not able to decode the message '{"type":"Buffer","data":[9,87,2,0,2,105,2,0,4,0,0,0,194]}'.

And the same problem I have with this device https://szneo.com/en/products/index.php?id=70.

closed 2020-05-06 by stale[bot]

jvrrr, 2020-06-09:
I have the same issue, this device is not reconized.
What do you need from me to solve the problem.

Thanks a lot!