Koenkk/zigbee2mqtt #2958: CC1352P-2 stability issues across the board

timdonovanuk, 2020-02-17:
## Debug Info
zigbee2mqtt version: tried zigbee2mqtt1.10.1 and zigbee2mqtt1.10.1-edge
CC253X firmware version: CC1352P_2_20191106

Am I the only one having massive issues with the CC1352P-2 across the board? I added a few devices then it stopped being able to pair anything (no error, it just wouldn't show initiation of paring). So I removed everything, changed PAN ID, and started again. Finally could add deviced but...

- Takes about 8 attempts to pair any Ikea bulbs with "Interview failed for...."
- When they are finally added, they eventually stop working with either 
     - Error: Data request failed with error: 'MAC channel access failure' (225)
     - Error: Data request failed with error: 'No network route' (205)

timdonovanuk, 2020-02-17:
Here is it failing to add an Ikea bulb:

```
2020-02-17T12:05:54.478Z zigbee-herdsman:controller:log Device '0x000b57fffe9b738b' joined
2020-02-17T12:05:54.479Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-02-17T12:05:54.479Z zigbee-herdsman:controller:log Device '0x000b57fffe9b738b' accepted by handler
2020-02-17T12:05:54.479Z zigbee-herdsman:controller:log New device '0x000b57fffe9b738b' joined
2020-02-17T12:05:54.479Z zigbee-herdsman:controller:log Creating device '0x000b57fffe9b738b'
zigbee2mqtt:info  2020-02-17 12:05:54: Device 'bedroom_1_ceiling_light_2' joined
zigbee2mqtt:info  2020-02-17 12:05:54: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"bedroom_1_ceiling_light_2"}}'
2020-02-17T12:05:54.484Z zigbee-herdsman:controller:log Interview '0x000b57fffe9b738b' start
zigbee2mqtt:info  2020-02-17 12:05:54: Starting interview of 'bedroom_1_ceiling_light_2'
zigbee2mqtt:info  2020-02-17 12:05:54: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"bedroom_1_ceiling_light_2"}}'
2020-02-17T12:05:54.487Z zigbee-herdsman:controller:device Interview - start device '0x000b57fffe9b738b'
2020-02-17T12:05:54.489Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":43977,"nwkaddrofinterest":43977}
2020-02-17T12:05:54.489Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,201,171,201,171,35]
2020-02-17T12:05:54.493Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102]
2020-02-17T12:05:54.493Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102]
2020-02-17T12:05:54.493Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102
2020-02-17T12:05:54.493Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0}
2020-02-17T12:05:54.493Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-02-17T12:06:04.492Z zigbee-herdsman:controller:device Interview - node descriptor request failed for '0x000b57fffe9b738b', attempt 1
2020-02-17T12:06:04.492Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":43977,"nwkaddrofinterest":43977}
2020-02-17T12:06:04.493Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,201,171,201,171,35]
2020-02-17T12:06:04.496Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102]
2020-02-17T12:06:04.496Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102]
2020-02-17T12:06:04.496Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102
2020-02-17T12:06:04.497Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0}
2020-02-17T12:06:04.497Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-02-17T12:06:14.495Z zigbee-herdsman:controller:device Interview - node descriptor request failed for '0x000b57fffe9b738b', attempt 2
2020-02-17T12:06:14.495Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":43977,"nwkaddrofinterest":43977}
2020-02-17T12:06:14.496Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,201,171,201,171,35]
2020-02-17T12:06:14.499Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102]
2020-02-17T12:06:14.499Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102]
2020-02-17T12:06:14.500Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102
2020-02-17T12:06:14.500Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0}
2020-02-17T12:06:14.500Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-02-17T12:06:24.498Z zigbee-herdsman:controller:device Interview - node descriptor request failed for '0x000b57fffe9b738b', attempt 3
2020-02-17T12:06:24.499Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":43977,"nwkaddrofinterest":43977}
2020-02-17T12:06:24.499Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,201,171,201,171,35]
2020-02-17T12:06:24.503Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102]
2020-02-17T12:06:24.503Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102]
2020-02-17T12:06:24.503Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102
2020-02-17T12:06:24.503Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0}
2020-02-17T12:06:24.503Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-02-17T12:06:34.501Z zigbee-herdsman:controller:device Interview - node descriptor request failed for '0x000b57fffe9b738b', attempt 4
2020-02-17T12:06:34.502Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":43977,"nwkaddrofinterest":43977}
2020-02-17T12:06:34.502Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,201,171,201,171,35]
2020-02-17T12:06:34.506Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102]
2020-02-17T12:06:34.506Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102]
2020-02-17T12:06:34.506Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102
2020-02-17T12:06:34.506Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0}
2020-02-17T12:06:34.506Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
[... cut ...]
2020-02-17T12:07:14.513Z zigbee-herdsman:controller:error Interview failed for '0x000b57fffe9b738b with error 'Error: Interview failed because can not get active endpoints ('0x000b57fffe9b738b')'
zigbee2mqtt:error 2020-02-17 12:07:14: Failed to interview 'bedroom_1_ceiling_light_2', device has not successfully been paired
zigbee2mqtt:info  2020-02-17 12:07:14: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"bedroom_1_ceiling_light_2"}}'
```

timdonovanuk, 2020-02-17:
Now the CC1352P-2 is back in the state where nothing will join, no matter what I do:

```
2020-02-17T12:16:46.426Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-02-17T12:16:46.426Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvRead - {"id":99,"offset":0,"osal":true,"product":-1}
2020-02-17T12:16:46.426Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,3,33,8,99,0,0,73]
2020-02-17T12:16:46.433Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,97,8,0,1,0,107]
2020-02-17T12:16:46.433Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,97,8,0,1,0,107]
2020-02-17T12:16:46.433Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 3 - 1 - 8 - [0,1,0] - 107
2020-02-17T12:16:46.433Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - osalNvRead - {"status":0,"len":1,"value":{"type":"Buffer","data":[0]}}
2020-02-17T12:16:46.433Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-02-17T12:16:46.433Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvRead - {"id":132,"offset":0,"osal":true,"product":-1}
2020-02-17T12:16:46.433Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,3,33,8,132,0,0,174]
2020-02-17T12:16:46.440Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,6,97,8,0,4,0,8]
2020-02-17T12:16:46.440Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,6,97,8,0,4,0,8]
2020-02-17T12:16:46.440Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [0,0,99]
2020-02-17T12:16:46.440Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,6,97,8,0,4,0,8,0,0,99]
2020-02-17T12:16:46.440Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 6 - 3 - 1 - 8 - [0,4,0,8,0,0] - 99
2020-02-17T12:16:46.440Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - osalNvRead - {"status":0,"len":4,"value":{"type":"Buffer","data":[0,8,0,0]}}
2020-02-17T12:16:46.440Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-02-17T12:16:46.441Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - nvLength - {"sysid":1,"itemid":4,"subid":0,"product":1,"osal":false,"offset":0}
2020-02-17T12:16:46.441Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,5,33,50,1,4,0,0,0,19]
2020-02-17T12:16:46.444Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,4,97,50,20,0,0,0]
2020-02-17T12:16:46.444Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,4,97,50,20,0,0,0]
2020-02-17T12:16:46.444Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [67]
2020-02-17T12:16:46.444Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,4,97,50,20,0,0,0,67]
2020-02-17T12:16:46.444Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 4 - 3 - 1 - 50 - [20,0,0,0] - 67
2020-02-17T12:16:46.444Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - nvLength - {"len":20}
2020-02-17T12:16:46.444Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-02-17T12:16:46.444Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - nvRead - {"sysid":1,"itemid":4,"subid":0,"product":1,"osal":false,"offset":0,"len":20}
2020-02-17T12:16:46.445Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,8,33,51,1,4,0,0,0,0,0,20,11]
2020-02-17T12:16:46.448Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,22,97,51,0,20,0,0]
2020-02-17T12:16:46.448Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,22,97,51,0,20,0,0]
2020-02-17T12:16:46.448Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [0,0,0,0,0,0,0,0]
2020-02-17T12:16:46.448Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,22,97,51,0,20,0,0,0,0,0,0,0,0,0,0]
2020-02-17T12:16:46.449Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [0,0,0,0,0,0,255,0]
2020-02-17T12:16:46.450Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,22,97,51,0,20,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,255,0]
2020-02-17T12:16:46.450Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [0,0,175]
2020-02-17T12:16:46.450Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,22,97,51,0,20,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,255,0,0,0,175]
2020-02-17T12:16:46.450Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 22 - 3 - 1 - 51 - [0,20,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,255,0,0,0] - 175
2020-02-17T12:16:46.450Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - nvRead - {"status":0,"len":20,"value":{"type":"Buffer","data":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,255,0,0,0]}}
2020-02-17T12:16:46.450Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-02-17T12:16:46.450Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - nvLength - {"sysid":1,"itemid":7,"subid":0,"product":1,"osal":false,"offset":0}
2020-02-17T12:16:46.450Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,5,33,50,1,7,0,0,0,16]
2020-02-17T12:16:46.454Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,4,97,50,12,0,0,0,91]
2020-02-17T12:16:46.454Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,4,97,50,12,0,0,0,91]
2020-02-17T12:16:46.454Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 4 - 3 - 1 - 50 - [12,0,0,0] - 91
2020-02-17T12:16:46.454Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - nvLength - {"len":12}
2020-02-17T12:16:46.454Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-02-17T12:16:46.455Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - nvRead - {"sysid":1,"itemid":7,"subid":0,"product":1,"osal":false,"offset":0,"len":12}
2020-02-17T12:16:46.455Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,8,33,51,1,7,0,0,0,0,0,12,16]
[... cut ...]
zigbee2mqtt:debug 2020-02-17 12:16:51: Received MQTT message on 'zigbee2mqtt/bridge/config/devices' with data '[{"ieeeAddr":"0x00124b001ca609fd","type":"Coordinator","networkAddress":0,"friendly_name":"Coordinator","softwareBuildID":"zStack3x0","dateCode":"20191106","lastSeen":1581941811530}]'
zigbee2mqtt:info  2020-02-17 12:16:51: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"devices","message":[{"ieeeAddr":"0x00124b001ca609fd","type":"Coordinator","networkAddress":0,"friendly_name":"Coordinator","softwareBuildID":"zStack3x0","dateCode":"20191106","lastSeen":1581941811532}]}'
```

timdonovanuk, 2020-02-17:
I had to unplug the CC1352P-2 and then reboot HassOS to get it to accept joining devices again...

timdonovanuk, 2020-02-17:
But then thats it. Added one device succesfully and it doesn't even show interviewing anything else after that. Despite:

```
2020-02-17T12:48:35.397Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-02-17T12:50:30.737Z zigbee-herdsman:controller:log Permit joining
2020-02-17T12:50:30.738Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - mgmtPermitJoinReq - {"addrmode":15,"dstaddr":65532,"duration":254,"tcsignificance":0}
2020-02-17T12:50:30.738Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,5,37,54,15,252,255,254,0,228]
2020-02-17T12:50:30.751Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,54,0,82,254,3]
2020-02-17T12:50:30.751Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,54,0,82,254,3]
2020-02-17T12:50:30.751Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 54 - [0] - 82
2020-02-17T12:50:30.751Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - mgmtPermitJoinReq - {"status":0}
2020-02-17T12:50:30.751Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3]
2020-02-17T12:50:30.751Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [69,182,0,0,0,240]
2020-02-17T12:50:30.751Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,182,0,0,0,240]
2020-02-17T12:50:30.751Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 182 - [0,0,0] - 240
2020-02-17T12:50:30.752Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - mgmtPermitJoinRsp - {"srcaddr":0,"status":0}
2020-02-17T12:50:30.752Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
```

timdonovanuk, 2020-02-17:
Latest...I went as far as reflashing the firmware to the CC1352P-2. Now everything is working great (even paring multiple Ikea bulbs at once). This is a little worrying as suggests the firmware is breakable.

tsprivate, 2020-02-17:
Hi @timdonovanuk ,

i do not have all your issues but the last one

-     Error: Data request failed with error: 'MAC channel access failure' (225)
-     Error: Data request failed with error: 'MAC no ack' (233))
-     Error: Data request failed with error: 'No network route' (205)

My system is working great for a while (24h-48h) but than I realize that some devices (others work fine) get sluggish and having the above mentioned issues.

to get this sorted I usually have to
- stop/start the zigbee2mqtt service on my RPI
- or even reboot the RPI

Does anybody else have this issue after a period of time ? ... Any idea on how to get it fixed?

Koenkk, 2020-02-18:
I'm currently working on fixing/investigating the 205, 233, 225 and 240 errors

dzungpv, 2020-02-18:
> 
> 
> I'm currently working on fixing/investigating the 205, 233, 225 and 240 errors

Hope for your success. They are long time error.

timdonovanuk, 2020-02-18:
Had reasonable success yesterday, but just come home to a household of "MAC channel access failure' (225)", and nothing works anymore :/

timdonovanuk, 2020-02-18:
Also a bunch of these which is new:

```
zigbee2mqtt:info  2020-02-18 20:11:04: MQTT publish: topic 'zigbee2mqtt/bedroom_1_ceiling_light_1', payload '{"state":"OFF","brightness":254,"color_temp":377,"linkquality":69,"color":{"x":0.4599,"y":0.4106}}'
(node:57) UnhandledPromiseRejectionWarning: Error: SREQ '--> ZDO - extRouteDisc - {"dstAddr":25347,"options":2,"radius":30}' failed with status '199' (expected '0')
    at Znp.<anonymous> (/zigbee2mqtt-1.10.0/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:246:27)
    at Generator.next (<anonymous>)
    at fulfilled (/zigbee2mqtt-1.10.0/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:5:58)
(node:57) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 16)
```

dede34fr, 2020-02-22:
I've got the 205 network route error too with a CC1352P-2 taking over a CC2531. Need to restart Zigbee2mqtt addon from time to time too (hassos - rpi4).
`zigbee2mqtt:error 2020-02-22 10:18:32: Publish 'set' 'state' to '0x00158d000398e829' failed: 'Error: Command 0x00158d000398e829/1 genOnOff.off({}, {"timeout":10000,"defaultResponseTimeout":15000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Data request failed with error: 'No network route' (205))'
zigbee2mqtt:info  2020-02-22 10:18:32: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to '0x00158d000398e829' failed: 'Error: Command 0x00158d000398e829/1 genOnOff.off({}, {\"timeout\":10000,\"defaultResponseTimeout\":15000,\"manufacturerCode\":null,\"disableDefaultResponse\":false}) failed (Error: Data request failed with error: 'No network route' (205))'","meta":{"friendly_name":"0x00158d000398e829"}}'`

Koenkk, 2020-02-22:
Plesse try with the latest dev branch (contains an attempt to fix these issues)

CodeFinder2, 2020-02-22:
> Latest...I went as far as reflashing the firmware to the CC1352P-2. Now everything is working great (even paring multiple Ikea bulbs at once). This is a little worrying as suggests the firmware is breakable.

I had similar issues. In particular, changing the config in the hassio plugin (although not affecting anything that actually requires repairing like pan ID or network key) seems to be "dangerous" for now. I took the "reflash + repair" way once some time ago. And a few days ago, I got my setup working again by repairing everything (reflashing was not necessary whatsoever). But practically, that's a mess 😞 

Glad that @Koenkk is taking a look at the 205, 233, 225, and 240 errors and HOPEFULLY they are related / the source of our issues.

DomiStyle, 2020-02-28:
I just got my CC1352P-2 on monday and everything worked until now but suddenly the network stopped working with these messages:

    (node:23) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 14)
    Publish 'set' 'brightness' to 'bathroom' failed: 'Error: Command 2 genLevelCtrl.moveToLevelWithOnOff({"level":95,"transtime":0}) failed (Error: SRSP - AF - dataRequestExt after 6000ms)'
    MQTT publish: topic 'zigbee/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'brightness' to 'bathroom' failed: 'Error: Command 2 genLevelCtrl.moveToLevelWithOnOff({\"level\":95,\"transtime\":0}) failed (Error: SRSP - AF - dataRequestExt after 6000ms)'","meta":{"friendly_name":"bathroom"}}'
    (node:23) UnhandledPromiseRejectionWarning: Error: AREQ - AF - dataConfirm after 3000ms
        at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
        at listOnTimeout (internal/timers.js:549:17)
        at processTimers (internal/timers.js:492:7)

After a restart of zigbee2mqtt the network still fails to recover:

    Using '/app/data' as data directory

    zigbee2mqtt@1.10.0-dev start /app
    node index.js

    Logging to console and directory: '/app/data/log/2020-02-28.07-33-39' filename: log.txt
    Starting zigbee2mqtt version 1.10.0-dev (commit #9cc84fe)
    Starting zigbee-herdsman...
    Error while starting zigbee-herdsman
    Failed to start zigbee
    Exiting...
    Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
        at ZStackAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:72:23)
        at Generator.throw (<anonymous>)
        at rejected (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:6:65)
    npm ERR! code ELIFECYCLE
    npm ERR! errno 1
    npm ERR! zigbee2mqtt@1.10.0-dev start: `node index.js`
    npm ERR! Exit status 1
    npm ERR! 
    npm ERR! Failed at the zigbee2mqtt@1.10.0-dev start script.
    npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

    npm ERR! A complete log of this run can be found in:
    npm ERR!     /root/.npm/_logs/2020-02-28T06_33_46_894Z-debug.log

Had to reboot the Pi to get things working again. I will try to get debug logs if it happens again.

dede34fr, 2020-02-29:
Tried the latest dev branch but no change. Zigbee2mqtt starts well with CC1352P-2, all devices discovered, but impossible to change the state of a single light in lovelace or automation (error 205 no network route). I tried the following with hassio on RPI4 : reinstall the addon, start / stop zigbee2mqtt, copy the old database (working with CC2531 before), and even flashed the zigbee 3.0 firmware again.

Koenkk, 2020-02-29:
@dede34fr could you flash your CC2531 as a sniffer and sniff the network when you get these errors? You can send the snif log to me on telegram (@koenkk)

dede34fr, 2020-03-01:
Sorry but I'm not using a CC2531 ; that's a new CC1352P-2. I migrated from CC2531 indeed.

Koenkk, 2020-03-02:
@dede34fr I mean that you can use the CC2531 as the sniffer to find out what goes wrong when using the CC1352P-2 as the coordinator.

dede34fr, 2020-03-05:
How do I do that? I've a RPI4 64 running hassio and use a Windows 10 PC. Do I plug the CC2531 onto it? which software should I use? please describe the detailed procedure and I'll do it. Tks

sjorge, 2020-03-05:
@dede34fr it's documented on the site: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html hopefully this helps.

dede34fr, 2020-03-07:
> @dede34fr could you flash your CC2531 as a sniffer and sniff the network when you get these errors? You can send the snif log to me on telegram (@Koenkk) https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

Hi, snif log sent to you on telegram (@Koenkk). Could add HA and zigbee2mqtt logs if helpful. Tks

dede34fr, 2020-03-14:
Solved. Repaired all devices from empty db with CC1352P-2, working fine, even behind the metal door between the laundry room and the garage. 
Also forgot to set permit_join to true in the default addon config ! Sorry for that Koen.

timdonovanuk, 2020-03-14:
So you..started from scratch? I'm not sure this is a solution - I started from scratch twice but within a day was seeing errors.

timdonovanuk, 2020-04-04:
I'm still facing: 

- unable to add some Ikea bulbs, despite dozens of attempts
- some ikea bulbs do work, but performance on the network is _strange_ - i.e. some bulbs turn on/off with a delay of up to 30 seconds, constantly report unresponsive etc
- the entire z2m network is now down - no errors in logs but not a single device is responsive
- z2m won't add any devices (it doesn't even see or report that new devices are trying to join)

I've flashed with the latest firmware and running on dev branch but really at a loss. Going to try and flash an old stick to act as a zigbee sniffer and collect logs, but there are so many different issues that happen at random times I don't even know where or when to begin collecting logs.

Koenkk, 2020-04-04:
I would suggest analyzing them one by one.

Lets start with: the entire z2m network is now down - no errors in logs but not a single device is responsive

Could you provide herdsman debug and sniff logs of this?

dzungpv, 2020-04-04:
Do you have any Router, it yes, seem it is unknown problem and we still could not fix it: https://e2e.ti.com/support/wireless-connectivity/zigbee-and-thread/f/158/t/890697

timdonovanuk, 2020-04-06:
Thanks @Koenkk - I finally got my lazy ass to wire up my CC2531 to reflash for sniffing! But...

you know when you manage to get a doctors appointment, by the time you see the doctor your illness is fixed? :) 

I booted up z2m today and the bulb I've not managed to pair in months suddenly paired first time. The only change is I pulled the latest dev branch (I've been doing this whenever there is a new one, but it seems there was one 18 hours ago I'd not tried).

Even the bulbs that act oddly are now responding fast. However, I did manage to capture a bulb failing to join.

I have two captures to share:

1) Just general startup + powering on some bulbs that have been problematic for a while. Probably lots of spurious errors in this, as devices were powered off

2) Adding a bulb that fails around 13:47 (failed for device '0x000b57fffe392635' with error 'Error: Interview failed because can not get node descriptor ('0x000b57fffe392635'))

And of course the herdsman log that ran throughout it all..

Whats the best way to share the files? Many thanks.

Koenkk, 2020-04-06:
Regarding the node descriptor error, can you try again with the latest dev branch?

timdonovanuk, 2020-04-07:
My docker (and dockerhub) isn't showing a devbranch newer than the one I'm already using:

/opt/zigbee2mqtt/log$ less zigbee2mqtt.log | grep "Starting zigbee"
info  2020-04-06 16:08:05: Starting zigbee2mqtt version 1.12.0-dev (commit #418b3d1)
info  2020-04-06 16:08:05: Starting zigbee-herdsman...
info  2020-04-06 16:20:02: Starting zigbee2mqtt version 1.12.0-dev (commit #418b3d1)
info  2020-04-06 16:20:02: Starting zigbee-herdsman...
info  2020-04-06 16:38:15: Starting zigbee2mqtt version 1.12.0-dev (commit #418b3d1)
info  2020-04-06 16:38:15: Starting zigbee-herdsman...
info  2020-04-06 16:44:55: Starting zigbee2mqtt version 1.12.0-dev (commit #418b3d1)
info  2020-04-06 16:44:55: Starting zigbee-herdsman...
info  2020-04-06 16:59:21: Starting zigbee2mqtt version 1.12.0-dev (commit #418b3d1)
info  2020-04-06 16:59:21: Starting zigbee-herdsman...
info  2020-04-06 17:17:55: Starting zigbee2mqtt version 1.12.0-dev (commit #418b3d1)
info  2020-04-06 17:17:55: Starting zigbee-herdsman...
info  2020-04-07 10:22:33: Starting zigbee2mqtt version 1.12.0-dev (commit #418b3d1)
info  2020-04-07 10:22:33: Starting zigbee-herdsman...
info  2020-04-07 10:24:22: Starting zigbee2mqtt version 1.12.0-dev (commit #418b3d1)
info  2020-04-07 10:24:22: Starting zigbee-herdsman...
info  2020-04-07 10:25:50: Starting zigbee2mqtt version 1.12.0-dev (commit #418b3d1)
info  2020-04-07 10:25:50: Starting zigbee-herdsman...

I got the bulb added by moving it bit closer and trying a few times. It was already 4 meters LOS to the coordinator but hey ho.

timdonovanuk, 2020-04-07:
I think things are working fairly okay now, but lets see. My herdsman debug log is littered with errors like this constantly though:

```
winston:file logged 266884 info  2020-04-07 10:41:15: MQTT publish: topic 'zigbee2mqtt/bathroom_2_climate_sensor', payload '{"temperature":21.65,"linkquality":36,"humidity":71.33,"pressure":1022.1,"battery":100,"voltage":3065}'
2020-04-07T09:41:15.128Z serialport/stream _read reading { start: 6807, toRead: 58729 }
2020-04-07T09:41:15.128Z serialport/binding-abstract read
2020-04-07T09:41:15.128Z serialport/bindings/unixRead Starting read
2020-04-07T09:41:15.129Z serialport/bindings/unixRead read error [Error: EAGAIN: resource temporarily unavailable, read] {
  errno: -11,
  code: 'EAGAIN',
  syscall: 'read'
}
2020-04-07T09:41:15.129Z serialport/bindings/unixRead waiting for readable because of code: EAGAIN
2020-04-07T09:41:15.129Z serialport/bindings/poller Polling for "readable"
2020-04-07T09:41:41.106Z serialport/bindings/poller received "readable"
2020-04-07T09:41:41.106Z serialport/bindings/unixRead Starting read
2020-04-07T09:41:41.106Z serialport/bindings/unixRead Finished read 8 bytes
2020-04-07T09:41:41.106Z serialport/stream binding.read finished { bytesRead: 8 }
```

Is that normal?

CodeFinder2, 2020-04-07:
Strange...seems that its unable or at least having problems (`EAGAIN`) to communicate with the hardware (CC1352) but I guess Koenkk knows more about this :thinking:

Koenkk, 2020-04-07:
@timdonovanuk that logging is normal, I would recommend `DEBUG=*` but rather `DEBUG=zigbee-herdsman:*`.

Regarding the sniff, it looks like it cannot be decrypted which means that the wireshark is missing the correct network key.

closed 2020-06-13 by stale[bot]