Koenkk/zigbee2mqtt #2959: Hi Guys,

nickman80, 2020-02-17:
Hi Guys,
sorry to bother you, but i have been stuck for couple days with zigbee herdsman, would appreciate for any help, this is my log [https://pastebin.com/i30b2kYH](url)
thank you

_Originally posted by @nickman80 in https://github.com/Koenkk/zigbee2mqtt/issues/2891#issuecomment-586819902_

Koenkk, 2020-02-17:
`Error: Error Resource temporarily unavailable Cannot lock port` means that other software on the system is already occupying the port, you need to find out which.

nickman80, 2020-02-17:
Thank you for fast reply and your time, i have already checked ports on raspberry and AM0 is for zigbee and other one AM1 is for rflink, is there other way how to check them? Even when i had rflink port disconnected did not work.

nickman80, 2020-02-18:
ok so, i have figured out, the same port ACM0 had been used in NodeRed, 
processes list using serial ports :
ls -l /proc/[0-9]*/fd/* |grep /dev/ACM0       ---replace ACM0 with actual port that you searching 
then you will get /proc/14683....your number will be different and then you have to run in terminal : ps aux     and from list you can find your proc number (PID) and see where is it linked to.

closed 2020-02-18 by nickman80