Koenkk/zigbee2mqtt #2961: Auto detection of serial port not working

bochelork, 2020-02-17:
I've multiple serial devices on my system, ports are not always the same, depending on boot order, so I would like to auto detect the coordinator (port: null).  But it's not working:

zigbee2mqtt:error 2020-02-17 21:22:32: Error: No path provided and failed to auto detect path

When adding 'console.log(devices);' after 'let devices = yield serialport_1.default.list();' in serialPortUtils.js I get the following results for my ttyACM devices:

  { manufacturer: undefined,
    serialNumber: undefined,
    pnpId: undefined,
    locationId: undefined,
    vendorId: undefined,
    productId: undefined,
    path: '/dev/ttyACM0' },
  { manufacturer: undefined,
    serialNumber: undefined,
    pnpId: undefined,
    locationId: undefined,
    vendorId: undefined,
    productId: undefined,
    path: '/dev/ttyACM1' },

Coordinator firmware version: '{"type":"zStack3x0","meta":{"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20191106}}'
Z-Stack 3 on CC2652R

Zigbee2mqtt version 1.10.0-dev (commit #f0903cd)

System:
Linux NAS2 4.4.59+ #24922 SMP Wed Oct 30 18:44:43 CST 2019 aarch64 GNU/Linux synology_rtd1296_ds418j

Koenkk, 2020-02-20:
Are you using docker?

bochelork, 2020-02-20:
No docker... No Intel...

Koenkk, 2020-02-20:
Is `eudev` or `udev` installed on this system? This is required in order for the feature to work.

closed 2020-04-30 by stale[bot]