Koenkk/zigbee2mqtt #2963: CC2531 issue (probably), no devices turn on/off

MonsieurSander, 2020-02-17:
Hi! 

I was sitting on my couch, not doing anything with hassio, when I noticed that the cc2351 produced a continuous green light instead of a blinking one as is usual. Curiously, I tried to turn on of my light, and it didn't work. I tried all other lights and the same issue was there. None of my zigbee devices could be actuated. Tried to power down the raspberry, replug the USB stick, restart both zigbee2mqtt and mosquito but none seems to work. Running zigbee2mqtt 1.10.0 and the stick is on 08062019. Bought the stick with the firmware installed, reckon I need to get a cc-debugger and see if that helps.

https://pastebin.com/FKywwjLK the log file

Koenkk, 2020-02-20:
Probably your CC2531 crashed, I would recommend flashing the source routing firmware to improve stability. https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing

MonsieurSander, 2020-02-20:
Thanks for your reply, was thinking along the same lines, cc debugger will arrive next week. Will post the results in here for future reference

MonsieurSander, 2020-03-01:
Reflashing the hub did the trick!

Koenkk, 2020-03-01:
Good, I assume this can be closed now

closed 2020-03-01 by Koenkk