Koenkk/zigbee2mqtt #2964: Philips hue motion sensor creating trouble in my zigbee network

pepp86, 2020-02-18:
Hi,

I have a pretty stable zigbee network wit CC2652R Board and zigbee 3x0 firmware. I'm running more than 50 devices (lot of IKEA and Osram routers).

At any time when I join a philips hue motion sensor, my networks get unstable - I'm receiving a lot of "Mac no Ack" messages immediatly after sending the command.

Turning logging to debug shows various situations. Sometimes the motion sensor is just quiet for a long time - other times I receive "Device '0x0017880106f731c9' announced itself" a lot without sending any movement.

Does anyone see the same issues at his network? Or any idea how to debug more in depth?

Thank you.

Koenkk, 2020-02-20:
Strange! Haven't seen this before.

To debug this issue you could sniff the traffic on your network: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

Fabiancrg, 2020-04-16:
Hi,

I was about to create a new issue for this.
So I have three Hue outdoor motion sensors, they worked fine with the CC2531 and ZHA 1.2 with source routing.
A few weeks ago, I upgraded to a CC2652R with Z-stack 3.x firmware and since I have to pair these outdoor sensors regularly as they drop from the network.
What I can see in the log are the folloiwng messages:

- There are 'Device announce' from time to time and normal message published
- At one point, the device will leave the network
- A 'Device join' will follow
- and righjt after a 'starting interview'
- A 'Device Interview' fails after a couple of minutes
- After that no other messages are reported

log --> https://pastebin.com/jYSHTnzr

This is happening randomly on each of these devices, one after the other, it could be after two days, a week, I don't see any logic.

I do have about 50 other various devices in my network and these Hue motion sensors are the only ones droping from the network.

@Koenkk I understand that with this normal log it won't be easy to find the root cause but what will you need to debug this ? I can run in debug mode but as it is done in the console and as the issue is sometimes taking a week to happen, not sure this will work.

Thank you

Koenkk, 2020-04-16:
@Fabiancrg this is a different issue, please check https://github.com/Koenkk/zigbee2mqtt/issues/2693#issuecomment-612957791

closed 2020-06-23 by stale[bot]