Koenkk/zigbee2mqtt #2965: RTCGQ11LM Xiaomi regular update

MaxBec, 2020-02-18:
Hey guys,

i just wanted to know how to set the illuminance and battery status update interval for the xiaomi RTCGQ11LM motion sensor. 60minutes (i think this is the default) is too long for me.

Thank you.
Kind Regards, Max

Koenkk, 2020-02-18:
This is not possible (limitation of the sensor itself). However you can modify the occupancy internal by applying a hardware hack: https://community.smartthings.com/t/making-xiaomi-motion-sensor-a-super-motion-sensor/139806

closed 2020-02-18 by Koenkk

MaxBec, 2020-02-18:
I soldered it now. The thing with the motion is working. But if there is no motion i want to have regular illuminance updates. He is sending them only if motion changes. Can i change the settings to send regular illuminance and battery updates?

Koenkk, 2020-02-18:
> Can i change the settings to send regular illuminance and battery updates?

No (limitation of the sensor itself)