Koenkk/zigbee2mqtt #2967: WSDCGQ11LM pressure value rounded to 0 decimals when drawn from database?

genezig, 2020-02-18:
# Bug Report
## What happened
Messages received from Aqara WSDCGQ11LM sensor vary in precision ignoring pressure_precision parameter. With pressure_precision: 1 and pressure of i.e. 1013,25 hPa value is sometimes 1013 (without any decimals), sometimes 1013,3. When no decimal is reported, the reported value is always rounded to the lower value, i.e. 1013,9 is reported as 1013.

I have 4 sensors, same behaviour observed on all of them.

Look at the log excerpt seeing the jumping values of pressure:
`info  2020-02-18 13:58:41: MQTT publish: topic 'zigbee2mqtt/temp3', payload '{"battery":57,"voltage":2925,"temperature":7.07,"humidity":49.24,"pressure":919,"linkquality":39}'
info  2020-02-18 14:23:13: MQTT publish: topic 'zigbee2mqtt/temp3', payload '{"battery":57,"voltage":2925,"temperature":7.46,"humidity":49.16,"pressure":919.7,"linkquality":52}'
info  2020-02-18 14:23:23: MQTT publish: topic 'zigbee2mqtt/temp3', payload '{"battery":57,"voltage":2925,"temperature":7.77,"humidity":49.16,"pressure":919.7,"linkquality":60}'
info  2020-02-18 14:23:23: MQTT publish: topic 'zigbee2mqtt/temp3', payload '{"battery":57,"voltage":2925,"temperature":7.77,"humidity":50.5,"pressure":919.7,"linkquality":60}'
info  2020-02-18 14:23:23: MQTT publish: topic 'zigbee2mqtt/temp3', payload '{"battery":57,"voltage":2925,"temperature":7.77,"humidity":50.5,"pressure":919,"linkquality":60}'
info  2020-02-18 14:26:04: MQTT publish: topic 'zigbee2mqtt/temp3', payload '{"battery":57,"voltage":2925,"temperature":7.45,"humidity":50.33,"pressure":919.7,"linkquality":52}'
info  2020-02-18 14:29:46: MQTT publish: topic 'zigbee2mqtt/temp3', payload '{"battery":57,"voltage":2925,"temperature":7.27,"humidity":50.33,"pressure":919.7,"linkquality":47}'
info  2020-02-18 14:29:46: MQTT publish: topic 'zigbee2mqtt/temp3', payload '{"battery":57,"voltage":2925,"temperature":7.27,"humidity":47.71,"pressure":919.7,"linkquality":47}'
info  2020-02-18 14:29:46: MQTT publish: topic 'zigbee2mqtt/temp3', payload '{"battery":57,"voltage":2925,"temperature":7.27,"humidity":47.71,"pressure":919,"linkquality":47}'
info  2020-02-18 14:48:03: MQTT publish: topic 'zigbee2mqtt/temp3', payload '{"battery":57,"voltage":2925,"temperature":7.83,"humidity":47.71,"pressure":919,"linkquality":44}'
`
## What did you expect to happen
When pressure_precision is set to 1 the pressure value in the MQTT message should always have the configured precision, i.e. for pressure of 1013,25 hPa and pressure_precision: 1 value should always be 1013,3. Only for i.e. 1013,0 I would expect a value of 1013 without decimal.
## How to reproduce it (minimal and precise)
Pair WSDCGQ11LM, configure pressure_precision: 1 in configuration.yaml for the paired device and wait for messages coming in. Some will report the pressure with one decimal, some without. See log in first section of this report.
## Debug Info
zigbee2mqtt version: 1.10.0
CC253X firmware version: 20190608

I believe this is a problem with database storing. Looking at database.db all pressure values are stored without decimals. Temperatures are stored with all decimals as reported by the device (i.e. 2304 for 23.04 degrees), pressure only i.e. as 1013 without decimal. When zigbee2mqtt derives values from the database it reports them without decimal. BTW: Shouldn't reading values from the database be inhibited with retain: false (which is set for all devices in my case)?

Koenkk, 2020-02-20:
Could you provide the debug logging of the messages received? Note that when the number is 919.0 you will always get 919, this is a limitation of javascript (it only has one number type, no floats, double, int etc.).

genezig, 2020-02-20:
Ok, 919 equals 919,0 - just what I am expecting.

Debug output resulting in rounded pressure value:
`debug 2020-02-20 22:26:10: Received Zigbee message from 'temp2', type 'attributeReport', cluster 'msTemperatureMeasurement', data '{"measuredValue":2340}' from endpoint 1 with groupID 0
debug 2020-02-20 22:26:10: Received Zigbee message from 'temp2', type 'attributeReport', cluster 'msRelativeHumidity', data '{"measuredValue":4589}' from endpoint 1 with groupID 0
debug 2020-02-20 22:26:10: Received Zigbee message from 'temp2', type 'attributeReport', cluster 'msPressureMeasurement', data '{"16":9154,"20":-1,"measuredValue":915}' from endpoint 1 with groupID 0
info  2020-02-20 22:26:11: MQTT publish: topic 'zigbee2mqtt/temp2', payload '{"battery":91,"voltage":2985,"temperature":23.4,"humidity":45.89,"pressure":915,"linkquality":52}'`

Debug output resulting in pressure value with one decimal:
`debug 2020-02-20 22:33:41: Received Zigbee message from 'temp2', type 'attributeReport', cluster 'genBasic', data '{"65281":{"1":2985,"4":5032,"5":9,"6":[0,1],"10":0,"100":2345,"101":4608,"102":91552}}' from endpoint 1 with groupID 0
info  2020-02-20 22:33:42: MQTT publish: topic 'zigbee2mqtt/temp2', payload '{"battery":91,"voltage":2985,"temperature":23.45,"humidity":46.08,"pressure":915.5,"linkquality":42}'`

Koenkk, 2020-02-22:
I'm not able to reproduce this. Could you post me your `configuration.yaml` file (remove password and url's from it)?

genezig, 2020-02-22:
My configuration.yaml:

_homeassistant: false
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://localhost'
advanced:
  channel: 25
  log_level: info
  log_directory: /opt/zigbee2mqtt/data
devices:
  '0x000d6ffffe9dxxxx':
    friendly_name: tradfri-repeater
    retain: false
  '0x00158d000485xxxx':
    friendly_name: temp1
    pressure_precision: 1
    debounce: 1
    retain: false
  '0x00158d000484xxxx':
    friendly_name: temp2
    pressure_precision: 1
    debounce: 1
    retain: false
  '0x00158d000487xxxx':
    friendly_name: temp3
    pressure_precision: 1
    debounce: 1
    retain: true_

Changing the debounce parameter does not change the behaviour. Same for the retain parameter. Just came across the repeater when going through the configuration.yaml, tried with another one (different type) and without it. First it looked like there is a change, but in the end the issue persists.

Thank you for looking into this.

Koenkk, 2020-02-23:
In the first log of https://github.com/Koenkk/zigbee2mqtt/issues/2967#issuecomment-589332496 the sensors sends a pressure of exactly `"measuredValue":915` because of this zigbee2mqtt drops the precision like I mentioned in https://github.com/Koenkk/zigbee2mqtt/issues/2967#issuecomment-589257154

Actually all pressure measurements that come from the `msPressureMeasurement` cluster dont have precision (this is what has been defined in the zigbee cluster library). Only the ones that come from the `genBasic` cluster will have it.

genezig, 2020-02-23:
Is that the expected behaviour? How can I change it?

Koenkk, 2020-02-23:
If you mean that `915` is always published as `915` (and not `915.0`) this is indeed expected behavior due to the technical reasons I mentioned here: https://github.com/Koenkk/zigbee2mqtt/issues/2967#issuecomment-589257154

genezig, 2020-02-23:
Well, no. The reason why I opened this report is that with a pressure of 915.5 i sometimes get 915 and sometimes 915.5 as value reported by zigbee2mqtt. Looking at the debug log you state that the genBasic cluster has the desired precision, the msPressureMeasurement has not. Behaviour that I would expect is that the output message has always the same precision no matter which cluster is used by the device, configured with the pressure_precision parameter in configuration.yaml.

So what my question is: Can I a) filter messages in order to only receive those coming from the genBasic cluster or b) change the behaviour of the msPressureMeasurement cluster to respect the pressure_precision parameter? The goal would be to always get mqtt messages with pressure values respecting the pressure_precision parameter, i.e. 915.5 is always reported as "915.5" and not as "915".

Koenkk, 2020-02-23:
A. You could achieve this by removing the `generic_pressure` converter from the `fromZigbee` in the WSDCGQ11LM in `devices.js` (see how to supported new devices in the zigbee2mqtt docs on how to get there). However I dont recommend this as you would mis a lot of pressure change events. The genBasic cluster message is only send every hour. But still when the genBasic would send a whole value you will get a number without precision (although chances for this to happen are low).
B. No, this is just the zigbee specification. The whole precision thing only rounds the received values, it doesnt make the devices report the pressure with more precision.

genezig, 2020-02-23:
Thank you for that quick answer. What I am wondering about: Looking at the debug log I can see that the device reports the following data:
`cluster 'msPressureMeasurement', data '{"16":9154,"20":-1,"measuredValue":915}'`
So in the number "9154" the required precision obviously is contained, pressure value is "915.4". That leads me to the conclusion that it should be possible to derive my desired and configured precision (one decimal) from that value by modifying zigbee2mqtt because the device reports it even in the msPressureMeasurement cluster.

genezig, 2020-02-23:
Ok, I think I now have a solution. While writing my previous post I stumbled about "16" vs. "measuredValue". "20" might be the report of the own precision of the device which I would like to ignore (if that is the right assumption at all).

I modified the "lumi.weather" block in devices.js as follows (changing fz.generic_pressure to fz.xiaomi_pressure):
`        fromZigbee: [
            fz.xiaomi_battery_3v, fz.xiaomi_temperature, fz.humidity, fz.xiaomi_pressure,
            fz.WSDCGQ01LM_WSDCGQ11LM_interval,
        ],
`

and adding the following block to fromZigbee.js:

`    xiaomi_pressure: {
        cluster: 'msPressureMeasurement',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            const pressure = parseFloat(msg.data['16']) / 10.0;
            return {pressure: calibrateAndPrecisionRoundOptions(pressure, options, 'pressure')};
        },
    },
`

This is working for me. I don't know if it is a clean approach, but if my assumption is right and "20":-1 is the measurement precision reported by the device this would mean that I just ignore this.

BTW: Sorry for the missing line feeds in the code, I tried to cope with that, no success.

Koenkk, 2020-02-23:
@genezig well found, I didn't notice the `16` property yet, it indeed seems to contain the pressure value with more precision. This value will now be used instead of the `measuredValue`, change is available in the latest dev branch.

Thanks! Assuming this can be closed now.

closed 2020-02-23 by Koenkk

smartynov, 2021-11-08:
I noticed that PR Koenkk/zigbee-herdsman-converters#3285 cancelled the code that was introduced in this Issue/PR. However it seems that the code was buggy anyway (see comments in that PR). At the same time I can see that current version (with patch applied) returns high precision data (not rounded).