Koenkk/zigbee2mqtt #2969: Samsung button IM6001-BTP02 not reporting temperature

sv87411, 2020-02-18:
# Bug Report
## What happened
Successfully paired Samsung model number IM6001-BTP02 button. All three capabilities appear in Home Assistant (screenshot attached):

<device name>button_click
<device name>button_linkquality
<device name>button_temperature

however, temperature is reporting as an empty string ""

This button was previously paired to Samsung hub and reporting temperature fine.

I noticed that the model number for the supported version of the Samsung button is "IM6001-BTP01". My button is clearly labelled as "M/N: IM6001-BTP02"

I have lights, outlets and Xiaomi temperature sensors paired fine with zigbee2mqtt and all are working fine so I am assuming this is a device/messaging issue where the BTP02 button is not reporting in the same way as the BTP01. Happy to provide whatever logs are needed to support troubleshooting.

## What did you expect to happen
I expected temperature to be reported as a centigrade value.

## How to reproduce it (minimal and precise)
Pair Samsung IM6001-BTP02 button to zigbee2mqtt, wait for it to appear in home assistant mqtt integration, check to see if temperature value is reporting.

## Debug Info
zigbee2mqtt version: 1.10.1
CC253X firmware version: Z-Stack_Home_1.2

[image]

sv87411, 2020-02-18:
I should add that button clicks are working correctly too with single, double and hold all working.

Shadowpheonix, 2020-02-19:
I am experiencing this same behavior with an IM6001-BTP01 (meaning this is not due to a difference between the BTP01 and BTP02 units).

Koenkk, 2020-02-20:
I think the entry in `devices.js` is not correct for this device.

Can you try updating it to

```js
{
    zigbeeModel: ['button'],
    model: 'IM6001-BTP01',
    vendor: 'SmartThings',
    description: 'Button',
    supports: 'single click, double click, hold and temperature',
    fromZigbee: [fz.st_button_state, fz.generic_battery, fz.temperature, fz.ignore_iaszone_attreport],
    toZigbee: [],
    meta: {configureKey: 1},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        await bind(endpoint, coordinatorEndpoint, ['msTemperatureMeasurement', 'genPowerCfg']);
        await configureReporting.temperature(endpoint);
        await configureReporting.batteryVoltage(endpoint);
    },
},
```

Afterwards remove the device through a `force_remove` and repair the device. https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigremove

How to get to `devices.js`: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

sv87411, 2020-02-21:
Thank you! I've updated the entry entry in devices.js as above and my buttons are now reporting temperature again.

[image]

Follow up question. I'm only just starting to use zigbee2mqtt and a local coordinator as I really want to move away from my Samsung hub.

This situation with the Samsung button suggests to me that maybe the code for the button has  changed and this caused them to stop reporting temperature? 

I've read through the adding devices section of your website and it suggests that anyone could make a change to the devices.js and commit those changes, including breaking existing devices. Do you check all committed changes to the file to ensure they don't break something that is already working?

Apologies if I've understood the mechanism for updating the devices.js file.

Thanks again for the quick response and fix!

Koenkk, 2020-02-22:
> However they are still not reporting battery. Samsung buttons integrated to my Home Assistant via Smartthings are reporting battery state so I know they can report this.

The battery sensor was not discovered for this device yet by zigbee2mqtt, added this in the latest dev branch.

> This situation with the Samsung button suggests to me that maybe the code for the button has changed and this caused them to stop reporting temperature?

Probably it has never been there in the first place, so the original PR to add this was not complete.

> I've read through the adding devices section of your website and it suggests that anyone could make a change to the devices.js and commit those changes, including breaking existing devices. Do you check all committed changes to the file to ensure they don't break something that is already working?

All changes are reviewed, if I seem something suspicious in this PR I will start asking question. But note that I don't have every device and it's impossible to verify each change for 100%.

sv87411, 2020-02-25:
> Please update to the latest zigbee2mqtt dev branch, the device should fully work now. https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html

Thanks for all the responses. Things are mostly working now. Thanks for the insight into the commit process.

I've just yesterday received a Ti CC26X2R1 board (which so far I'm loving). Therefore, I've moved to the dev Z2M branch and re-paired the few devices I've been experimenting with on zigbee2mqtt instead of my Samsung hub to the CC26X2 as my main Z2M coordinator. 

These are mostly my study devices and this includes the Samsung button.

The Samsung button paired OK and is now reporting click, link quality, temperature and battery to Home Assistant. But, the battery is reporting as 180% - see screenshot and Z2M log extract below.  While I'm impressed by this and I wish all my batteries had 180% charge, obviously I don't believe this is true.

I've re-paired it back to the Samsung hub to check the battery and as I suspected its reporting 90%. So the Z2M converter is reporting double the actual % value. Re-paired to Z2M and still reporting 180%.

So it's nearly reporting perfectly, just doubling the battery level.

Keep up all the amazing work, I'm learning so much and loving the process of moving from my Samsung hub to Z2M.

[image]

Log extract:
`Feb 25 15:24:16 dtop npm[16392]: zigbee2mqtt:info  2020-02-25 15:24:16: MQTT publish: topic 'z2m/m-study-button', payload '{"linkquality":219,"temperature":19.25,"battery":180,"click":"single"}'`

Koenkk, 2020-02-25:
Fixed in the latest dev branch.

sv87411, 2020-02-27:
Hi, sorry only just got around to testing.

Just updated to latest dev branch and still my button is reporting 161 as battery 

`Feb 27 23:01:13 dtop npm[30650]: zigbee2mqtt:info  2020-02-27 23:01:13: MQTT publish: topic 'z2m/m-study-button', payload '{"linkquality":165,"temperature":19.94,"battery":161,"click":"single"}'`

[image]

Thanks

sv87411, 2020-02-28:
Apologies... I just tried to pair a new Samsung button and during that process it didn't report it's battery initially. I recall now that the Samsung buttons take quite a while to report battery state.

Could it be that what I am seeing with the study button is a cached value that will take a while to update to the correct value when it next reports?

Koenkk, 2020-02-28:
Yes it takes some time (should be max 24 hours)

sv87411, 2020-02-28:
Thanks. I've now migrated all of my Samsung buttons from SmartThings to Z2M and 3 of them are reporting battery. The other 4 I only migrated today so I'll give them 24 hours. 

Thanks for all your help and I think I can close this issue now.

Stuart

closed 2020-02-28 by sv87411