Koenkk/zigbee2mqtt #2970: Allow a read-only devices.yml

rhuss, 2020-02-19:
## Feature request

Actually `devices.yml` (as referenced in the configuratin's `devices:` field) can't be used in a read-only fashion, leading to errors when a new device is detected:

```
mqtt zigbee2mqtt-0 zigbee2mqtt (node:17) UnhandledPromiseRejectionWarning: Error: EROFS: read-only file system, open '/app/data/devices.yaml'
mqtt zigbee2mqtt-0 zigbee2mqtt     at Object.openSync (fs.js:440:3)
mqtt zigbee2mqtt-0 zigbee2mqtt     at Object.writeFileSync (fs.js:1265:35)
mqtt zigbee2mqtt-0 zigbee2mqtt     at Object.writeIfChanged (/app/lib/util/yaml.js:32:12)
mqtt zigbee2mqtt-0 zigbee2mqtt     at write (/app/lib/util/settings.js:274:14)
mqtt zigbee2mqtt-0 zigbee2mqtt     at Object.addDevice (/app/lib/util/settings.js:472:5)
mqtt zigbee2mqtt-0 zigbee2mqtt     at Controller.onZigbeeEvent (/app/lib/controller.js:174:40)
mqtt zigbee2mqtt-0 zigbee2mqtt     at Zigbee.emit (events.js:223:5)
mqtt zigbee2mqtt-0 zigbee2mqtt     at Controller.<anonymous> (/app/lib/zigbee.js:55:57)
mqtt zigbee2mqtt-0 zigbee2mqtt     at Controller.emit (events.js:223:5)
mqtt zigbee2mqtt-0 zigbee2mqtt     at Controller.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/controller.js:329:22)
```

A read-only configuration makes sense in a Kubernetes context where this configuration can be served by a K8s `ConfigMap` or `Secret` which is mounted as a read-only file. So, a clear separation between files that can be user-provided and which are written by zigbee2mqtt during runtime would be very helpful, not only in this situation but also to separate concerns and don't have two parties writing into the same files (which could lead to conflicts in other context, too).

A possible syggestion would be to use 2 files, one user-provided (like "devices.yml") and one which is written by the system (like "devices-run.yml"), and that could be merged, with the data in "devices.yml" taking precedence (or it could also two different schemas, and then distinguished within the code)

rhuss, 2020-04-24:
Is there anything I could to help to get this implemented ? I have a workaround, however that requires multiple restarts though when adding a new device.

rhuss, 2020-06-23:
Any chance to bring this forward ?

rhuss, 2020-08-06:
Stale bot is merciless :)

rhuss, 2020-08-06:
(really would love this feature but understand that it's of low priority. And if I would have some more spare time, then its would be definitely one of the stories I would tackle first)

Koenkk, 2020-08-07:
Please help me understand the following:

> A possible syggestion would be to use 2 files, one user-provided (like "devices.yml") and one which is written by the system (like "devices-run.yml"), and that could be merged, with the data in [...]

Since it's possible to make files writable in this context (`devices-run.yml` like you proposed), why can this be done with `devices.yml`?

rhuss, 2020-08-13:
My use case is quite simple: In need to be able to operate all user provided configuration in a read-only fashion so that they can be served in Kubernetes by `ConfigMaps` which is a read-only concept for delivering key-value maps and that can be mounted as read-only files. For anything written by zigbee2mqtt a separate file would be needed.

The issue described here is when a new device is auto-detected and its config (ids etc) are written to a configuration file. My workflow is to make the auto-detection work and then copy over the auto-detected information into the `ConfigMap` so that i can add e.g. a proper name to the device.

I'm not sure, but if the auto-detected information is written into the devices.yml and I use the main configuration file for the sanitized device info, that would work, too. But I was not aware that you can mix device informations in both, `configuration.xml` and a `device.yml` (thought it was an exclusive choice).

Let me try if I can get this to work, because then this issue is indeed obsolete.

closed 2020-09-21 by github-actions[bot]

rhuss, 2020-11-27:
@Koenkk I'm trying now the approach that you suggest, e.g. to do both, using a `devices.yaml` file and inline configuration, however I'm not sure how this should work, as the key `devices:` can be only used once: Either for referring to an external file or for holding a list of device declarations.

Do you have an example how this can be achieved ?

Koenkk, 2020-11-27:
I probably misread your question at the time, but this was not possible yet. I've added a new feature to allow this, docs: https://github.com/Koenkk/zigbee2mqtt.io/blob/develop/docs/information/configuration.md#specifying-devices-and-groups-in-a-separate-file

Spoiler
```yaml
devices:
  - devices1.yaml   # <- only this file will be written
  - devices2.yaml
```

MoYapro, 2020-11-29:
Just tried this new configuration option from the dev branch. Works like a charm <3