Koenkk/zigbee2mqtt #2971: Has someone tried these wired all switches?

JuanGrados, 2020-02-19:
I have looked for them in the "supported devices" list and I haven't found them.
I also suspect that many of these wall switches might be very similar to some of the officially supported ones.

I have bought the Xiaomi without neutral wire and works fine, unfortunately I didn't realise that the "body" is not EU standard and I cannot fit it in the hole of the wall :(
That is why I am looking for alternatives.

[No Neutral Lerlink](https://www.aliexpress.com/item/4000423219189.html?trace=wwwdetail2mobilesitedetail&gps-id=platformRecommendH5&scm=1007.18499.134609.0&scm_id=1007.18499.134609.0&scm-url=1007.18499.134609.0&pvid=d38be7b9-1cf0-4585-a4fd-924645a9f443&_t=gps-id:platformRecommendH5,scm-url:1007.18499.134609.0,pvid:d38be7b9-1cf0-4585-a4fd-924645a9f443&spm=a2g0n.detail-amp.moretolove.4000423219189&aff_trace_key=1ad61c4409034fafb39a1e0256fc2f51-1582064530747-08500-UneMJZVf&aff_platform=msite&m_page_id=4240amp-NJOrYiYiD82joMvqgLN5ZA1582098090071&browser_id=22837ce7af5c438ba259c0feb043919c&is_c=N&gclid=Cj0KCQiAs67yBRC7ARIsAF49CdVgTC6e3FjH-tCfNflVvvvZ3jl46pGRaZbI7mG2efEJTis-Upt6WkQaAp8AEALw_wcB&gclsrc=aw.ds)

[Lellki with neutral wire](https://www.aliexpress.com/item/4000497939899.html?spm=a2g0o.productlist.0.0.1f6f71a57Gqn4T&algo_pvid=8b58b777-4267-4ef5-86dd-ce04819fcabd&algo_expid=8b58b777-4267-4ef5-86dd-ce04819fcabd-1&btsid=0b0a0ac215821432916244213e89c0&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_)

Thank you guys :)

sjorge, 2020-02-19:
Ooh they do look nice (moving soon so not sure if I would need with or without neutral wire, else I'd order one to try).

I wonder if they would offer decouple mode like the Aqara ones because I;d need that
Have the button control a zigbee group and the bulbs always be powered.

Koenkk, 2020-02-19:
I think these are already supported: https://www.zigbee2mqtt.io/devices/GDKES-03TZXD.html (this is just another rebrand).

JuanGrados, 2020-02-19:
@Koenkk that was what I was suspecting, is it “save” to asume that switches using the Tuya chipset are compatible?

Koenkk, 2020-02-20:
@JuanGrados if they have the same appearance, probably yes.

JuanGrados, 2020-02-21:
I found another rebrand that also looks nice, with or without neutral, 1/2/3 gang:

JuanGrados, 2020-02-21:
[Lonsonho Zigbee](https://www.aliexpress.com/item/4000298926256.html?spm=a2g0o.detail.1000060.1.3bd2589frho1YL&gps-id=pcDetailBottomMoreThisSeller&scm=1007.13339.146401.0&scm_id=1007.13339.146401.0&scm-url=1007.13339.146401.0&pvid=1a372d62-aab9-4a25-aeb6-66e2b2c51612)

JuanGrados, 2020-02-21:
I have bought the Lonsonho, with and without neutral. 
I will let you know if they work fine or not as soon as I get them (it might take a while)

sjorge, 2020-02-21:
If they work, I wonder if you can decouple them like with the aqara ones

JuanGrados, 2020-02-21:
I hope so! otherwise they will be useless to me, I do not want to collect random hubs with forced connections to some servers. I have enough with the Roborock (waiting for rooting of 3rd generation)

JuanGrados, 2020-03-15:
Ok, after 23 days, I got the switches without neutral (the ones with neural still to be shipped).

fr00kt, 2020-05-01:
I bought the same switches as yours, but there is 1 problem when using them when the state is off, the LED does not light, and when you turn it on, it lights up in red. When using Tuya hub they shine red and green respectively

JuanGrados, 2020-05-01:
Hi @fr00kt for me show the same behavior as yours, I didn’t know they suppose to show green/red depending on the state.

closed 2020-07-10 by stale[bot]