Koenkk/zigbee2mqtt #2972: EcoSmart A9A19A60WESDZ02 device not supported (narrator: it is)

roygbiv856, 2020-02-20:
## What happened
When trying to pair EcoSmart A9A19A60WESDZ02 with new CC26X2R1 coordinator, I get message "Device not supported. Visit this link to support new devices"

## What did you expect to happen
I expected the bulb to pair successfully considering that it is indeed already is a supported device.

## How to reproduce it (minimal and precise)
Place coordinator into pairing mode, turn bulb off/on 5 times in 6 seconds

## Debug Info
zigbee2mqtt version: 1.6 (hassio add on)
CC253X firmware version: newest available

Koenkk, 2020-02-20:
You are running an old zigbee2mqtt version, try updating to zigbee2mqtt 1.10.

closed 2020-04-30 by stale[bot]

mkaatman, 2022-04-01:
zigbee2mqtt version: 1.25
zStack3x0 20211217

Same issue.