Koenkk/zigbee2mqtt #2974: CC2530 coordinator via ESP8266 and Docker

githubwesseln, 2020-02-20:
The CC2530 coordinator is connected via an ESP8266, actually installation is running well without socat:

configuration.yaml:
serial:
  port: 'tcp://192.168.178.152:1775' 

Now I want to use Zigbee2MQTT with Docker koenkk/zigbee2mqtt, but it doesn't work.
What is the correct entry for ‘device:’ in the ‘docker-compose.yml’ file ?
Is a modification in the ‘configuration.yaml’ for Zigbee2MQTT necessary ?

The test with a USB CC2531 coordinator and Docker is working.

configuration.yaml:
serial:
  port: /dev/ttyACM0

docker-compose.yml:
devices:
   - /dev/ttyACM0:/dev/ttyACM0

## Debug Info
zigbee2mqtt version: 1.10.1
CC253X firmware version: Z-Stack_Home_1.2

markuspeter, 2020-04-06:
Hi,
Were you ever able to make this work ? Is there a particular problem running Zigbee2MQTT from docker with a remote coordinator (via esp8266)?

kabakakao, 2020-04-17:
I am running docker with nearly the same constellation. But is only working with Version >= 1.8.0
If I use a version above, I ran into OOM. But I don't know why
You don't have to set device. Delete the whole section. Can you check it on your site?

roesl, 2020-04-25:
I want to install the cc2530 + 2592 coordinator on Docker for Synology, but I cant install ser2net or another VSP Software on Synology to connect the device.
Is there a solution ?

closed 2020-07-03 by stale[bot]