Koenkk/zigbee2mqtt #2975: Heiman M420

geeano, 2020-02-20:
# Bug Report
Heiman Smart Siren failed to pair:
Failed to interview '0xccccccfffed86303', device has not successfully been paired

## What happened
After pairing process is initiated, process will not finish succesfull 

## What did you expect to happen
Pairing process succeeded

## How to reproduce it (minimal and precise)
press network button long (inititate pairing) to add device

## Debug Info
zigbee2mqtt version: zigbee2mqtt@1.9.0 
CC253X firmware version: ? (where to find)

Logfile: https://hastebin.com/ibubikefec.bash

geeano, 2020-02-20:
meanwhile device added in device.js (
New Device - Heiman - Model ID: WarningDevice-EF-3.0
#2877 )

latest log:
error 2020-02-20 14:43:23: Failed to interview '0xccccccfffed86303', device has not successfully been paired
info  2020-02-20 14:43:23: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_nam
info  2020-02-20 14:43:23: Configuring '0xccccccfffed86303'
info  2020-02-20 14:43:24: Successfully configured '0xccccccfffed86303'
debug 2020-02-20 14:43:24: Received Zigbee message from '0xccccccfffed86303', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"zonoint 1 with groupID 0

Koenkk, 2020-02-20:
Please try with zigbee2mqtt 1.10 (contains some improvements regarding interviewing).

geeano, 2020-02-20:
Hi Coen, thx for your quick response. 
I upgraded meanwhile to latest version 1.10, but without any result.
 The device was not added to the latest latest version of device.js 

I added some logging:

zigbee2mqtt:debug 2020-02-20 22:54:46: Received Zigbee message from '0xccccccfffed86303', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"zonestatus":16,"extendedstatus":0}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-02-20 22:54:46: No converter available for 'HS2WD-E' with cluster 'ssIasZone' and type 'commandStatusChangeNotification' and data '{"zonestatus":16,"extendedstatus":0}'
  zigbee-herdsman:controller:endpoint DefaultResponse 0xccccccfffed86303/1 1280(0, {"timeout":10000,"defaultResponseTimeout":15000,"manufacturerCode":null,"disableDefaultResponse":true}) +15s

geeano, 2020-02-20:
Furthermore I added a more detailed log in mode: DEBUG=zigbee-herdsman* npm start 
Please have a look
https://hastebin.com/jivagifecu.makefile

geeano, 2020-02-21:
With further pairing attempts is does not appear anymore in the logs, in contrary the the previous log.

Koenkk, 2020-02-23:
When I open the link now (https://hastebin.com/jivagifecu.makefile) it appear empty, could you perhaps post it on pastebin.com?

geeano, 2020-02-23:
Hi Koen, unfortunately I don't have a local copy. As I mentioned in my previous response, the device does not appear anymore in the console log during DEBUG=zigbee-herdsman* npm start

Koenkk, 2020-02-23:
- Can you try disconnecting the battery of the device for a few minutes and repair?
- If this doesn't help try reflashing the CC2531 and repair.

geeano, 2020-02-23:
I opened the device yesterday and disconnected the battery connector. After reconnecting the backup battery I tried to pair it again, without success. It did still not appear in the log. I prefer to avoid reflashing of the CC2531 upfront. Why do you suggest?

geeano, 2020-02-23:
Tomorrow I will disconnect it a second time and wait a couple of minutes. I will let you know.

geeano, 2020-02-24:
It does not make any difference. Device does not appear anymore in logging after press button to initiate pairing.

Koenkk, 2020-02-24:
Also check https://www.zigbee2mqtt.io/information/FAQ.html#why-does-my-device-not-or-fail-to-pair

geeano, 2020-02-24:
Hi Koen, although I had some reluctance to reflash the CC2531, I followed up your advice.
Now the device is paired.

Remaining is a message in my domoticz logging:
2020-02-24 20:01:47.507 (zigbee) Unhandled message from zigbee2mqtt: zigbee2mqtt/0xccccccfffed86303 {'battery': 100, 'linkquality': 60}

For me it is unclear if it is caused by the domoticz-zigbee2mqtt-plugin. Can you let know your expert view?

geeano, 2020-02-24:
The detailed Zigbee is available at https://pastebin.com/nnWi2W6Q

geeano, 2020-02-24:
Koen, from the log the message: No converter available for 'M420' with cluster 'ssIasZone' remains.
What is the cause of this message?

Koenkk, 2020-02-25:
The no converter available message is not a problem (and not relevant for this device so you can ignore it).

You can now trigger the alarm by sending to `zigbee2mqtt/0xccccccfffed86303/set` with payload `{"warning": {"mode": "emergency"}}`

> 2020-02-24 20:01:47.507 (zigbee) Unhandled message from zigbee2mqtt: zigbee2mqtt/0xccccccfffed86303 {'battery': 100, 'linkquality': 60}

Probably it still has to be added to the domitcz plugin, please create an issue here: https://github.com/stas-demydiuk/domoticz-zigbee2mqtt-plugin/issues

geeano, 2020-02-25:
Thanks again for your support !

What am I doing wrong ?:

pi@raspberrypi:~ $ mosquitto_pub -t zigbee2mqtt/0xccccccfffed86303/set -m {"warning": {"mode": "emergency"}}
Error: Unknown option '{mode:'

pi@raspberrypi:~ $ mosquitto_pub -t zigbee2mqtt/0xccccccfffed86303/set {"warning": {"mode": "emergency"}}
Error: Unknown option '{warning:'.

Koenkk, 2020-02-25:
quotes are missing: ` mosquitto_pub -t zigbee2mqtt/0xccccccfffed86303/set -m '{"warning": {"mode": "emergency"}}'`

geeano, 2020-02-25:
Thanks Koen for all your advises, it works, and deaf for a while :-)
I will create an issue : https://github.com/stas-demydiuk/domoticz-zigbee2mqtt-plugin/issues

geeano, 2020-02-25:
Is there any abnormality in the log (before requesting support for new device / for the avoidance of doubt):

My concern is the row " .. failed: 'Error: Command.." 

debug 2020-02-25 15:07:19: Received MQTT message on 'zigbee2mqtt/0xccccccfffed86303/set' with data '{"warning": {"mode": "emergency", "level": "medium", "strobe": "false", "duration": "5"}}'
debug 2020-02-25 15:07:19: Publishing 'set' 'warning' to '0xccccccfffed86303'
debug 2020-02-25 15:07:19: Received Zigbee message from '0xccccccfffed86303', type 'attributeReport', cluster 'ssIasWd', data '{"61440":3}' from endpoint 1 with groupID 0
debug 2020-02-25 15:07:19: No converter available for 'M420' with cluster 'ssIasWd' and type 'attributeReport' and data '{"61440":3}'
debug 2020-02-25 15:07:24: Received Zigbee message from '0xccccccfffed86303', type 'attributeReport', cluster 'ssIasWd', data '{"61440":0}' from endpoint 1 with groupID 0
debug 2020-02-25 15:07:24: No converter available for 'M420' with cluster 'ssIasWd' and type 'attributeReport' and data '{"61440":0}'
error 2020-02-25 15:07:34: Publish 'set' 'warning' to '0xccccccfffed86303' failed: 'Error: Command 0xccccccfffed86303/1 ssIasWd.startWarning({"startwarninginfo":53,"warningduration":"5"}, {"timeout":10000,"defaultResponseTimeout":15000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Timeout - 16339 - 1 - 33 - 1282 - 11 after 15000ms)'
debug 2020-02-25 15:07:34: Error: Command 0xccccccfffed86303/1 ssIasWd.startWarning({"startwarninginfo":53,"warningduration":"5"}, {"timeout":10000,"defaultResponseTimeout":15000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Timeout - 16339 - 1 - 33 - 1282 - 11 after 15000ms)
    at Endpoint.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:347:23)
    at Generator.throw (<anonymous>)
    at rejected (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65)
info  2020-02-25 15:07:34: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'warning' to '0xccccccfffed86303' failed: 'Error: Command 0xccccccfffed86303/1 ssIasWd.startWarning({\"startwarninginfo\":53,\"warningduration\":\"5\"}, {\"timeout\":10000,\"defaultResponseTimeout\":15000,\"manufacturerCode\":null,\"disableDefaultResponse\":false}) failed (Error: Timeout - 16339 - 1 - 33 - 1282 - 11 after 15000ms)'","meta":{"friendly_name":"0xccccccfffed86303"}}'
debug 2020-02-25 15:08:10: Saving state to file /opt/zigbee2mqtt/data/state.json

Koenkk, 2020-02-25:
Should be fixed in the latest zigbee2mqtt dev branch: https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html

geeano, 2020-02-25:
Thank you Koen !

geeano, 2020-02-25:
### ugraded to zigbee2mqtt dev branch.

**Result**

command:
mosquitto_pub -t zigbee2mqtt/0xccccccfffed86303/set -m '{"warning": {"mode": "emergency", "ĺevel": "high", "duration": "2"}}'

Log result:
debug 2020-02-26 00:31:32: Received MQTT message on 'zigbee2mqtt/0xccccccfffed86303/set' with data '{"warning": {"mode": "emergency", "ĺevel": "high", "duration": "2"}}'
debug 2020-02-26 00:31:32: Publishing 'set' 'warning' to '0xccccccfffed86303'
debug 2020-02-26 00:31:32: Received Zigbee message from '0xccccccfffed86303', type 'attributeReport', cluster 'ssIasWd', data '{"61440":3}' from endpoint 1 with groupID 0
debug 2020-02-26 00:31:32: No converter available for 'HS2WD-E' with cluster 'ssIasWd' and type 'attributeReport' and data '{"61440":3}'
debug 2020-02-26 00:31:34: Received Zigbee message from '0xccccccfffed86303', type 'attributeReport', cluster 'ssIasWd', data '{"61440":0}' from endpoint 1 with groupID 0
debug 2020-02-26 00:31:34: No converter available for 'HS2WD-E' with cluster 'ssIasWd' and type 'attributeReport' and data '{"61440":0}'

No error anymore !
thanks

Koenkk, 2020-02-26:
Great! I assume this can be closed now.

closed 2020-02-26 by Koenkk

geeano, 2020-03-24:
I sent a donation !
Thanks for the good work