Koenkk/zigbee2mqtt #2976: Timeout error with Tradfri LED1623G12

louismulder, 2020-02-20:
# Bug Report
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/4230998/log.txt)

## What happened
An timeout error is given, when a get command is given.
The connection with the MQTT server is also short broken. 
Sometimes the device is working and sometimes not 
## What did you expect to happen
No timeout error
## How to reproduce it (minimal and precise)
It always happens
## Debug Info
zigbee2mqtt version:1.10.0
CC253X firmware version:20190608

dani203, 2020-02-20:
I have the same issue with many of my Ikea lights, aswell as my Osram Smart Socket and my Osram Garden pole lights. Sometimes just a simple powercycle fixes it, sometimes i actually need to completely re- pair the lights in order to fix the Problem. Very annoying!

Koenkk, 2020-02-20:
Please try with the latest dev branch. Yesterday I've pushed a fix to rediscover the route on timeout.

dani203, 2020-02-20:
Just installed Z2M-edge, i will keep you updated if that fixes it.
Thank you for your work!

louismulder, 2020-02-21:
I have installed the last dev branch and the good news is that the timeout error is gone, but the bad news is that it will took 18 seconds before the command is executed on the device.
See the logging:
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/4236952/log.txt)

Koenkk, 2020-02-21:
But after this does the bulb work as expected?

louismulder, 2020-02-21:
Hi Koen,

Yes I can give Brightness, On and Off, but always executing with a delay of 18 seconds. So this is not a working situation.

louismulder, 2020-02-21:
Here is the log with the DEBUG=zigbee-herdsman* 
[debug.txt](https://github.com/Koenkk/zigbee2mqtt/files/4237679/debug.txt)

Koenkk, 2020-02-23:
- Do you really have only 1 device connected in your network?
- Is your CC2531 connected through a USB extension cable? (to avoid interference)
- How far is the bulb from the coordinator?
- Can you try with the source routing firmware: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing

dani203, 2020-02-23:
I'm still encountering timeout issues even with the developement build (zigbee2mqtt-edge for home assistent/hassio)

the weird thing i noticed is, that it doesn't seem to happen when i call a light from a group, in that case it responds instantaniously. 

The Problem accours with "0x7cb03eaa00adfdd4", which is part of the group "balkon" as defined in the groups.yaml

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/4242355/log.txt)

louismulder, 2020-02-23:
> * Do you really have only 1 device connected in your network?
> * Is your CC2531 connected through a USB extension cable? (to avoid interference)
> * How far is the bulb from the coordinator?
> * Can you [...]

I have tested it again with version 1.10.0 with the following:
I'am testing it with only one device
The CC2531 is connected with an USB extension cable
The bulb is 50 cm far from the coordinator
I have also tried source routing. Same result

Koenkk, 2020-02-23:
@dani203 the zigbee2mqtt edge addon update takes a little bit longer (could be a few hours), please try again in a few hours and provide the herdsman debug logging. https://www.zigbee2mqtt.io/information/debug.html#hassio-addon

@louismulder could you also provide the herdmsan debug logging with the latest zigbee2mqtt dev branch? (https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging). Try turning the device on/off a few times.

louismulder, 2020-02-23:
Here are the tests.
First one with zstack1.2 default
[debug_default.txt](https://github.com/Koenkk/zigbee2mqtt/files/4242464/debug_default.txt)

Second with zstack1.2 source
[debug_source.txt](https://github.com/Koenkk/zigbee2mqtt/files/4242465/debug_source.txt)

Koenkk, 2020-02-24:
@louismulder very strange issue, could you sniff the traffic of this? https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

dani203, 2020-02-25:
> @dani203 the zigbee2mqtt edge addon update takes a little bit longer (could be a few hours), please try again in a few hours and provide the herdsman debug logging. https://www.zigbee2mqtt.io/information/debug.html#hassio-addon
> 
Here is my Logfile, as you can see, trying to switch the lamp 0x7cb03eaa00adfdd4 directly ended in a timeout. However switching it's corresponding group "Balkon" worked flawlessley.
It doesn't always fail to switch by the way, sometimes it works great, often it doesn't.

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/4249701/log.txt)

Koenkk, 2020-02-25:
Via a group it's broadcast so all devices will receive it. Could you sniff the traffic?

louismulder, 2020-02-25:
> @louismulder very strange issue, could you sniff the traffic of this? https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

@Koenkk I have tried to setup a Zboss sniffer according the documentation, but I couldn't get it working.
I only see broadcast messages.
[image]

dani203, 2020-02-25:
Sorry, can't help with that at the moment, i don't have a way of flashing software to the dongle, 
 since i bought it with the right firmware already on it. I will look into getting a programmer and a second dongle to sniff the traffic

louismulder, 2020-02-26:
@Koenkk I have now found, why zboss didn't work:
ZBoss starts default with 0xC(12) but should start with 0xB(11)
Tomorrow I'll try to create the Wireshark dump.
[image]

louismulder, 2020-02-27:
@Koenkk I think that the issue is solved.
I have a production environment (1.10.0) and a test environment (1.10.1). Only one environment could be active was my issue.
So I paused my production environment and started after that the test environment and now there are NO issues.

Pls can you add the settings for ZBoss in the documentation?

Koenkk, 2020-02-27:
Good that you found the problem!

- When there are 2 networks using the same `pan_id` and `channel` you will indeed run into these strange problems.
- I've updated the sniffing ZBOSS documentation

closed 2020-02-27 by Koenkk