Koenkk/zigbee2mqtt #2977: Error: SRSP - SYS - osalNvWrite after 6000ms

timdonovanuk, 2020-02-20:
Just logging this as I seem to be breaking the record for new errors this week :)

# Bug Report
## What happened

Clean install of debian VM on proxmox. Clean install of docker, and z2m:

```
zigbee2mqtt:info  2020-02-20 14:06:46: Starting zigbee-herdsman...
zigbee2mqtt:error 2020-02-20 14:07:32: Error while starting zigbee-herdsman
zigbee2mqtt:error 2020-02-20 14:07:32: Failed to start zigbee
zigbee2mqtt:error 2020-02-20 14:07:32: Exiting...
zigbee2mqtt:error 2020-02-20 14:07:32: Error: SRSP - SYS - osalNvWrite after 6000ms
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at listOnTimeout (internal/timers.js:531:17)
    at processTimers (internal/timers.js:475:7)
npm ERR! code ELIFECYCLE

CC1352P-2 with latest firmware.

timdonovanuk, 2020-02-20:
> test -w /dev/ttyACM0 && echo success || echo failure
this reports success

Just saw the FAQ about needing to push the CC1352P-2 button on intel NUC. Tried that also.

AndrzejOlender, 2020-02-20:
I had that error yesterday, the FAQ description worked.

timdonovanuk, 2020-02-20:
The odd thing is I had the CC1352P-2 working "fine" in HassOS on Proxmox with USB passed through, never needing to push any button. Only just found out about this.

I've tried the button push thing anyway, no joy.

timdonovanuk, 2020-02-20:
My CC1352P-2 seems to be bricked. Reflashed it even, still same thing.

Koenkk, 2020-02-20:
Make sure to execute step 5 from https://www.zigbee2mqtt.io/information/flashing_via_uniflash.html

timdonovanuk, 2020-02-20:
Pretty sure I did that (although it says not required in the steps) but tried again just to be sure. Different error this time but similar "Error: AREQ - SYS - resetInd after 30000ms", which is already logged elsewhere. I seem to get lots of different errors, but maybe this was caused by a missteps in flashing so will close for now. Cheers.

closed 2020-02-20 by timdonovanuk