Koenkk/zigbee2mqtt #2978: Support of ORVIBO Neutral Smart Switch 3 Gang T18W3Z

aivs, 2020-02-20:
Hi! 
I successfully paired Switch. In node_modules/zigbee-herdsman-converters/devices.js added:
```
        zigbeeModel: ['31c989b65ebb45beaf3b67b1361d3965'],
        model: 'T18W3Z',
        vendor: 'ORVIBO',
        description: 'Neutral Smart Switch 3 Gang',
        supports: 'on/off',
        fromZigbee: [],
        toZigbee: [tz.on_off],
        endpoint: (device) => {
            return {'l1': 1, 'l2': 2, 'l3': 3};
        },
```
After that I can control each of the 3 relays. But I don't get any data when I press any of the switch buttons. I think I need to add some configurations, but what?

Here it is full pairing log:
```
zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,12,69,202,104,231,89,92,160,23,0,75,18,0,0,0,231] +20s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,12,69,202,104,231,89,92,160,23,0,75,18,0,0,0,231] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 12 - 2 - 5 - 202 - [104,231,89,92,160,23,0,75,18,0,0,0] - 231 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - tcDeviceInd - {"nwkaddr":59240,"extaddr":"0x00124b0017a05c59","parentaddr":0} +20s
  zigbee-herdsman:controller:log Device '0x00124b0017a05c59' joined +20s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:controller:log Device '0x00124b0017a05c59' accepted by handler +1ms
  zigbee-herdsman:controller:log New device '0x00124b0017a05c59' joined +0ms
  zigbee-herdsman:controller:log Creating device '0x00124b0017a05c59' +1ms
zigbee2mqtt:info  2020-02-21 01:12:09: Device '0x00124b0017a05c59' joined
zigbee2mqtt:info  2020-02-21 01:12:09: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x00124b0017a05c59"}}'
  zigbee-herdsman:controller:log Interview '0x00124b0017a05c59' start +11ms
zigbee2mqtt:info  2020-02-21 01:12:09: Starting interview of '0x00124b0017a05c59'
zigbee2mqtt:info  2020-02-21 01:12:09: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x00124b0017a05c59"}}'
  zigbee-herdsman:controller:device Interview - start device '0x00124b0017a05c59' +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":59240,"nwkaddrofinterest":59240} +1m
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,104,231,104,231,35] +1m
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102] +42ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102 +1ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0} +1m
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,18,69,130,104,231,0,104,231,1,64,142,0,0,80,160,0,0,0,160,0,0,74] +641ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,18,69,130,104,231,0,104,231,1,64,142,0,0,80,160,0,0,0,160,0,0,74] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 18 - 2 - 5 - 130 - [104,231,0,104,231,1,64,142,0,0,80,160,0,0,0,160,0,0] - 74 +1ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - nodeDescRsp - {"srcaddr":59240,"status":0,"nwkaddr":59240,"logicaltype_cmplxdescavai_userdescavai":1,"apsflags_freqband":64,"maccapflags":142,"manufacturercode":0,"maxbuffersize":80,"maxintransfersize":160,"servermask":0,"maxouttransfersize":160,"descriptorcap":0} +687ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:controller:device Interview - got node descriptor for device '0x00124b0017a05c59' +670ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - activeEpReq - {"dstaddr":59240,"nwkaddrofinterest":59240} +666ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,5,104,231,104,231,36] +666ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,13,69,193,104,231,104,231,89,92,160,23,0,75,18,0,142,236] +10ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,13,69,193,104,231,104,231,89,92,160,23,0,75,18,0,142,236] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 13 - 2 - 5 - 193 - [104,231,104,231,89,92,160,23,0,75,18,0,142] - 236 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - endDeviceAnnceInd - {"srcaddr":59240,"nwkaddr":59240,"ieeeaddr":"0x00124b0017a05c59","capabilities":142} +12ms
  zigbee-herdsman:controller:log Device announce '0x00124b0017a05c59' +686ms
zigbee2mqtt:debug 2020-02-21 01:12:10: Device '0x00124b0017a05c59' announced itself
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +5ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,5,0,97] +4ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,5,0,97] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 5 - [0] - 97 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - activeEpReq - {"status":0} +665ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,9,69,133,104,231,0,104,231,3,3,2,1,202] +26ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,9,69,133,104,231,0,104,231,3,3,2,1,202] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 9 - 2 - 5 - 133 - [104,231,0,104,231,3,3,2,1] - 202 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - activeEpRsp - {"srcaddr":59240,"status":0,"nwkaddr":59240,"activeepcount":3,"activeeplist":[3,2,1]} +38ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:controller:device Interview - got active endpoints for device '0x00124b0017a05c59' +51ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - simpleDescReq - {"dstaddr":59240,"nwkaddrofinterest":59240,"endpoint":1} +51ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,5,37,4,104,231,104,231,1,37] +51ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,4,0,96] +20ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,4,0,96] +1ms
[... 134 lines cut ...]
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":5} +30ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,34,68,129,0,0,0,0,104,231,1,1,0,44,0,227,97,216,0,0,14,24,6,1,0,64,0,66,6,118,51,46,49,46,50,104,231,29,223] +20ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,34,68,129,0,0,0,0,104,231,1,1,0,44,0,227,97,216,0,0,14,24,6,1,0,64,0,66,6,118,51,46,49,46,50,104,231,29,223] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 34 - 2 - 4 - 129 - [0,0,0,0,104,231,1,1,0,44,0,227,97,216,0,0,14,24,6,1,0,64,0,66,6,118,51,46,49,46,50,104,231,29] - 223 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":0,"srcaddr":59240,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":44,"securityuse":0,"timestamp":14180835,"transseqnumber":0,"len":14,"data":{"type":"Buffer","data":[24,6,1,0,64,0,66,6,118,51,46,49,46,50]}} +21ms
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true},"transactionSequenceNumber":6,"manufacturerCode":null,"commandIdentifier":1},"Payload":[{"attrId":16384,"status":0,"dataType":66,"attrData":"v3.1.2"}]},"address":59240,"endpoint":1,"linkquality":44,"groupID":0}' +52ms
zigbee2mqtt:debug 2020-02-21 01:12:10: Received Zigbee message from '0x00124b0017a05c59', type 'readResponse', cluster 'genBasic', data '{"swBuildId":"v3.1.2"}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-02-21 01:12:10: No converter available for 'T18W3Z' with cluster 'genBasic' and type 'readResponse' and data '{"swBuildId":"v3.1.2"}'
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +6ms
  zigbee-herdsman:controller:device Interview - got 'swBuildId' for device '0x00124b0017a05c59' +51ms
  zigbee-herdsman:controller:device Interview - completed for device '0x00124b0017a05c59' +3ms
  zigbee-herdsman:controller:log Succesfully interviewed '0x00124b0017a05c59' +9ms
zigbee2mqtt:info  2020-02-21 01:12:10: Successfully interviewed '0x00124b0017a05c59', device has successfully been paired
zigbee2mqtt:info  2020-02-21 01:12:10: Device '0x00124b0017a05c59' is supported, identified as: ORVIBO Neutral Smart Switch 3 Gang (T18W3Z)
[... cut ...]
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 13 - 2 - 5 - 193 - [104,231,104,231,89,92,160,23,0,75,18,0,142] - 236 +1ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - endDeviceAnnceInd - {"srcaddr":59240,"nwkaddr":59240,"ieeeaddr":"0x00124b0017a05c59","capabilities":142} +24s
  zigbee-herdsman:controller:log Device announce '0x00124b0017a05c59' +24s
zigbee2mqtt:debug 2020-02-21 01:12:35: Device '0x00124b0017a05c59' announced itself
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +5ms
```

aivs, 2020-02-22:
Sniffer shows that device send to 10 endpoint.  This configuration not works!
```
{
        zigbeeModel: ['31c989b65ebb45beaf3b67b1361d3965'],
        model: 'T18W3Z',
        vendor: 'ORVIBO',
        description: 'Neutral Smart Switch 3 Gang',
        supports: 'on/off',
        fromZigbee: [],
        toZigbee: [tz.on_off],
        meta: {configureKey: 1},
        configure: async (device, coordinatorEndpoint) => {
            await bind(device.getEndpoint(10), coordinatorEndpoint, ['genOnOff']);
            await configureReporting.onOff(device.getEndpoint(10));
        },
        endpoint: (device) => {
            return {'l1': 1, 'l2': 2, 'l3': 3};
        },
    }
```
[image]

aivs, 2020-02-22:
And this didn't work

```
{
        zigbeeModel: ['31c989b65ebb45beaf3b67b1361d3965'],
        model: 'T18W3Z',
        vendor: 'ORVIBO',
        description: 'Neutral Smart Switch 3 Gang',
        supports: 'on/off',
        fromZigbee: [],
        toZigbee: [tz.on_off],
        meta: {configureKey: 1},
        configure: async (device, coordinatorEndpoint) => {
                const coordinatorEndpoint10 = coordinatorEndpoint;
                coordinatorEndpoint10.ID = 10;
                await bind(device.getEndpoint(1), coordinatorEndpoint10, ['genOnOff']);
                console.log("-------------------------------------",coordinatorEndpoint10);
        },
        endpoint: (device) => {
            return {'l1': 1, 'l2': 2, 'l3': 3};
        },
    }
```

Koenkk, 2020-02-23:
Try with

```js
{
    zigbeeModel: ['31c989b65ebb45beaf3b67b1361d3965'],
    model: 'T18W3Z',
    vendor: 'ORVIBO',
    description: 'Neutral Smart Switch 3 Gang',
    supports: 'on/off',
    fromZigbee: [],
    toZigbee: [tz.on_off],
    meta: {configureKey: 2},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint1 = device.getEndpoint(1);
        await bind(endpoint1, coordinatorEndpoint, ['genOnOff']);
        await configureReporting.onOff(endpoint1);
        const endpoint2 = device.getEndpoint(1);
        await bind(endpoint2, coordinatorEndpoint, ['genOnOff']);
        await configureReporting.onOff(endpoint2);
        const endpoint3 = device.getEndpoint(1);
        await bind(endpoint3, coordinatorEndpoint, ['genOnOff']);
        await configureReporting.onOff(endpoint3);
    },
    endpoint: (device) => {
        return {'l1': 1, 'l2': 2, 'l3': 3};
    },
}
```

aivs, 2020-02-24:
I noticed, you changed configureKey from 1 to 2. What this mean?
I think you made a typo with:
```
const endpoint2 = device.getEndpoint(1);
const endpoint3 = device.getEndpoint(1);
```
I changed to:
```
const endpoint2 = device.getEndpoint(2);
const endpoint3 = device.getEndpoint(3);
```

I added code:
{
```
        zigbeeModel: ['31c989b65ebb45beaf3b67b1361d3965'],
        model: 'T18W3Z',
        vendor: 'ORVIBO',
        description: 'Neutral Smart Switch 3 Gang',
        supports: 'on/off',
        fromZigbee: [],
        toZigbee: [tz.on_off],
        meta: {configureKey: 2},
        configure: async (device, coordinatorEndpoint) => {
                console.log("---------------------------------- START CONFIGURE");
                const endpoint1 = device.getEndpoint(1);
                await bind(endpoint1, coordinatorEndpoint, ['genOnOff']);
                await configureReporting.onOff(endpoint1);
                const endpoint2 = device.getEndpoint(2);
                await bind(endpoint2, coordinatorEndpoint, ['genOnOff']);
                await configureReporting.onOff(endpoint2);
                const endpoint3 = device.getEndpoint(3);
                await bind(endpoint3, coordinatorEndpoint, ['genOnOff']);
                await configureReporting.onOff(endpoint3);
        },
        endpoint: (device) => {
                return {'l1': 1, 'l2': 2, 'l3': 3};
        }
    }
```

After pairing I'm again started configure:
`mosquitto_pub -t zigbee2mqtt/bridge/configure -m '0x00124b0017a05c59'`

But an error appears:
```
zigbee2mqtt:debug 2020-02-24 13:43:15: Received MQTT message on 'zigbee2mqtt/bridge/configure' with data '0x00124b0017a05c59'
zigbee2mqtt:info  2020-02-24 13:43:15: Configuring '0x00124b0017a05c59'
---------------------------------- START CONFIGURE
  zigbee-herdsman:controller:endpoint Bind 0x00124b0017a05c59/1 genOnOff from 'undefined' +17s
  zigbee-herdsman:controller:endpoint Bind 0x00124b0017a05c59/1 genOnOff from 'undefined' failed (TypeError: Cannot read property 'toString' of undefined) +17s
zigbee2mqtt:error 2020-02-24 13:43:15: Failed to configure '0x00124b0017a05c59', attempt 2 (Error: Bind 0x00124b0017a05c59/1 genOnOff from 'undefined' failed (TypeError: Cannot read property 'toString' of undefined)
    at Endpoint.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:235:23)
    at Generator.throw (<anonymous>)
    at rejected (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65)
    at processTicksAndRejections (internal/process/task_queues.js:97:5))
```

aivs, 2020-02-24:
I added logs output:
```
{
        zigbeeModel: ['31c989b65ebb45beaf3b67b1361d3965'],
        model: 'T18W3Z',
        vendor: 'ORVIBO',
        description: 'Neutral Smart Switch 3 Gang',
        supports: 'on/off',
        fromZigbee: [],
        toZigbee: [tz.on_off],
        meta: {configureKey: 2},
        configure: async (device, coordinatorEndpoint) => {
                console.log("---------------------------------- START CONFIGURE");
                console.log("---------------------------------- ",device.getEndpoint(1));
                console.log("---------------------------------- ",coordinatorEndpoint);
                const endpoint1 = device.getEndpoint(1);
                await bind(endpoint1, coordinatorEndpoint, ['genOnOff']);
                await configureReporting.onOff(endpoint1);
                const endpoint2 = device.getEndpoint(2);
                await bind(endpoint2, coordinatorEndpoint, ['genOnOff']);
                await configureReporting.onOff(endpoint2);
                const endpoint3 = device.getEndpoint(3);
                await bind(endpoint3, coordinatorEndpoint, ['genOnOff']);
                await configureReporting.onOff(endpoint3);
        },
        endpoint: (device) => {
                return {'l1': 1, 'l2': 2, 'l3': 3};
        }
    }
``` 

The problem with **coordinatorEndpoint** is ** undefined**. I don't understand why this happens.
```
zigbee2mqtt:info  2020-02-24 23:34:41: Configuring '0x00124b0017a05c59'
---------------------------------- START CONFIGURE
----------------------------------  Endpoint {
  ID: 1,
  profileID: 260,
  deviceID: 0,
  inputClusters: [ 0, 5, 4, 6 ],
  outputClusters: [ 0 ],
  deviceNetworkAddress: 7221,
  deviceIeeeAddress: '0x00124b0017a05c59',
  clusters: { genBasic: { attributes: [Object] } },
  _binds: []
}
----------------------------------  undefined
  zigbee-herdsman:controller:endpoint Bind 0x00124b0017a05c59/1 genOnOff from 'undefined' +21s
  zigbee-herdsman:controller:endpoint Bind 0x00124b0017a05c59/1 genOnOff from 'undefined' failed (TypeError: Cannot read property 'toString' of undefined) +21s
zigbee2mqtt:error 2020-02-24 23:34:41: Failed to configure '0x00124b0017a05c59', attempt 3 (Error: Bind 0x00124b0017a05c59/1 genOnOff from 'undefined' failed (TypeError: Cannot read property 'toString' of undefined)
    at Endpoint.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:235:23)
    at Generator.throw (<anonymous>)
    at rejected (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65))
```

aivs, 2020-02-24:
I found out that coordinator don't have first endpoint:
```
Device {
  ID: 1,
  _type: 'Coordinator',
  _ieeeAddr: '0x00124b0012022dc2',
  _networkAddress: 0,
  _manufacturerID: 0,
  _endpoints: [
    Endpoint {
      ID: 2,
      profileID: 257,
      deviceID: 5,
      inputClusters: [],
      outputClusters: [],
      deviceNetworkAddress: 0,
      deviceIeeeAddress: '0x00124b0012022dc2',
      clusters: {},
      _binds: []
    },
    Endpoint {
      ID: 3,
      profileID: 261,
      deviceID: 5,
      inputClusters: [],
      outputClusters: [],
      deviceNetworkAddress: 0,
      deviceIeeeAddress: '0x00124b0012022dc2',
      clusters: {},
      _binds: []
    },
    Endpoint {
      ID: 4,
      profileID: 263,
      deviceID: 5,
      inputClusters: [],
      outputClusters: [],
      deviceNetworkAddress: 0,
      deviceIeeeAddress: '0x00124b0012022dc2',
      clusters: {},
      _binds: []
    },
    Endpoint {
      ID: 5,
      profileID: 264,
      deviceID: 5,
      inputClusters: [],
      outputClusters: [],
      deviceNetworkAddress: 0,
      deviceIeeeAddress: '0x00124b0012022dc2',
      clusters: {},
      _binds: []
    },
    Endpoint {
      ID: 6,
      profileID: 265,
      deviceID: 5,
      inputClusters: [],
      outputClusters: [],
      deviceNetworkAddress: 0,
      deviceIeeeAddress: '0x00124b0012022dc2',
      clusters: {},
      _binds: []
    },
    Endpoint {
      ID: 8,
      profileID: 260,
      deviceID: 5,
      inputClusters: [],
      outputClusters: [],
[... 84 lines cut ...]
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:controller:endpoint ConfigureReporting 0x00124b0017a05c59/1 genOnOff([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":0}], {"timeout":10000,"defaultResponseTimeout":15000,"manufacturerCode":null,"disableDefaultResponse":true}) +58ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":59506,"destendpoint":1,"srcendpoint":1,"clusterid":6,"transid":3,"options":0,"radius":30,"len":11,"data":{"type":"Buffer","data":[16,4,6,0,0,0,16,0,0,16,14]}} +59ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,21,36,1,114,232,1,1,6,0,3,0,30,11,16,4,6,0,0,0,16,0,0,16,14,166] +60ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100] +18ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100 +1ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0} +59ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,3,197] +5ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,3,197] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,3] - 197 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":3} +28ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:controller:endpoint ConfigureReporting 0x00124b0017a05c59/1 genOnOff([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":0}], {"timeout":10000,"defaultResponseTimeout":15000,"manufacturerCode":null,"disableDefaultResponse":true}) failed (Error: Timeout - 59506 - 1 - 4 - 6 - 7 after 10000ms) +24s
zigbee2mqtt:error 2020-02-25 00:27:04: Failed to configure '0x00124b0017a05c59', attempt 3 (Error: ConfigureReporting 0x00124b0017a05c59/1 genOnOff([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":0}], {"timeout":10000,"defaultResponseTimeout":15000,"manufacturerCode":null,"disableDefaultResponse":true}) failed (Error: Timeout - 59506 - 1 - 4 - 6 - 7 after 10000ms)
    at Endpoint.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:316:23)
    at Generator.throw (<anonymous>)
    at rejected (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65))
```
Sniffer shows that binding configured:
[image]
[orvibo_binding_and_report_configuration.pcapng.zip](https://github.com/Koenkk/zigbee2mqtt/files/4247001/orvibo_binding_and_report_configuration.pcapng.zip)

**If I can't configure binding and report, so is there a way to see all packets on all endpoint from this switch?**

Koenkk, 2020-02-25:
Coordinator endpoint 1 is mandatory, I'm not sure why it's not there for you.

Can you manually modify your `database.db` and add the endpoint?

aivs, 2020-02-25:
I added endpoint 1 to database.db. But this didn't solve the problem.
```
----------------------------------  Endpoint {
  ID: 1,
  profileID: 257,
  deviceID: 5,
  inputClusters: [],
  outputClusters: [],
  deviceNetworkAddress: 0,
  deviceIeeeAddress: '0x00124b0012022dc2',
  clusters: {},
  _binds: []
}
  zigbee-herdsman:controller:endpoint Bind 0x00124b0017a05c59/1 genOnOff from '0x00124b0012022dc2/1' +19s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - bindReq - {"dstaddr":59506,"srcaddr":"0x00124b0017a05c59","srcendpoint":1,"clusterid":6,"dstaddrmode":3,"dstaddress":"0x00124b0012022dc2","dstendpoint":1} +19s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,23,37,33,114,232,89,92,160,23,0,75,18,0,1,6,0,3,194,45,2,18,0,75,18,0,1,193] +19s
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,33,0,69] +23ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,33,0,69] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 33 - [0] - 69 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - bindReq - {"status":0} +19s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,69,161,114,232,0,125] +36ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,161,114,232,0,125] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 161 - [114,232,0] - 125 +1ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - bindRsp - {"srcaddr":59506,"status":0} +66ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:controller:endpoint ConfigureReporting 0x00124b0017a05c59/1 genOnOff([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":0}], {"timeout":10000,"defaultResponseTimeout":15000,"manufacturerCode":null,"disableDefaultResponse":true}) +58ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":59506,"destendpoint":1,"srcendpoint":1,"clusterid":6,"transid":2,"options":0,"radius":30,"len":11,"data":{"type":"Buffer","data":[16,3,6,0,0,0,16,0,0,16,14]}} +59ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,21,36,1,114,232,1,1,6,0,2,0,30,11,16,3,6,0,0,0,16,0,0,16,14,160] +59ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100] +19ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0} +59ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,2,196] +4ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,2,196] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,2] - 196 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":2} +28ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:controller:endpoint ConfigureReporting 0x00124b0017a05c59/1 genOnOff([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":0}], {"timeout":10000,"defaultResponseTimeout":15000,"manufacturerCode":null,"disableDefaultResponse":true}) failed (Error: Timeout - 59506 - 1 - 3 - 6 - 7 after 10000ms) +19s
zigbee2mqtt:error 2020-02-25 10:48:54: Failed to configure '0x00124b0017a05c59', attempt 3 (Error: ConfigureReporting 0x00124b0017a05c59/1 genOnOff([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":0}], {"timeout":10000,"defaultResponseTimeout":15000,"manufacturerCode":null,"disableDefaultResponse":true}) failed (Error: Timeout - 59506 - 1 - 3 - 6 - 7 after 10000ms)
    at Endpoint.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:316:23)
    at Generator.throw (<anonymous>)
    at rejected (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65))
```

Koenkk, 2020-02-25:
Could you try to reflash your CC2531? Looks that something fishy is going on.

aivs, 2020-02-25:
YES!
It works!
I used the shepherd firmware created in 2016. With latest firmware https://github.com/Koenkk/Z-Stack-firmware/raw/master/coordinator/Z-Stack_Home_1.2/bin/default/CC2531_DEFAULT_20190608.zip it works!

closed 2020-02-25 by aivs