Koenkk/zigbee2mqtt #2980: Aqara Opple Catch all problem

kisseler, 2020-02-21:
I received the new Aqara Opple Switches. Paring is working flawlessly, but 2 out of 3 are being paired to "catch all" devices in my ZigBee network.

I saw that people are having the same problem, but nowhere I found a solution. Does anyone know how to avoid that behaviour? I would like to pair the device and ONLY bind it to one device or group.

Thank you ver much for your help!

tubededentifrice, 2020-02-23:
Have you tried https://www.zigbee2mqtt.io/information/binding.html ? Something I do is just unplugging the Zigbee devices that are getting caught to avoid bothering, but that's because I only have 2 plugs...

kisseler, 2020-03-04:
Issue solved after upgrading to 1.11-dev - and after zigbee2mqtt confirming successfull interview by naming correct device.
Now also single, double, triple, hold and release work on all buttons.
Fantastic! Thanks, @Koenkk !

closed 2020-03-04 by kisseler