Koenkk/zigbee2mqtt #2986: Question: HA New Brightness Stepping Feature

flexage, 2020-02-23:
Quick question, will/does the HA integration add remotes/dimmers in a way that will work with the new Brightness Stepping feature in the next HA release? (https://rc.home-assistant.io/blog/2020/02/17/release-106/) It looks like it's based on triggers such as "Button Rotated Right".

[image]

It would be great to take advantage of these simple automation within HA for devices such as the Ikea 5 Button Remote Control (E1524/E1810) and Ikea Wireless Dimmer (ICTC-G-1). Currently, I end up producting fairly complex login in Node Red to accomplish this. Would be much easier to use the logic that will be baked into HA.

Koenkk, 2020-02-23:
If its also added to https://www.home-assistant.io/integrations/light.mqtt/ it could be added.

flexage, 2020-02-23:
Looks like they've added device trigger abilities to MQTT.
Think I've found the related pull request: https://github.com/home-assistant/home-assistant/pull/31679
Home assistant.io documentation is still pending though.

Koenkk, 2020-02-23:
Let's wait until the next Home Assistant release (so we have the updated docs and to easily test this).

flexage, 2020-02-23:
Amazing, just wanted to put it on radar👍

Koenkk, 2020-02-28:
I've just tested this but it already seems to work fine, no additional changes from the Zigbee2mqtt side are required.

closed 2020-02-28 by Koenkk

flexage, 2020-02-29:
Just installed the latest update and had a play. You are correct, you can use the new brightness stepping feature if you set the trigger as a `state` trigger, however this requires the end user to know the possible values of the action that the remote/button provides.

I guess something I didn't fully qualify in my original question was the ability to use `device` triggers. The seem to be ready to go triggers specified on HA's new device model. In the screenshot from the OP, you can see that the trigger is of type `device` the device is a `SYMFONISK Sound Controller` and the trigger action type is `Button Rotated "Right"`.

Using a Tradfri Remote Control I couldn't see any device actions for button presses or holds, as in the screenshot below:

[image]

HA seems to be making a big push towards Devices, and these 'device actions' seem to be baked right in to device integrations.

Does Z2M have any plans to support device actions? I've not looked at the Z2M code since the move to herdsman, but could these actions be included on a device level in the device schema?

Koenkk, 2020-02-29:
Ive implemented that today see #3033

flexage, 2020-03-01:
Take this official 'Ultimate Legend' medal 🏅
You win at GitHub!

flexage, 2020-03-08:
So I've had chance to play with the brightness stepping feature, along with the new device actions on the edge branch, it works really well for devices. However it appears to not able to be used with Z2M groups, in fact the light group I have set up for the kitchen doesn't seem to appear as a group at all.

As a workaround I tried enabling optimistic mode on the group, and using the device brightness stepping feature against one of the bulbs in that group, but only that single bulb changes state.

Is there possibly any other workaround for groups?