Koenkk/zigbee2mqtt #2987: Is there a dependency between the stick and the configuration files?

wsw70, 2020-02-23:
I would like to put in place an (artisanal)  High Availability system for my switches. 

I currently run Zigbee2MQTT from my main server (where the Zigbee USB stick is plugged in) and would like to reproduce this setup (with another stick) on Raspberry Pi.

That RPi would constantly check for the availability of critical services on the main server and start the backup ones if needed.

I run all servies in docker containers but Zigbee2MQTT has a hardware dependency (the USB stick). 

My question: is there anything in the configuration which refers to a specific USB stick? I am thinking about some kind of unique ID which would be provided by the device and used to pair switches etc.

In other words: if I just copy the configuration and use a different stick, would my switches still be paired and Zigbee2MQTT work? Or is the pairing done at hardware level? (= a switch is paired with a specific dongle only)

Koenkk, 2020-02-23:
See https://www.zigbee2mqtt.io/information/FAQ.html#what-does-and-does-not-require-repairing-of-all-devices

wsw70, 2020-02-23:
@Koenkk Thank you . I had a look at the FAQ but I was searching for something around "backup" and missed the (unmissable) part you pointed to.

Specifically in my case, the last two bullets:

> - Switching from CC2530/CC2531 device (physically).
> - Switching the system running Zigbee2mqtt.
>   - When doing this, make sure to copy over the contents of the data directory.

closed 2020-02-23 by wsw70

turboproc, 2020-02-23:
@Koenkk What about the issue we discovered recently related to devices having a bind to the coordinator which is based on the MAC address. MAC will change when when changing enddevice which will result the enddevice having a bind to a non-existing device.

Koenkk, 2020-02-23:
@turboproc you are right, create a bug report: https://github.com/Koenkk/zigbee-herdsman/issues/141