Koenkk/zigbee2mqtt #2989: Lqi between router's and coordinator 0 or 1

Diddlik, 2020-02-23:
While building the ZigBee network I noticed that all routers have lqi 1 or 0 to the coordinator. E.g. DW.Testing (door contact) are much further than both routers. 
Coordinator: 1352P with external antenna (capacitor re-soldered).
RouterFlurEG -> CC1352 with Router Firmware
Router.TVLED -> Xiaomi Plug

## Debug Info
zigbee2mqtt version: 1.10.0
CC2531 firmware: CC2531_router_2019_02.zip
CC1352P firmware: 20191106 Z-Stack_3.x.0 firmware.

[image]

xefil, 2020-02-23:
Hello!
Sorry for OT information, but how do you generate the map? Looks cool!
I'm having CC2531 routers with `CC2531_router_2019_02` 1.2 and I'm getting 'Failed to execute LQI'...
Thanks, Simon

Diddlik, 2020-02-23:
Hi, i used Zigbee2MqttAssistant

1igeza, 2020-02-25:
I have the same with CC2530 coordinator and CC2530 routers. Zigbee2mqtt version 1.10.0-dev (commit #9cc84fe)

[image]

highlandr, 2020-04-26:
WiFi network most probably, move the WiFi away from channel 1.
Had the same...

closed 2020-07-03 by stale[bot]