Koenkk/zigbee2mqtt #2990: zigbee2mqtt does not startup when advanced keyword is in configuration.yaml

robin-s-007, 2020-02-23:
#Bug Report
## What happened
zigbee2mqtt does not startup when advanced keyword has been added to configuration.yaml
I added this to provide a network_key to zigbee2mqtt to secure my network.
The configuration.yaml was verified with http://www.yamllint.com/, and contained no errors.
After playing the the configuration.yaml by adding and removing lines, zigbee2mqtt fails to start when advanced keyword was added.
Note: I also tested with a full configuration.yaml for the example on https://www.zigbee2mqtt.io/information/configuration.html, this one also failed.

## What did you expect to happen
zigbee2mqtt to startup and the need to repair all devices due to a key change.

## How to reproduce it (minimal and precise)
Uncomment advanced from example below
homeassistant: false
permit_join: false
mqtt:
  base_topic: /huis/zigbee2mqtt
  server: 'mqtt://homeserver.fritz.box'
serial:
  port: /dev/ttyACM0
devices: devices.yaml
groups: groups.yaml
#advanced:

## Debug Info
zigbee2mqtt version: 1.10.0
CC253X firmware version: ?

[Paste bin with tests and log output ](https://pastebin.com/NP0jtWta)

tubededentifrice, 2020-02-23:
Posting the config as code would help provide better readability, especially to preserve indentation, which matters.
Your config is probably valid YAML but invalid for zigbee2mqtt. In particular, `advanced:` is invalid by itself, it should be `advanced: {}`
Example of a working config (in hassio):

```
data_path: /share/zigbee2mqtt
devices: devices.yaml
groups: groups.yaml
homeassistant: true
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://core-mosquitto'
  user: zigbee2mqtt
  password: 123456
serial:
  port: /dev/serial/by-id/usb-Texas_Instruments_xxx
advanced:
  network_key: [1, 3, 5, 7, 9, 11, 13, 15, 0, 2, 4, 6, 8, 10, 12, 13]
  pan_id: 0x1a62
  baudrate: 115200
  report: true
  rtscts: false
  channel: 25
  availability_blacklist: []
ban: []
whitelist: []
queue: {}
socat:
  enabled: false
  master: 'pty,raw,echo=0,link=/dev/ttyZ2M,mode=777'
  slave: 'tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5'
  restartdelay: 1
  initialdelay: 1
  options: '-d -d'
  log: false
```

robin-s-007, 2020-02-23:
@tubededentifrice ,
Ok, thats good to know, here is the yaml config file for closer examination. (modified the filename to be able to upload it)
[configuration_yaml.txt](https://github.com/Koenkk/zigbee2mqtt/files/4242357/configuration_yaml.txt)

Best Regards,

Robin

robin-s-007, 2020-02-25:
Note: 
I have removed V 1.10.0 because of unstability reasons.
zigbee2mqtt remained running, but did not post any events on MQTT broker, nor were any new updates posted in the log. PI 3B+ was slow in response.

closed 2020-05-02 by stale[bot]