Koenkk/zigbee2mqtt #2992: [docs] device_options and groups

sjorge, 2020-02-23:
I was reading the docs and it's not 100% clear of `device_options` also applies to groups or not?
It looks like `retain` specified in `device_options` seems to be applied.

But this could be a side effect of the bulbs in the groups having retain param applied?

Perhaps clarify it in the docs if it does or doesn't apply to groups.

Koenkk, 2020-02-23:
Yes, these are also applied to groups. thanks!

closed 2020-02-23 by Koenkk

sjorge, 2020-02-24:
Could it be debounce is not correctly applied? Retain seems to work fine, so maybe just denounce that is ignored?

Koenkk, 2020-02-24:
Debounce is only used for received messages from zigbee, groups dont send messages from zigbee (only generated by zigbee2mqtt itself).

sjorge, 2020-02-24:
So if I have debounce: 2 in all bulbs in a group, does it mean the group gets the message delayed by 2 sec? Or will it immediately update?

It looks like I am seeing the later behavior.

~ sjorge

Koenkk, 2020-02-25:
It should be delayed by 2 sec (also checked this with the code):