Koenkk/zigbee2mqtt #2993: Can´t pair Ledvance Smart+ Plug AC10691 (again)

kirbyDE1, 2020-02-23:
# Bug Report
Hi, similar to #2557 I have a couple of Ledvance Smart+ Plug AC10691 that won't connect to Z2M.

The coordinator seems to work fine (couple of Aqara sensors connected). Also, the Ledvance should be ok in theory; a similar one is currently connected to my Hue network. The one I am experimenting with was not connected to any other network before.

I tried resetting with Touchlink, however, Z2M reports "Failed to factory reset device through Touchlink".

The user in that case sadly gave up but I want to try again, because the plug seems to be kind of common. Any starting points?

## What happened
Pairing did not happen

## What did you expect to happen
Pairing should have happened

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version: 1.10.0 (Docker)
CC253X firmware version: Z-Stack_Home_1.2 (default)

kirbyDE1, 2020-02-24:
Today, I made a fresh installation of Z2M onto a Pi3, together with a another Zigbee stick that also had  Z-Stack_Home_1.2 installed.

The Ledvance plug paired instantly and I was able to successfully toggle it on and off by sending payloads via MQTT. It behaves exactly as https://www.zigbee2mqtt.io/devices/AB3257001NJ.html. I guess it is basically the same hardware.

However, I am not sure what exactly went different this time.

A) My test install runs Z2M on Pi3 directly vs. in a Docker Container on x86 hardware for production.
B) By second (test) CC stick ist stock and has no modded antenna like my main stick.
C) In the room where my main system is, there is a Hue Bride and many Hue bulbs, as well as my Wifi access point.

Any ideas? I'll try to eliminate the variables one by one.

kirbyDE1, 2020-02-24:
Using the antenna-modded CC stick in my test room yielded no success. Again, there was absolutely no reaction in Z2M whatsoever. I could however pair my Aqara sensors just fine and link quality is considerably better for them with the antenna-modded stick.

So strange...

Koenkk, 2020-02-25:
Running multiple zigbee network at the same time can cause problems related to pairing (especially when a device was first connected to the old network). Make sure to check: https://www.zigbee2mqtt.io/information/FAQ.html#no-logging-is-shown-at-all

kirbyDE1, 2020-02-25:
Thank you for that suggestion. I habe tried some more things:

- Powering down my other Zigbee (Hue) network.
- Adding a USB-extension to both devices (server and test-Pi)
- Reflashing the (antenna-modded) CC stick with Zigbee 3.0 firmware

However, none of these changes made a difference:

I can pair the Ledvance perfectly fine to one of the sticks (the unmodded one) and it makes no difference if it is in directly in front of my Hue stuff or in the other room across the house.
I cannot pair it with the other stick (the antenna-modded one) and it also makes no difference where that stick is in the house or what the host computer is.

I have now flashed the misbehaving stick with the router firmware and will try to setup my devices this way (using both sticks). I might come back to this, if I come around buying a few more CC sticks. It seems to be a purely the problem of that one CC stick.

kirbyDE1, 2020-03-01:
Final update: Seems to be that the Ledvance Plug has really bad antennas that barely want to communicate with my unmodded stick and refuse zu work with my modded stick, because it also does not want to connect to the modded stick as router. They work a bit better on the Hue network, so I have placed them in that.

closed 2020-03-01 by kirbyDE1