Koenkk/zigbee2mqtt #2994: Xiaomi Smoke Detector not pairing.. Maximum devices reached?

ljpaff, 2020-02-24:
# Bug Report
## What happened

I have xiaomi devices installed within a zigbee network with a coordinator and a router. I have attached the device map to the issue. When I try to pair another smoke detector, it is not able to join the current network. As I read the maximum number of devices including the link to the coordinator is 21 devices. Currently there are 20 but I find it strange that they do not let me connect such a device if I have another one installed.

[image]

## What did you expect to happen
Joining device into Zigbee network.
## How to reproduce it (minimal and precise)
n/A
## Debug Info
zigbee2mqtt version: 1.9.0 (commit #287f843)
CC253X firmware version: 20190608

Koenkk, 2020-02-25:
Can you try if reflashing the CC2531 fixes the problem? (doesn't require repairing).

ljpaff, 2020-02-26:
> Can you try if reflashing the CC2531 fixes the problem? (doesn't require repairing).

Which one do i have to reflash, the coordinator, the router or both?

Koenkk, 2020-02-26:
coordinator.

ljpaff, 2020-02-26:
It worked!!!!!!. What is the problem that makes me reflash the coordinator?

Koenkk, 2020-02-26:
I think that devices that were previously a child of the coordinator still occupy a place in the direct children table. I've added this to the FAQ (as I think it's not easy solvable as it has to be done in the firmware which is not developed anymore by Texas Instruments). https://www.zigbee2mqtt.io/information/FAQ.html#no-logging-is-shown-at-all

closed 2020-03-06 by ljpaff