Koenkk/zigbee2mqtt #2995: Failed to call 'DeviceEvent' 'onZigbeeEvent'

gimlichael, 2020-02-24:
# Bug Report
## What happened
After updating to  zigbee2mqtt@1.10.0, errors are being thrown consistently every now an then.
I have tried both master and development; currently running on the dev branch.

zigbee2mqtt:error 2020-02-24 17:20:04: Failed to call 'DeviceEvent' 'onZigbeeEvent' (TypeError: zigbeeHerdsmanConverters.onEvent is not a function
    at DeviceEvent.callOnEvent (C:\zigbee2mqtt\lib\extension\deviceEvent.js:24:34)
    at DeviceEvent.onZigbeeEvent (C:\zigbee2mqtt\lib\extension\deviceEvent.js:13:18)
    at Controller.callExtensionMethod (C:\zigbee2mqtt\lib\controller.js:327:44))

zigbee2mqtt:error 2020-02-24 17:12:42: Failed to call 'DeviceEvent' 'onZigbeeStarted' (TypeError: zigbeeHerdsmanConverters.onEvent is not a function
    at DeviceEvent.callOnEvent (C:\zigbee2mqtt\lib\extension\deviceEvent.js:24:34)
    at DeviceEvent.onZigbeeStarted (C:\zigbee2mqtt\lib\extension\deviceEvent.js:7:18)
    at Controller.callExtensionMethod (C:\zigbee2mqtt\lib\controller.js:327:44))

## What did you expect to happen
No errors when running zigbee2mqtt.

## How to reproduce it (minimal and precise)
I am running the zigbee2mqtt in a Windows environment.

## Debug Info
zigbee2mqtt version: 1.10.0
CC253X firmware version: CC2531_DEFAULT_20190608

Koenkk, 2020-02-24:
After updating make sure you do a `npm ci` in the zigbee2mqtt directory

gimlichael, 2020-02-24:
> After updating make sure you do a `npm ci` in the zigbee2mqtt directory

Thank you - now it works like a charm!

closed 2020-02-24 by Koenkk