Koenkk/zigbee2mqtt #2996: Can't pair previously working Osram Flex LED strips

onik, 2020-02-24:
# Bug Report
## What happened
Device doesn't pair with interview failed error. The same behaviour happens with two similar devices, other lights and a temperature sensor work.

## What did you expect to happen
Device pairs and is usable.

## How to reproduce it (minimal and precise)
Just reset the light strip and monitor the topics under zigbee2mqtt

## Debug Info
zigbee2mqtt version: 1.10.0
CC253X firmware version: 20190608

## Logs
`  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,69,196,70,199,0,3] +1m
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,196,70,199,0,3] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 196 - [70,199,0] - 3 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":51014,"relaycount":0,"relaylist":[]} +1m
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,12,69,202,59,48,111,179,202,0,0,38,24,132,70,199,165] +105ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,12,69,202,59,48,111,179,202,0,0,38,24,132,70,199,165] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 12 - 2 - 5 - 202 - [59,48,111,179,202,0,0,38,24,132,70,199] - 165 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - tcDeviceInd - {"nwkaddr":12347,"extaddr":"0x8418260000cab36f","parentaddr":51014} +107ms
  zigbee-herdsman:controller:log Device '0x8418260000cab36f' joined +1m
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +2ms
  zigbee-herdsman:controller:log Device '0x8418260000cab36f' accepted by handler +1ms
  zigbee-herdsman:controller:log New device '0x8418260000cab36f' joined +0ms
  zigbee-herdsman:controller:log Creating device '0x8418260000cab36f' +1ms
zigbee2mqtt:info  2020-02-24 21:37:40: Device '0x8418260000cab36f' joined
zigbee2mqtt:info  2020-02-24 21:37:40: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x8418260000cab36f"}}'
  zigbee-herdsman:controller:log Interview '0x8418260000cab36f' start +14ms
zigbee2mqtt:info  2020-02-24 21:37:40: Starting interview of '0x8418260000cab36f'
zigbee2mqtt:info  2020-02-24 21:37:40: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x8418260000cab36f"}}'
  zigbee-herdsman:controller:device Interview - start device '0x8418260000cab36f' +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":12347,"nwkaddrofinterest":12347} +1m
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,59,48,59,48,35] +1m
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102] +64ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0} +1m
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +2ms
  zigbee-herdsman:controller:device Interview - node descriptor request failed for '0x8418260000cab36f', attempt 1 +10s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":12347,"nwkaddrofinterest":12347} +10s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,59,48,59,48,35] +10s
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102] +10s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0} +10s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:controller:device Interview - node descriptor request failed for '0x8418260000cab36f', attempt 2 +10s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":12347,"nwkaddrofinterest":12347} +10s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,59,48,59,48,35] +10s
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102] +10s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0} +10s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:controller:device Interview - node descriptor request failed for '0x8418260000cab36f', attempt 3 +10s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":12347,"nwkaddrofinterest":12347} +10s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,59,48,59,48,35] +10s
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102] +10s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0} +10s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +0ms
  zigbee-herdsman:controller:device Interview - node descriptor request failed for '0x8418260000cab36f', attempt 4 +10s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":12347,"nwkaddrofinterest":12347} +10s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,59,48,59,48,35] +10s
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102] +10s
[... 12 lines cut ...]
  zigbee-herdsman:controller:device Interview - node descriptor request failed for '0x8418260000cab36f', attempt 6 +10s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - activeEpReq - {"dstaddr":12347,"nwkaddrofinterest":12347} +10s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,5,59,48,59,48,36] +10s
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,5,0,97] +10s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,5,0,97] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 5 - [0] - 97 +1ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - activeEpReq - {"status":0} +10s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +2ms
  zigbee-herdsman:controller:device Interview - active endpoints request failed for '0x8418260000cab36f', attempt 1 +10s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - activeEpReq - {"dstaddr":12347,"nwkaddrofinterest":12347} +10s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,5,59,48,59,48,36] +10s
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,5,0,97] +10s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,5,0,97] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 5 - [0] - 97 +1ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - activeEpReq - {"status":0} +10s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +2ms
  zigbee-herdsman:controller:device Interview - active endpoints request failed for '0x8418260000cab36f', attempt 2 +10s
  zigbee-herdsman:controller:device Interview - failed for device '0x8418260000cab36f' with error 'Error: Interview failed because can not get active endpoints ('0x8418260000cab36f')
  zigbee-herdsman:controller:device     at Device.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/device.js:275:23)
  zigbee-herdsman:controller:device     at Generator.throw (<anonymous>)
  zigbee-herdsman:controller:device     at rejected (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/device.js:6:65)' +1ms
  zigbee-herdsman:controller:error Interview failed for '0x8418260000cab36f with error 'Error: Interview failed because can not get active endpoints ('0x8418260000cab36f')' +0ms
zigbee2mqtt:error 2020-02-24 21:39:00: Failed to interview '0x8418260000cab36f', device has not successfully been paired
zigbee2mqtt:info  2020-02-24 21:39:00: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"0x8418260000cab36f"}}'
^C`

Koenkk, 2020-02-25:
Did you check https://bro-gitlab.w2k.feico.com/tem/tem/tem_installer/blob/master/TEM_INSTALLER/MasterInstaller/MasterInstaller.Plugin.Vacuum/VacuumSetup.cs ?

Koenkk, 2020-02-25:
Did you check: https://www.zigbee2mqtt.io/information/FAQ.html#frequently-asked-questions ?

onik, 2020-02-25:
Yes, and I followed the directions. Tried repairing each device at least 3 times, device doesn't join with 0x000000000000000, so I created a new issue with zigbee-herdsman logging enabled and posted the log. Anything else I can do?

Koenkk, 2020-02-25:
Can you try relfashing the CC2531 (doesn't require repairing of your existing devices).

closed 2020-05-02 by stale[bot]

buzzdev, 2020-10-19:
I have the same issue. Osram RGBW Outdoor Flex was working previously. Since certain time it is unable to pair.
@Koenkk What would the stick re-flashing do here?

Koenkk, 2020-10-20:
@buzzdev it clear the memory of the adapter, sometimes this helps when a device refuses to pair.

onik, 2020-12-30:
For me the issue was resolved by bringing the control box of the strip very close to the coordinator, apparently during pairing the device needs to be <1m away from the coordinator. After pairing the range is longer, but still very sporadic, so I'm looking into getting some signal amplifiers.