Koenkk/zigbee2mqtt #2997: Failed to connect to the  adapter (Error: SRSP - SYS - ping after 6000ms)

fantasyjqc, 2020-02-25:
#Bug Report

What happened
zigbee2mqtt cannot connect to adpater

```
zigbee2mqtt:info  2020-02-25 09:27:08: Starting zigbee2mqtt version 1.
10.0-dev (commit #ad59ff5)
zigbee2mqtt:info  2020-02-25 09:27:08: Starting zigbee-herdsman...
zigbee2mqtt:error 2020-02-25 09:27:16: Error while starting zigbee-her
dsman
zigbee2mqtt:error 2020-02-25 09:27:16: Failed to start zigbee
zigbee2mqtt:error 2020-02-25 09:27:16: Exiting...
zigbee2mqtt:error 2020-02-25 09:27:16: Error: Failed to connect to the 
adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (/opt/zigbee2mqtt3/node_modules
/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:72:2
3)
    at Generator.throw (<anonymous>)
    at rejected (/opt/zigbee2mqtt3/node_modules/zigbee-herdsman/di
st/adapter/z-stack/adapter/zStackAdapter.js:6:65)

```

What did you expect to happen
zigbee2mqtt 

How to reproduce it (minimal and precise)
homeassistant: false
permit_join: true
mqtt:
  base_topic: zigbee2mqtt3
  server: 'mqtt://192.168.123.133:1883'
serial:
  port: 'tcp://192.168.123.139:8881'

ping and telnet is ok:

[image]

[image]

Debug Info
zigbee2mqtt version: 1.10.0-dev (commit #ad59ff5)
CC2530 firmware version: CC2530_SOURCE_ROUTING_20190619.zip

mibas74, 2020-03-02:
i confirm this issue after upgrade to 1.11.0:

●```
 zigbee2mqtt.service - zigbee2mqtt
   Loaded: loaded (/etc/systemd/system/zigbee2mqtt.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2020-03-02 21:31:58 CET; 52s ago
 Main PID: 6915 (npm)
    Tasks: 13 (limit: 2319)
   Memory: 13.7M
   CGroup: /system.slice/zigbee2mqtt.service
           ├─6915 npm
           └─6926 sh -c node index.js

Mar 02 21:32:02 openhab npm[6915]: zigbee2mqtt:info  2020-03-02 21:32:02: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-03-02.21-32-02' filename: log.txt
Mar 02 21:32:02 openhab npm[6915]: zigbee2mqtt:info  2020-03-02 21:32:02: Starting zigbee2mqtt version 1.11.0 (commit #31e5678)
Mar 02 21:32:02 openhab npm[6915]: zigbee2mqtt:info  2020-03-02 21:32:02: Starting zigbee-herdsman...
Mar 02 21:32:15 openhab npm[6915]: zigbee2mqtt:error 2020-03-02 21:32:15: Error while starting zigbee-herdsman
Mar 02 21:32:15 openhab npm[6915]: zigbee2mqtt:error 2020-03-02 21:32:15: Failed to start zigbee
Mar 02 21:32:15 openhab npm[6915]: zigbee2mqtt:error 2020-03-02 21:32:15: Exiting...
Mar 02 21:32:15 openhab npm[6915]: zigbee2mqtt:error 2020-03-02 21:32:15: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
Mar 02 21:32:15 openhab npm[6915]:     at ZStackAdapter.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:72:23)
Mar 02 21:32:15 openhab npm[6915]:     at Generator.throw (<anonymous>)
Mar 02 21:32:15 openhab npm[6915]:     at rejected (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:6:65)
```

a20gard, 2020-03-02:
I would suggest you to run 
sudo npm rebuild

and then unplug and plug again the usb key

mibas74, 2020-03-02:
hi,
thanks, i've tried this but unfortunately without success yet..

Miguelin96, 2020-03-02:
i confirm this issue after upgrade to 1.11.0:

zigbee2mqtt:info  2020-03-03 00:22:00: Logging to console and directory: '/share/zigbee2mqtt/log/2020-03-03.00-22-00' filename: log.txt
zigbee2mqtt:info  2020-03-03 00:22:01: Starting zigbee2mqtt version 1.11.0 (commit #unknown)
zigbee2mqtt:info  2020-03-03 00:22:01: Starting zigbee-herdsman...
zigbee2mqtt:error 2020-03-03 00:22:09: Error while starting zigbee-herdsman
zigbee2mqtt:error 2020-03-03 00:22:09: Failed to start zigbee
zigbee2mqtt:error 2020-03-03 00:22:09: Exiting...
zigbee2mqtt:error 2020-03-03 00:22:09: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (/zigbee2mqtt-1.11.0/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:72:23)
    at Generator.throw (<anonymous>)
    at rejected (/zigbee2mqtt-1.11.0/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:6:65)
npm
 
ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.11.0 start: `node index.js`
npm ERR!
 Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.11.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

Koenkk, 2020-03-03:
Please check your adapter communication: https://www.zigbee2mqtt.io/information/FAQ.html#help-zigbee2mqtt-fails-to-start

mibas74, 2020-03-03:
it was working and stopped after upgrade to current version.

[21:14:55] openhabian@openhab:~$ ls -l /dev/serial/by-id
lrwxrwxrwx 1 root root 13 Mar  2 23:00 usb-Texas_Instruments_TI_CC2531_USB_CDC__                    _0X00124B0018E248BF-if00 -> ../../ttyACM0

[21:15:12] openhabian@openhab:~$ test -w /dev/ttyACM0 && echo success || echo failure
success

[21:15:37] openhabian@openhab:~$ sudo apt-get purge modemmanager
[sudo] password for openhabian:
Reading package lists... Done
Building dependency tree
Reading state information... Done
Package 'modemmanager' is not installed, so not removed

Koenkk, 2020-03-03:
@mibas74 so when switching back to zigbee2mqtt 1.10.0 it works again?

CyberianD, 2020-03-03:
I have the same problem.

The adapter (CC2530) communication is success.

```
Mar 03 21:23:11 Zigbee2MQTT npm[1550]: zigbee2mqtt:info  2020-03-03 21:23:11: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-03-03.21-23-11' filename: log.txt
Mar 03 21:23:11 Zigbee2MQTT npm[1550]: zigbee2mqtt:info  2020-03-03 21:23:11: Starting zigbee2mqtt version 1.10.0-dev (commit #c25c32f)
Mar 03 21:23:11 Zigbee2MQTT npm[1550]: zigbee2mqtt:info  2020-03-03 21:23:11: Starting zigbee-herdsman...
Mar 03 21:23:18 Zigbee2MQTT npm[1550]: zigbee2mqtt:error 2020-03-03 21:23:18: Error while starting zigbee-herdsman
Mar 03 21:23:18 Zigbee2MQTT npm[1550]: zigbee2mqtt:error 2020-03-03 21:23:18: Failed to start zigbee
Mar 03 21:23:18 Zigbee2MQTT npm[1550]: zigbee2mqtt:error 2020-03-03 21:23:18: Exiting...
Mar 03 21:23:18 Zigbee2MQTT npm[1550]: zigbee2mqtt:error 2020-03-03 21:23:18: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
Mar 03 21:23:18 Zigbee2MQTT npm[1550]:     at ZStackAdapter.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:72:23)
Mar 03 21:23:18 Zigbee2MQTT npm[1550]:     at Generator.throw (<anonymous>)
Mar 03 21:23:18 Zigbee2MQTT npm[1550]:     at rejected (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:6:65)
Mar 03 21:23:18 Zigbee2MQTT npm[1550]: npm ERR! code ELIFECYCLE
Mar 03 21:23:18 Zigbee2MQTT npm[1550]: npm ERR! errno 1
Mar 03 21:23:18 Zigbee2MQTT npm[1550]: npm ERR! zigbee2mqtt@1.10.0-dev start: `node index.js`
Mar 03 21:23:18 Zigbee2MQTT npm[1550]: npm ERR! Exit status 1
Mar 03 21:23:18 Zigbee2MQTT npm[1550]: npm ERR!
Mar 03 21:23:18 Zigbee2MQTT npm[1550]: npm ERR! Failed at the zigbee2mqtt@1.10.0-dev start script.
Mar 03 21:23:18 Zigbee2MQTT npm[1550]: npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
```

```
My configuration.yaml

homeassistant: false
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://192.168.178.95'
  user: ****
  password: ****
serial:
  port: /dev/ttyUSB0
  baudrate: 115200
```

mibas74, 2020-03-03:
> 
> 
> @mibas74 so when switching back to zigbee2mqtt 1.10.0 it works again?

could You provide a link how to downgrade to one of previous versions please?

mibas74, 2020-03-03:
Hi,
i've reflashed CC2531 USB dongle with latest coordinator source routing firmware and it seems that it started properly and works OK.

RuvdH, 2020-03-05:
I thought it was me. I had to do a reinstall of my pi and after the new installation I run into the same issues. I'm not sure what version I ran before my crash of my SD card, but I haven't updated it in a while. I haven't flashed my dongle yet with the latest version so if there is anything I can do to test please let me know.

hoedha12, 2020-03-07:
I was having the very same issue with CC2530 and external antena after updating from 1.4 (very old, I know).. Here are the things I've done, I don't know which one solved the problem : 
1. uninstall modemManager
2. unplug and plugging the stick again
3. rebooting the pc
4. making sure rtscts is set to false
5. making sure disable_led entry is deleted

jorgensmit, 2020-03-08:
I updated the CC2531 USB stick and it's working again.

RuvdH, 2020-03-10:
I reflashed the stick and tried it again with success. This is odd as I had the same version on the stick already. But at least it's working again.

sergeylanzman, 2020-03-10:
maybe it's this https://github.com/home-assistant/supervisor/issues/1556

RadianceNL, 2020-03-11:
Maybe a silly suggestion. I just received my CC26x2R today and put in the a 3d printed enclosure and had the same error. I just figured out that the reset button was pressed by the enclosure. Snapped the part of that pushes against the reset button and it's running now.

ayukhno, 2020-03-11:
I have the same problem wiht cc2538 on 1.11.0 ver of z2m.
i can't reflash my stick. what i can do?
thnx

ayukhno, 2020-03-11:
i foung some warning in "hassio_supervisor" container - might be it could help: 
20-03-11 21:31:43 WARNING (MainThread) [supervisor.store.data] Can't read /data/addons/git/7ad98f9c/zigbee2mqtt-edge/config.json: does not match regular expression for dictionary value @ data['schema']['experimental']['transmit_power']. Got 'int(-22,19)?'

vdkeybus, 2020-03-12:
Yesterday, my Zigbee network was working fine and suddenly failed completely. I got the same problem (cannot set dtr/rts in dmesg, Error: SRSP - SYS - ping after 6s). Power cycling / restarting z2m had no effect.

Short story: I had to reflash the CC2531 to get it working again. I don't know which version was on the stick (bought it pre-programmed), but I believe I wrote the same version 20190619.

However, I'm 100% sure it happened exactly when I touched an Ikea ICTC-G-1 dimmer.

gombal88, 2020-03-14:
I had the same error. I'm running docker version of zigbee2mqtt and I was trying to reflash stick but no effect.

vdkeybus, 2020-03-14:
The green light also turned off when my CC2531 was still failing. I retried at least 5 times (power on/off, press button, reload serial driver, restart z2m).

After I re-flashed using cc-tool it worked immediately again on first attempt.

keniji, 2020-03-17:
I got a CC26X2R1 yesterday and have this issue with this new adapter. I made a frash Raspbian for my 3B+ from:
[Raspbian Buster Lite](https://downloads.raspberrypi.org/raspbian_lite_latest)
And then follow the instruction to start ZigBee2mqtt but still have the same issue. Here is the log with zigbee-herdsman debug mode enabled:
`> zigbee2mqtt@1.11.0 start /opt/zigbee2mqtt
> node index.js

zigbee2mqtt:info  2020-03-17 02:10:32: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-03-17.02-10-32' filename: log.txt
zigbee2mqtt:debug 2020-03-17 02:10:32: Can't load state from file /opt/zigbee2mqtt/data/state.json (doesn't exist)
zigbee2mqtt:info  2020-03-17 02:10:32: Starting zigbee2mqtt version 1.11.0 (commit #e140cf5)
zigbee2mqtt:info  2020-03-17 02:10:32: Starting zigbee-herdsman...
zigbee2mqtt:debug 2020-03-17 02:10:32: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM0"}}'
  zigbee-herdsman:adapter Path '/dev/ttyACM0' is valid for 'ZStackAdapter' +0ms
  zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM0"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json"}' +0ms
  zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with /dev/ttyACM0 and {"baudRate":115200,"rtscts":true,"autoOpen":false} +0ms
  zigbee-herdsman:adapter:zStack:znp:log Serialport opened +8ms
  zigbee-herdsman:adapter:zStack:znp:log Writing skip bootloader payload +1ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> buffer [239] +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +0ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +1s
zigbee2mqtt:error 2020-03-17 02:10:39: Error while starting zigbee-herdsman
zigbee2mqtt:error 2020-03-17 02:10:40: Failed to start zigbee
zigbee2mqtt:error 2020-03-17 02:10:40: Exiting...
zigbee2mqtt:error 2020-03-17 02:10:40: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:72:23)
    at Generator.throw (<anonymous>)
    at rejected (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:6:65)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.11.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.11.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2020-03-17T02_10_40_066Z-debug.log
`
Port checked and it should be alright:
`pi@raspberrypi:/opt/zigbee2mqtt/data $ ls -l /dev/serial/by-id
total 0
lrwxrwxrwx 1 root root 13 Mar 16 17:11 usb-Texas_Instruments_XDS110__03.00.00.07__Embed_with_CMSIS-DAP_L1100115-if00 -> ../../ttyACM0
lrwxrwxrwx 1 root root 13 Mar 16 17:11 usb-Texas_Instruments_XDS110__03.00.00.07__Embed_with_CMSIS-DAP_L1100115-if03 -> ../../ttyACM1`
And the write access to the port has been verified too:
`pi@raspberrypi:/opt/zigbee2mqtt/data $ test -w /dev/ttyACM0 && echo success || echo failure
success
`
I also tried version 1.10 but still got the same issue... BTW I use UNIFLASH to flash this firmware many times:
[CC26X2R1_20191106.zip](https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_3.x.0/bin/CC26X2R1_20191106.zip?raw=true)
And here is my configuration:
`homeassistant: false
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://ADDRESS:1883'
  user: USER
  password: PASSWORD

serial:
  port: /dev/ttyACM0`

nldroid, 2020-03-17:
> Is there anything else I can try? Many thanks!

You can try with: 
`port: null 
`
to let zigbee2mqtt auto detect the port

keniji, 2020-03-17:
> > Is there anything else I can try? Many thanks!
> 
> You can try with:
> `port: null `
> to let zigbee2mqtt auto detect the port

Thanks, I tried this but still getting the same error...

And I tried to change the panid, channel and set rtscts to "false", none of them work:
```
zigbee2mqtt:debug 2020-03-17 07:24:07: Using zigbee-herdsman with settings: '{"network":{"panID":6755,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[12],"networkKey":"HIDDEN"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":false,"path":null}}'
```

keniji, 2020-03-17:
Below is the debug log with path set to "null":
```
pi@raspberrypi:/opt/zigbee2mqtt $ DEBUG=* npm start

> zigbee2mqtt@1.11.0 start /opt/zigbee2mqtt
> node index.js

winston:create-logger Define prototype method for "error"
winston:create-logger Define prototype method for "warn"
winston:create-logger Define prototype method for "info"
winston:create-logger Define prototype method for "http"
winston:create-logger Define prototype method for "verbose"
winston:create-logger Define prototype method for "debug"
winston:create-logger Define prototype method for "silly"
winston:create-logger Define prototype method for "error"
winston:create-logger Define prototype method for "warn"
winston:create-logger Define prototype method for "info"
winston:create-logger Define prototype method for "http"
winston:create-logger Define prototype method for "verbose"
winston:create-logger Define prototype method for "debug"
winston:create-logger Define prototype method for "silly"
zigbee2mqtt:info  2020-03-17 07:43:15: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-03-17.07-43-15' filename: log.txt
winston:file written true false
zigbee2mqtt:debug 2020-03-17 07:43:15: Can't load state from file /opt/zigbee2mqtt/data/state.json (doesn't exist)
winston:file written true false
winston:file logged 127 info  2020-03-17 07:43:15: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-03-17.07-43-15' filename: log.txt

winston:file logged 230 debug 2020-03-17 07:43:15: Can't load state from file /opt/zigbee2mqtt/data/state.json (doesn't exist)

winston:file ENOENT ok /opt/zigbee2mqtt/data/log/2020-03-17.07-43-15/log.txt
winston:file stat done: log.txt { size: 0 }
winston:file create stream start /opt/zigbee2mqtt/data/log/2020-03-17.07-43-15/log.txt { flags: 'a' }
winston:file create stream ok /opt/zigbee2mqtt/data/log/2020-03-17.07-43-15/log.txt
winston:file file open ok /opt/zigbee2mqtt/data/log/2020-03-17.07-43-15/log.txt
zigbee2mqtt:info  2020-03-17 07:43:16: Starting zigbee2mqtt version 1.11.0 (commit #e140cf5)
winston:file written true false
zigbee2mqtt:info  2020-03-17 07:43:16: Starting zigbee-herdsman...
winston:file written true false
zigbee2mqtt:debug 2020-03-17 07:43:16: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":null}}'
winston:file written true false
  serialport/bindings loading LinuxBinding +0ms
  zigbee-herdsman:adapter No path provided, auto detecting path +0ms
  serialport/stream .list +0ms
winston:file logged 81 info  2020-03-17 07:43:16: Starting zigbee2mqtt version 1.11.0 (commit #e140cf5)

winston:file logged 136 info  2020-03-17 07:43:16: Starting zigbee-herdsman...

winston:file logged 555 debug 2020-03-17 07:43:16: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":null}}'

  zigbee-herdsman:adapter Auto detected path '/dev/ttyACM0' from adapter 'ZStackAdapter' +244ms
  zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM0"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json"}' +0ms
  zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with /dev/ttyACM0 and {"baudRate":115200,"rtscts":true,"autoOpen":false} +0ms
  serialport/stream opening path: /dev/ttyACM0 +251ms
  serialport/binding-abstract open +0ms
  serialport/stream _read queueing _read for after open +2ms
  serialport/bindings/poller Creating poller +0ms
  serialport/stream opened path: /dev/ttyACM0 +3ms
  serialport/stream _read reading { start: 0, toRead: 65536 } +1ms
  serialport/binding-abstract read +7ms
  zigbee-herdsman:adapter:zStack:znp:log Serialport opened +11ms
  zigbee-herdsman:adapter:zStack:znp:log Writing skip bootloader payload +1ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> buffer [239] +0ms
  serialport/stream _write 1 bytes of data +3ms
  serialport/binding-abstract write 1 bytes +3ms
  serialport/bindings/unixRead Starting read +0ms
  serialport/bindings/unixWrite Starting write 1 bytes offset 0 bytesToWrite 1 +0ms
  serialport/bindings/unixRead read error [Error: EAGAIN: resource temporarily unavailable, read] {
  errno: -11,
  code: 'EAGAIN',
  syscall: 'read'
[... 11 lines cut ...]
  serialport/bindings/unixWrite write returned: wrote 5 bytes +1ms
  serialport/bindings/unixWrite Finished writing 5 bytes +1ms
  serialport/stream binding.write write finished +4ms
zigbee2mqtt:error 2020-03-17 07:43:23: Error while starting zigbee-herdsman
winston:file written true false
zigbee2mqtt:error 2020-03-17 07:43:23: Failed to start zigbee
[... cut ...]
    at rejected (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:6:65)
winston:file written true false
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.11.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.11.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2020-03-17T07_43_23_552Z-debug.log
```

hughb8on, 2020-03-17:
I was having the same problem today following a soft reboot to apply updates to Ubuntu.
Unplugging and replugging my CC26X2R1 fixed it. This is on 1.11.
I can see nothing in the logs; it seems as far as linux is concerned the adapter was available the whole time.

keniji, 2020-03-17:
> I was having the same problem today following a soft reboot to apply updates to Ubuntu.
> Unplugging and replugging my CC26X2R1 fixed it. This is on 1.11.
> I can see nothing in the logs; it seems as far as linux is concerned the adapter was available the whole time.

Thanks hughb8on, but I think I have already unplug&replug the board but it's all in vain. I also tried it on Windows and the error is the same.
And I have just verified the firmware in UniFlash and the verfication is success:
```
2020/3/17 下午10:07:12] [SUCCESS] Program Load completed successfully.
[2020/3/17 下午10:07:46] [INFO] Cortex_M3_0: GEL Output: Memory Map Initialization Complete.
[2020/3/17 下午10:07:49] [INFO] Cortex_M4_0: GEL Output: Memory Map Initialization Complete.
[2020/3/17 下午10:07:56] [SUCCESS] Cortex_M4_0: Program verification successful for C:/Users/keniji/Desktop/CC26X2R1_20191106.hex
```
I am thinking may be I should submit an issue under zigbee-herdsman or Z-Stack-firmware instead?

Koenkk, 2020-03-17:
@keniji what CC26X2R1  revision do you have? (you will find it on the white sticker on the back of the board)

keniji, 2020-03-17:
> @keniji what CC26X2R1 revision do you have? (you will find it on the white sticker on the back of the board)

Hi Koenkk, the board shows Rev 1.0.1, but in UniFlash it shows Revision: C (1.1)

Koenkk, 2020-03-17:
@keniji thats the wrong revision (https://www.zigbee2mqtt.io/information/FAQ.html#cc26xr1-launchpad-coordinators-only-verify-that-you-have-the-correct-revision)

keniji, 2020-03-18:
> @keniji thats the wrong revision (https://www.zigbee2mqtt.io/information/FAQ.html#cc26xr1-launchpad-coordinators-only-verify-that-you-have-the-correct-revision)

Hi @Koenkk , is this a new FAQ? I seem don't remember have seen this part before... lol

Koenkk, 2020-03-18:
Yes I added it after your comment

keniji, 2020-03-18:
> Yes I added it after your comment

Well, hope this will help the others to identify a similar issue ;)

sirs2k, 2020-03-20:
I'm having the same problem since updating from 1.05.0 I always have to re-plug the usb for it to work. 
The port is correct and I even ran `test -w /dev/ttyACM0 && echo success || echo failure` and it returned success.

Can someone please tell me what firmware to flash on my CC2531?
It's currently on DEFAULT_20190608

vdkeybus, 2020-03-20:
I had some version on the 2531 that was on it when I bought it.

I reflashed it with CC2531_SOURCE_ROUTING_20190619.zip

But if you have to re-plugin I would look at the port permissions before
and after the re-plugging. You're doing the 'test' with the same user that
runs z2m ? Could be an udev issue

Also, you could be low on power (current) during initial config (many
devices being configured in short succession) and when re-plugging, this
does not occur ? Perhaps test with a powered hub ? Just some thoughts.

J.

sirs2k, 2020-03-20:
> Also, you could be low on power (current) during initial config (many devices being configured in short succession) and when re-plugging, this does not occur ? Perhaps test with a powered hub ? Just some thoughts.

Thank you, great point. Forgive my ignorance, but how can I power the stick externally when the usb has to be connected to the Pi?

vdkeybus, 2020-03-20:
Get a powered hub, i.e. one with an external power supply connection. Plug
the CC2531 in one of its ports, connect the hub to the Pi. Keep the hub
powered while you power cycle the Pi.

henkiejan1, 2020-03-29:
Hmm is has the same issue with the CC2531. But what i discover is that when i just plug it in my laptop. Did not flash or anything en plug it in back it works! So maybe is the short workarround to plug the stick a short time in your laptop and then back in your pi.

m-vanraamsdonk, 2020-03-31:
I've got the same issue, tried all suggested solutions.. No luck.. Tried to re-flash but the CC Debugger will not detect the device (red led). Tried a spare CC2531.. same thing.. Might be the debugger..

joaomruas, 2020-04-09:
Having similar problem to users in this thread.
 
Other users are reporting similar problem in Home assistant plugin. But didn't show herdsman debug logging so it's hard to tell
https://github.com/danielwelch/hassio-zigbee2mqtt/issues/267 
And one user in #3148 had similar log as me.

Current setup: Raspberry Pi 3 home assistant plugin. 

I'm going to install zigbee2mqtt without home assistant and check if error persists using either CC2530 or CC2531.

- CC2531(normal) CC2531_DEFAULT_20190608, Problem occurring,No by-pass
- CC2530(normal) CC2530_DEFAULT_20190608 No problems
- CC2530(normal) CC2530_SOURCE_ROUTING_20190619, Problem occurring, re-plugging usb corrects problem but after some hours it happens again.

On first look it seems hardware dependent but can only confirm after reinstalling without home assistant.

## Log Dump:
-----------------------------------
> node index.js
2020-04-09T00:08:58.878Z zigbee-herdsman:adapter Failed to validate path: 'Error: spawn udevadm ENOENT'
2020-04-09T00:08:58.888Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[143,7,174,14,37,151,164,186,98,10,228,120,21,87,40,249],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[25]},"serialPort":{"baudRate":115200,"rtscts":false,"path":"/dev/ttyUSB0"},"databasePath":"/share/zigbee2mqtt/database.db","databaseBackupPath":"/share/zigbee2mqtt/database.db.backup","backupPath":"/share/zigbee2mqtt/coordinator_backup.json"}'
2020-04-09T00:08:58.907Z zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with /dev/ttyUSB0 and {"baudRate":115200,"rtscts":false,"autoOpen":false}
2020-04-09T00:08:58.929Z zigbee-herdsman:adapter:zStack:znp:log Serialport opened
2020-04-09T00:08:58.930Z zigbee-herdsman:adapter:zStack:znp:log Writing skip bootloader payload
2020-04-09T00:08:58.931Z zigbee-herdsman:adapter:zStack:unpi:writer --> buffer [239]
2020-04-09T00:08:59.950Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}
2020-04-09T00:08:59.964Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]
zigbee2mqtt:error 2020-04-09 01:09:05: Error while starting zigbee-herdsman
zigbee2mqtt:error 2020-04-09 01:09:05: Failed to start zigbee
zigbee2mqtt:error 2020-04-09 01:09:05: Exiting...
zigbee2mqtt:error 2020-04-09 01:09:06: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (/zigbee2mqtt-1.12.0/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:71:23)
    at Generator.throw (<anonymous>)
    at rejected (/zigbee2mqtt-1.12.0/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:6:65)
npm
 ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.12.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.12.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

malt3, 2020-04-13:
For me the issue seems to have started after running the raspberry pi 3 from a power bank with only 1A. Is it possible that the under-voltage of the CC2531 could be responsible for errors in the device firmware requiring a reflash of the firmware?

Koenkk, 2020-04-13:
Recommendation for pi 3 is 2.5 amps, anything below can give strange behavior so I guess yes (I remember users previously solving issues by using a decent power supply). https://www.raspberrypi.org/documentation/hardware/raspberrypi/power/README.md

jd1900, 2020-04-14:
I've just had the same problem in CC26X2R1 / CC2652R. I've tried unplugging / re-plugging the CC26X2R1 with no luck. It was working yesterday, but this morning I saw it was not working at all.

I had to reboot the whole Raspberry Pi, and it worked again.

Just in case it can help debugging the error.

harisosmancevic, 2020-04-15:
I am having the same issue with a CC26X2R1. HW Rev: B. Worked for 2-3 months until 6-7 days ago...

zigbee2mqtt:info  2020-04-15 10:53:57: Logging to console and directory: '/share/zigbee2mqtt/log/2020-04-15.10-53-57' filename: log.txt
zigbee2mqtt:info  2020-04-15 10:53:58: Starting zigbee2mqtt version 1.12.2 (commit #unknown)
zigbee2mqtt:info  2020-04-15 10:53:58: Starting zigbee-herdsman...
zigbee2mqtt:error 2020-04-15 10:54:05: Error while starting zigbee-herdsman
zigbee2mqtt:error 2020-04-15 10:54:05: Failed to start zigbee
zigbee2mqtt:error 2020-04-15 10:54:05: Exiting...
zigbee2mqtt:error 2020-04-15 10:54:05: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (/zigbee2mqtt-1.12.2/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:71:23)
    at Generator.throw (<anonymous>)
    at rejected (/zigbee2mqtt-1.12.2/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:6:65)
npm 
ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.12.2 start: `node index.js`
npm
 ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.12.2 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-04-15T08_54_05_891Z-debug.log
2020-04-15T10:54:08: PM2 log: App [npm:0] exited with code [1] via signal [SIGINT]
2020-04-15T10:54:08: PM2 log: App [npm:0] starting in -fork mode-
2020-04-15T10:54:08: PM2 log: App [npm:0] online
> zigbee2mqtt@1.12.2 start /zigbee2mqtt-1.12.2
> node index.js

exetico, 2020-04-15:
I have one devices, and it's working just fine. 

However, another contact of mine, can't get it to work. I've tried everything, but it's the same story over and over again.

I'ts tried both the latest version, the version i'm running, and a few others.

The devices has been flashed a few times, just to make sure that evertything is OK.

Permissions is like it should be. They are working...

```
zigbee2mqtt:info  2020-04-15 18:50:07: Logging to console and directory: '/app/data/log/2020-04-15.18-50-07' filename: log.txt
zigbee2mqtt:info  2020-04-15 18:50:07: Starting zigbee2mqtt version 1.12.0 (commit #840b9d9)
zigbee2mqtt:info  2020-04-15 18:50:07: Starting zigbee-herdsman...
zigbee2mqtt:error 2020-04-15 18:50:15: Error while starting zigbee-herdsman
zigbee2mqtt:error 2020-04-15 18:50:15: Failed to start zigbee
zigbee2mqtt:error 2020-04-15 18:50:15: Exiting...
zigbee2mqtt:error 2020-04-15 18:50:15: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:71:23)
    at Generator.throw (<anonymous>)
    at rejected (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:6:65)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.12.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.12.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-04-15T18_50_15_370Z-debug.log
```
Here's a example with 1.12.0.

Anyone?It's the CC26X2R1, and it's the current revision, too. We've also followed the flashing-guide step by step (and did it, the same way, as with mine).

Both machines are Debian, behind ESXi. There's no other USB-devices in the machines.

TL;DR: Mine are working, but his are stuck in this status.. I've tried almost everything. I can't ship it to him, right away - so, we can't compare like that, but i'd like to do it...

Did anyone else, manage to get this fixed?

exetico, 2020-04-16:
So, the device was moved to an backplane connector, instead of a front connector.

And, it worked - for some reason. Here is the two dmesg-outputs, where it's working - and it's not.
Didn't work:
```
[ 2187.947145] usb 1-1: new high-speed USB device number 3 using ehci-pci
[ 2188.528281] usb 1-1: New USB device found, idVendor=0451, idProduct=bef3, bcdDevice= 1.00
[ 2188.528284] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2188.528285] usb 1-1: Product: XDS110 (03.00.00.07) Embed with CMSIS-DAP
[ 2188.528286] usb 1-1: Manufacturer: Texas Instruments
[ 2188.528287] usb 1-1: SerialNumber: L1100GNO
[ 2188.557412] cdc_acm 1-1:1.0: ttyACM0: USB ACM device
[ 2188.658466] cdc_acm 1-1:1.3: ttyACM1: USB ACM device
[ 2188.764488] hid-generic 0003:0451:BEF3.0003: hiddev0,hidraw0: USB HID v1.11 Device [Texas Instruments XDS110 (03.00.00.07) Embed with CMSIS-DAP] on usb-0000:02:03.0-1/input5
```

Worked like it should:
```
[49832.479142] usb 2-1: new high-speed USB device number 2 using ehci-pci
[49833.056558] usb 2-1: New USB device found, idVendor=0451, idProduct=bef3, bcdDevice= 1.00
[49833.056560] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[49833.056561] usb 2-1: Product: XDS110 (03.00.00.07) Embed with CMSIS-DAP
[49833.056562] usb 2-1: Manufacturer: Texas Instruments
[49833.056563] usb 2-1: SerialNumber: L1100GNO
[49833.079335] cdc_acm 2-1:1.0: ttyACM0: USB ACM device
[49833.182585] cdc_acm 2-1:1.3: ttyACM1: USB ACM device
[49833.290458] hid-generic 0003:0451:BEF3.0006: hiddev0,hidraw1: USB HID v1.11 Device [Texas Instruments XDS110 (03.00.00.07) Embed with CMSIS-DAP] on usb-0000:02:03.0-1/input5
```
So, we tried to move it back to the frontpanel, and the device failed again. Hereafter, to the backpanel, again. Same result. Failed again.

Now we:
1) Removed USB Controller and USB Device in ESXi
2) Stopped the container, and removed it
3) Powered down the VM, with `sudo poweroff`
4) Removed the cable from the device
5) Waited a few seounds
6) Plugged the cable back into the device
7) Added USB Controller and USB Device in ESXi
8) Turned on the VM
9) Created the container, again

This time, it worked like it should.

I've also tried to stop the container, remove it, and recreate it. It also worked, like should.

So, for us, it must be something with how the device is handling the pass-through to our host VM, or, maybe just how it's power cycled, or so. It's hard to say.

However, now we know the steps, to get it up and running. I would like to try and remove steps, until we found our, what's actually fixing the issues, but i think it's isolated to power cycle, and the host VM. Cause we removed and added the device multiple times in ESXi yesterday, with no luck.

notlistening, 2020-05-06:
So also been seeing this problem, with two different CC2531 boards. When this happens you can also see when the firmware is loading that the device disconnects. At this point you get various different behaviours. 

dmesg when running the zigbee2mqtt docker image on a corrupted stick:
[206335.875770] usb 5-5: new full-speed USB device number 2 using ohci-pci
[206336.078801] usb 5-5: New USB device found, idVendor=0451, idProduct=16a8
[206336.078804] usb 5-5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[206336.078805] usb 5-5: Product: TI CC2531 USB CDC
[206336.078806] usb 5-5: Manufacturer: Texas Instruments
[206336.078807] usb 5-5: SerialNumber: __0X00124B00014D8F93
[206336.545673] cdc_acm 5-5:1.0: ttyACM0: USB ACM device
[206336.547013] usbcore: registered new interface driver cdc_acm
[206336.547014] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters
[206401.046661] usb 5-5: USB disconnect, device number 2
[206401.647682] usb 5-5: new full-speed USB device number 3 using ohci-pci
[206417.123670] usb 5-5: device descriptor read/64, error -110
[206432.739647] usb 5-5: device descriptor read/64, error -110

So the information being returned by the firmware to identify the device is bad. This make me think there is some sort of corruption happening on the stick causing the USB enumeration to fail after the corruption. This does not seem OS dependant as I have checked this across Alpine/Ubuntu/HAS.io also I have run it with 9 IKEA GU10s and found this problem and then again with 20 devices. Each time the 'corruption' happens at a random time during normal operation. It might be interesting if a developer could inspect a device after this corruption had occurred so see 'what damage' is happening and where. Then maybe add some breakpoints and see when that part of the flash is written to. This will help find which part of the code is responsible. I can quite reliably reproduce this here but just waiting for a debugger.

lukyjay, 2020-05-09:
> So the information being returned by the firmware to identify the device is bad. This make me think there is some sort of corruption happening on the stick causing the USB enumeration to fail after [...]

For me this happens almost any time I reboot my machine. I would be happy to supply any information or debugging if someone can instruct me on what to do.

I wonder if this issue is specific CC2531

dedors, 2020-05-14:
I'm using a CC26X2R1. Hass.io running as VM on Unraid.

Replug -> didn't help
Restarting VM -> didn't help
Power down Host (Unraid) and replug -> worked

lukyjay, 2020-05-17:
I got a new adapter and it was working for a day or so then this happened again.

I noticed the host (proxmox) outputs this error each time the problem first occurs, then afterward the dongle will not work:
cp210x ttyUSB0: failed set request 0x7 status: -32

I reflashed the firmware then got the same error above again and it still wont work. This is extremely frustrating.

sermayoral, 2020-05-20:
Hi. I got the zig-a-zig-ah! (zzh) new adapter. Hass.io running in a Raspberry Pi 3 B+ and zigbee2mqtt v1.13.0: 

Replug -> didn't help
Restarting Home Assistant -> didn't help
Rebooting Raspberry Pi 3 B+ -> in process

blakadder, 2020-05-20:
Is it stable after reboots?

dedors, 2020-05-20:
I had this issue only once in 5 months and since the reboot 7 days ago it's stable again.

ginodp, 2020-05-22:
Same problem here.
It was working like a champ for a while but never flawlessly. But now it doesn't work at all. I updated all and after the updates I got this problem.

And it has to be after one of the updates because it was working before and suddenly we all have this problem. 

I love home assistant but I think it's time to switch back to old school again more reliable then ha there seems that there is always something wrong with it.

It rather pay for something that works then something that keeps me in the dark most of the times.

lukyjay, 2020-05-25:
Found a solution that seems to be working for me. I've managed to successfully restart around 5~ times with no issues.

Use a Raspberry Pi and direct connect the CC2530 to the GPIO headers on the Pi. The zigbee2mqtt website has instructions on connecting. I also ran zigbee2mqtt using the bare metal instructions since the pi _only_ runs zigbee2mqtt (I run everything else in a Proxmox Server).

I assume the issue is related to USB. Maybe I've been lucky and this hasn't fixed the issue, but I hope this helps someone.

monotonus, 2020-05-28:
I have a similar behavior with proxmox and a linux container (LXC). The stick is working for some days and then no more data is received. If I then restart zigbee2mqtt, I also get this error. It also just helps to powercycle the complete proxmox host, even if I have write access according to the documentation.

root@homeassistant:~# test -w /dev/ttyACM0 && echo success || echo failure
success

dmesg shows this:

[ 6609.294201] cdc_acm 2-3:1.0: failed to set dtr/rts
[ 6652.046431] cdc_acm 2-3:1.0: failed to set dtr/rts
[ 6658.446565] cdc_acm 2-3:1.0: failed to set dtr/rts
root@proxmox:~# dmesg | grep Texas
[    1.707136] usb 2-3: Manufacturer: Texas Instruments

I had this issue this evening and before restarting I exchanged the CC2531 to a spare one with the same coordinator version. But now after restarting z2m several times, it stopped working again. So my approch to restart z2m regularly whith a cronjob wouldn't help much, I guess.

z2m version 1.13.1
coordinator 20190608

planetar, 2020-05-29:
I saw the same problem here, after months of smooth operation zigbee2mqtt turned inactive. dmesg was full of 
```
[46839.546643] cdc_acm 1-1.3:1.0: failed to set dtr/rts
[46844.383366] cdc_acm 1-1.3:1.0: failed to set dtr/rts
```
and the log was like 

```
zigbee2mqtt:error 2020-05-29 13:01:59: Error while starting zigbee-herdsman
zigbee2mqtt:error 2020-05-29 13:01:59: Failed to start zigbee
zigbee2mqtt:error 2020-05-29 13:01:59: Exiting...
zigbee2mqtt:error 2020-05-29 13:01:59: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:72:23)
    at Generator.throw (<anonymous>)
    at rejected (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:6:65)
```

Testing the port ended with success. Taking the stick out and reinsert it again didnt work, systemctl restart didnt do it, rebooting the raspberry host had no success. 
So I followed the advice I found here and reflashed the stick with the firmware it already had. CC2531_DEFAULT_20190608 
(I found out that Dupont cables fit on the debug port once I took off the plastic isolation and isolated the cables with paper strips.) The flasher counted up to 512 and signalled success and indeed dmesg shows none of the previous errors and zigbee2mqtt starts.

Unfortunately, now every action results in 

`Error: Data request failed with error: 'No network route' (205)`
Does it mean the reflashing failed?

Koenkk, 2020-05-29:
@planetar in case of stability issue I would recommend switching to the source routing firmware. https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing

A no network route after a reflash can most of the times resolved by power cycling that specific device.

planetar, 2020-05-29:
Thanks a lot Koen, 
power cycling mixed with some patience on the grounds of your good advice and software seems to have brought them up one by one. :-)

boesing, 2020-06-04:
I am experiencing this issue after upgrading homeassistant OS from 3.13 to 4.8 (now 4.9).
I had to replug the stick (CC2531) sometimes but now, I cannot make it work with 4.8+...

Using latest stable of zigbee2mqtt `1.13.1.1`. Sadly, I dont get how to rollback to 3.13 to verify if its an homeassistant OS issue or not.

[115 more replies]

closed 2020-09-15 by github-actions[bot]