Koenkk/zigbee2mqtt #2998: Pairing doesn't work with latest-dev

cjtenny, 2020-02-25:
# Bug Report
## What happened
Even with permit_join = true, no new devices join or appear in logs.

Running docker build https://hub.docker.com/layers/koenkk/zigbee2mqtt/latest-dev/images/sha256-c1f01dc740a3528b62462d3f8897f273d9277d23b92980ff4594ae441b4bff83?context=explore

Downgrading to zigbee2mqtt:1.9.0 docker image with same configuration allows pairing.

## What did you expect to happen
New devices in pairing mode can join the zigbee coordinator

## How to reproduce it (minimal and precise)
Attempt to pair new device with docker image latest-dev as of today.

## Debug Info
zigbee2mqtt version: latest-dev, see SHA above
CC253X firmware version: 20191106 commit

sjorge, 2020-02-25:
I have this too, the following steps fix it for me:
- stop z2m
- unplug coordinator
- wait 10 seconds
- plug in coordinator
- start z2m

ellavas, 2020-02-25:
I had this issue also. Unplug/Plug the coordinator helps.

Koenkk, 2020-02-26:
@sjorge thanks, added this to the FAQ!

sjorge, 2020-02-26:
It seems after about ~12h of uptime I need to replug to get pairing to work... I wonder what changed.

closed 2020-05-03 by stale[bot]