Koenkk/zigbee2mqtt #2999: Duplicate entry in database.db

claudio-petrini, 2020-02-25:
Hello,
when pairing a Philips HUE model 9290012573A two entries are inserted in the database.db file:
- one has the key `modelId` and has `"type": "Router"`
- one doesn't have the key `modelId` and it's an `EndDevice`.

When trying to control the bulb by publishing to `zigbee2mqtt/${friendlyName}/set` we've noticed that only the entry with `modelId` can be controlled while the other generates the following warning:

```
Device with modelID 'undefined' is not supported.
```

Is correct the assumption that we can control only the entry with the `modelId` key defined?

Thank you

Koenkk, 2020-02-25:
I don't think it's the bulb but another device, could you share your `database.db`?

claudio-petrini, 2020-02-26:
Hello @Koenkk, you were right there was another device that was connected to the coordinator. 
Apologize, and thank you again for the assistance.

closed 2020-02-26 by claudio-petrini