Koenkk/zigbee2mqtt #3000: lumi.sen_ill.mgl01 has odd softwareBuildID

sjorge, 2020-02-25:
```json
{
  "ieeeAddr": "0x04cf8cdf3c771820",
  "type": "EndDevice",
  "networkAddress": 24216,
  "model": "GZCGQ01LM",
  "friendly_name": "0x04cf8cdf3c771820",
  "manufacturerID": 4718,
  "manufacturerName": "LUMI",
  "powerSource": "Battery",
  "modelID": "lumi.sen_ill.mgl01",
  "hardwareVersion": 1,
  "softwareBuildID": "2019\u0000www.",
  "dateCode": "20191118",
  "lastSeen": 1582650212229
}
```

The bsoftwareBuildID is very weird hier, not sure this is normal.
I just got this device today, I paired it twice but both times the same result.

Koenkk, 2020-02-26:
Sniffing reveals that this is indeed the value that's being send by the device. However wireshark seems to strip everything including the UTF8 null character.

[image]

sjorge, 2020-02-26:
:| seems like sort of bufferoverflow or something... '2019' would make some sense at a but the 'www.' after it is just weird.

I got a 2nd one on the way now that i know it works... but I guess closing for now as this is correct data.

closed 2020-02-26 by sjorge