Koenkk/zigbee2mqtt #3001: osram_remember_state not working - AC03641

M0ebiu5, 2020-02-25:
# Bug Report
## What happened
the Osram bulb turn on everytime after a power cycle
bulb type AC03641 - OSRAM LIGHTIFY LED Classic A60 clear

## What did you expect to happen
the bulb should stay off, if it is set to "off" before the power cycle
(and the "osram_remember_state" parameter is set to true)

## How to reproduce it (minimal and precise)
send the following json the the device:
`{
    "state": "OFF",
    "osram_remember_state": true
}`
cut the power to the bulb and reconnect -> bulb is glowing

## Debug Info
zigbee2mqtt version:` 1.10.0`

CC253X firmware version: 
`Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190619}}'`

M0ebiu5, 2020-03-05:
Could somebody please confirm or deny this behavior?

copystring, 2020-03-05:
I have this too. I set my osram light to off. when actually removing power from it and turning it back on, it turns on with brightness of 1 instead of 0

M0ebiu5, 2020-03-06:
@copystring : Thanks for your feedback!
So this problem is related to the bulb (AC03641) or it's firmware?
I wonder if it's working for anyone, cause if not, it should be mentioned on the "supported devices page"...

tam1m, 2020-03-13:
Same for me with AC03642

Koenkk, 2020-03-13:
I doubt if all Osram devices support this (it seems not)

copystring, 2020-03-15:
https://www.zigbee2mqtt.io/devices/4052899926158.html is mine. It does seem to support it. However it remembers brightness of 1 instead of 0 which makes the light turn on after powering it.
Setting it to 255/on is remembered correctly

sjorge, 2020-03-15:
@copystring isn't brightness 0 mapped to off in most firmwares? I remmeber my trådfri bulbs doing that atleast.

copystring, 2020-03-15:
This is what I think. brightness 0 = off.
So setting it to 0 and removing power from the bulb and then powering it again makes it have brightness 1.

M0ebiu5, 2020-03-15:
So maybe "osram_remember_brightness" would be a better name for the parameter...

If somebody is interested, i have a workaround (using tradfri gu10 spots): 

They also remember the transition time. So if you set the transition time to min. 3 seconds before cutting the power, it is possible to turn them off before they shine, if power is restored.
I had to enable "availability" in zigbee2mqtt and made a simple node-red flow, which sends the turn off command, as soon as a light goes "online".

Maybe this could also be done directly in zigbee2mqtt with some configuration settings?

copystring, 2020-03-21:
So I created this log and filtered it by the lamp with the issue described above.
First I made sure the light is in the OFF state with brightness of 0. Then removed power of it for 5 seconds and returned power. Light is then in the ON state with brightness of 1.

Making the light have a brightness of 250 is remembered correctly.

Here is the log. I added two lines where I turned the lights off and on after 5 seconds.

https://pastebin.com/N2w8vgJX

copystring, 2020-03-30:
@Koenkk would it be possible for you to have a look at this? Do you need more info?
It seems osram remember state works but not the on/off state.

Koenkk, 2020-03-30:
@copystring I cannot contribute much here ATM as I don't have Osram bulbs. To fix this sniffing should be done with the original gateway.

@qosmio since you originally implemented this, could you maybe help with this?

Nephiel, 2020-05-31:
Same here, AC03641 always turns back on after a power loss.

qosmio, 2020-05-31:
Sorry all, I've been away from my development machine for a while. I'll take a look at this in the next coming days.

qosmio, 2020-06-01:
OK, just read the bug report a little more closely. I'm not sure I actually ever tested with a brightness setting of "0" and have it retain that after power loss. Are you able to confirm if this is capable with the Osram supplied gateway? 

A little difficult to test right now as I don't have a spare Osram bulb to tinker with (all being used in groups and linked with HA). As others have stated, I do believe it's the default behavior of the Osram bulbs and many other bulbs to return to their brightness states only if > 0.

qosmio, 2020-06-01:
> So maybe "osram_remember_brightness" would be a better name for the parameter...

@M0ebiu5 : My RGBW BR30 and RGBW A19 bulbs both remember brightness, color_temp, and xy color attributes. I know "STATE" in regards to Z2M and lightbulbs is ON/OFF, but it's a pretty general term to encompass all the attributes (minus being ON/OFF in this instance).

> Edit: It also does not seem to remeber color_temp.

@copystring I'm not sure that applies, you said your lightbulb was this one? https://www.zigbee2mqtt.io/devices/4052899926158.html it's brightness and on/off only.

copystring, 2020-06-01:
@qosmio I have that one too but have not tried it with this one (like you said it does not support it)
I tried it with this one just now to confirm the problem: https://www.zigbee2mqtt.io/devices/AC03645.html
The problem persists.

qosmio, 2020-06-01:
@copystring ah ok that bulb wasn't mentioned. I'm not sure what I could do there, don't have that bulb and sadly isn't sold in the states (at least US Amazon). I will say though, I DO remember before moving these bulbs to Z2M I had to update the firmware of my A19s to utilize the "remember" feature. 

Have you already made sure you've updated all your Osram devices through the vendor gateway? I know Z2M supports OTA but it isn't 100% coverage ATM. And again, before confirming if it is in fact a bug, we'd need to know if the capability for it exists with vendor's gateway.

copystring, 2020-06-01:
Not sure if it is related to the bulb. None of my Osram bulbs remember the brightness.
Yes. I was one of the first people to try the new OTA feature and updated all my bulbs to the latest firmware.

Nephiel, 2020-06-02:
I also updated mine using the OTA feature. Can confirm it doesn't remember state nor brightness after a power loss. Always defaults to ON at max brightness at power on.

Also, for some reason it doesn't allow setting brightness to 255, max value seems to be 254.

qosmio, 2020-06-03:
I can't really help much in this regard without having access to the bulb or at the very least, a sniffer dump where it's paired to the factory gateway and you're able to confirm that it's able to remember its state with the manufacturer supplied gateway.

Again, I reverse engineered all of these based on sniffing the data specific to the A19, BR30, and Lightify RGBW strip. And has for the most part worked with the rest of Osram/LEDVANCE/Sylvania ecosystem.

closed 2020-07-26 by stale[bot]

SimpleUserHA, 2023-01-08:
I would like to reopen somehow this issues as I recently tried the option to set the default poweron behaviour for the AC03641 and received the following errors while trying the options  OFF / ON / TOGGLE / PREVIOUS:

```
pressing OFF
error 2023-01-08 21:19:05Publish 'set' 'power_on_behavior' to 'Stehlampe' failed: 'Error: Write 0x7cb03eaa0a040e09/3 genOnOff({"startUpOnOff":0}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'

pressing ON
error 2023-01-08 21:19:20Publish 'set' 'power_on_behavior' to 'Stehlampe' failed: 'Error: Write 0x7cb03eaa0a040e09/3 genOnOff({"startUpOnOff":1}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'

pressing TOGGLE
error 2023-01-08 21:19:22Publish 'set' 'power_on_behavior' to 'Stehlampe' failed: 'Error: Write 0x7cb03eaa0a040e09/3 genOnOff({"startUpOnOff":2}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'

pressing PREVIOUS
error 2023-01-08 21:19:26Publish 'set' 'power_on_behavior' to 'Stehlampe' failed: 'Error: Write 0x7cb03eaa0a040e09/3 genOnOff({"startUpOnOff":255}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
```

After doing some research I came across this issue here. So to be clear, I am not 100% sure which hardware versions are out there for this device, but it seems mine does not support the power on behaviour.
```
Firmware-Datum
    20140331CNEF****
Firmware-Version
    V1.05.10
```

However, after further investigation I came across the Deconz implementation:
1. via a request for ZHA / zigpy quirk in home assistant community:
https://community.home-assistant.io/t/osram-lightify-ledvance-smart-set-power-on-defaults-for-bulbs/355687
that linked to 

1. https://github.com/dresden-elektronik/deconz-rest-plugin/issues/102
which contains a bitcatcher log file and lots of explanations by @ebaauw 

1. deconz implemenation:
https://github.com/dresden-elektronik/deconz-rest-plugin/pull/1837/commits/dc83c121cf40e6d4331765f811623631c443b5d5

1. 2nd issue containing screenshots of the deconz gui:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1833

**Long story short**:  When executing the following command to the manufacturer specific cluster, I am able to write the **_current_ brightness as default values** to the device.   After the power cut, the device successfully recovers the values stored previously and does not go to full brightness anymore by default.
[image]

@qosmio & @Koenkk  are these information sufficient to 
a) implement the command so each user can access it easily via the gui / device config page ?
b) could the default power on option be disabled again / removed, as it seems it is not  working....  

Unfortunately I have no access to the OEM bridge and I am not able to sniff anything.
In case of any questions / details, maybe @ebaauw could help out or provide some details about the devices ?

jakvot, 2023-01-25:
The same here:
`
Zigbee2MQTT:info: Sxxxx (0x7cb03eaa00xxxxxx): 4058075816794
 OSRAM Smart+ Ceiling TW (Router)
Zigbee2MQTT:error: Publish 'set' 'power_on_behavior' to 'Sxxxx' failed: 'Error: Write 0x7cb03eaa00xxxxxx/3 genOnOff({"startUpOnOff":255}, {"sendWhen":
"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaul
tResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,
"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUT
E')'
`

maurizioandreotti60, 2023-07-25:
I have the same issue even using remember_state, the the bulb always glows.
The z2m documentation says the it should work, but it seems not the case.