Koenkk/zigbee2mqtt #3003: Schneider 2 Gang Dimmer Switch Problem

onewaylifestreet, 2020-02-26:
Hello,

I need to make my Schneider 2 Gang Dimmer Switch work on my hassio.
I went in the device.js and change the // ELKO // zigbeeModel to my schneider model one and it worked fine but only with one gang, the other gang is not working at all.

anyone know how to add or configure the code to make it work maybe?
Thank you for the upcoming support.

[image]

Koenkk, 2020-02-26:
Could you share your `database.db` file?

onewaylifestreet, 2020-02-26:
Here it is sir, apologize for my late reply.

Thank you

Koenkk, 2020-02-27:
Please try with:

```js
{
    zigbeeModel: ['U202DST600ZB'],
    model: 'U202DST600ZB',
    vendor: 'Schneider Electric',
    description: 'EZinstall3 2 gang 2x300W dimmer module',
    supports: 'on/off, brightness',
    fromZigbee: [fz.generic_state_multi_ep],
    toZigbee: [tz.light_onoff_brightness, tz.ignore_transition],
    meta: {configureKey: 2},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint1 = device.getEndpoint(1);
        await bind(endpoint1, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
        await configureReporting.onOff(endpoint1);
        await configureReporting.brightness(endpoint1);
        const endpoint2 = device.getEndpoint(2);
        await bind(endpoint2, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
        await configureReporting.onOff(endpoint2);
        await configureReporting.brightness(endpoint2);
    },
    endpoint: (device) => {
        return {'l1': 10, 'l2': 11};
    },
},
```

The only thing I expect not to work is the feedback when clicking the brightness buttons manually, but this should also be fixable.

You can now control the device through:
- `zigbee2mqtt/[FRIENDLY_NAME]/l1/set` payload e.g. `{"brightness": 20}`
- `zigbee2mqtt/[FRIENDLY_NAME]/l2/set` payload e.g. `{"brightness": 20}`

onewaylifestreet, 2020-02-28:
It is working now as you said, but the button wont stay on; when pressing on the switch moved back to off but the light is on.

The lights worked on both gang now.
how do it make the button stay on as the light is on?

also when pressing the physical switch, there is no feedback to the button in home assistant.

BIG THANK THANK YOU!

Koenkk, 2020-02-28:
Could you provide the debug log when controlling the switch physically?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

onewaylifestreet, 2020-02-29:
> Could you provide the debug log when controlling the switch physically?
> 
> To enable debug logging set in `configuration.yaml`:
> 
> ```yaml
> advanced:
>   log_level: debug
> ```

I tried but there is nothing coming up when pressing the physical button. any other way to do this correctly?

Thank you!

Koenkk, 2020-02-29:
Can you check with the herdsman debug logging?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

onewaylifestreet, 2020-02-29:
> Can you check with the herdsman debug logging?
> 
> To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

`info  2020-02-29 18:46:16: Logging to console and directory: '/share/zigbee2mqtt/log/2020-02-29.18-46-16' filename: log.txt
debug 2020-02-29 18:46:16: Removing old log directory '/share/zigbee2mqtt/log/2020-02-28.18-16-49'
debug 2020-02-29 18:46:17: Loaded state from file /share/zigbee2mqtt/state.json
info  2020-02-29 18:46:17: Starting zigbee2mqtt version 1.11.0 (commit #unknown)
info  2020-02-29 18:46:17: Starting zigbee-herdsman...
debug 2020-02-29 18:46:17: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/share/zigbee2mqtt/database.db","databaseBackupPath":"/share/zigbee2mqtt/database.db.backup","backupPath":"/share/zigbee2mqtt/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM0"}}'
info  2020-02-29 18:46:19: zigbee-herdsman started
info  2020-02-29 18:46:19: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20180815}}'
debug 2020-02-29 18:46:19: Zigbee network parameters: {"panID":6754,"extendedPanID":"0xdddddddddddddddd","channel":11}
info  2020-02-29 18:46:19: Currently 7 devices are joined:
info  2020-02-29 18:46:19: 0x00124b000e9071f5 (0x00124b000e9071f5): U202DST600ZB - Schneider Electric EZinstall3 2 gang 2x300W dimmer module (Router)
info  2020-02-29 18:46:19: 0x00124b000aa0dbaf (0x00124b000aa0dbaf): U202DST600ZB - Schneider Electric EZinstall3 2 gang 2x300W dimmer module (Router)
info  2020-02-29 18:46:19: 0x00124b0018df58fa (0x00124b0018df58fa): U202DST600ZB - Schneider Electric EZinstall3 2 gang 2x300W dimmer module (Router)
info  2020-02-29 18:46:19: 0x00124b00114a02d0 (0x00124b00114a02d0): U202DST600ZB - Schneider Electric EZinstall3 2 gang 2x300W dimmer module (Router)
info  2020-02-29 18:46:19: 0x00124b0014ae066d (0x00124b0014ae066d): U202DST600ZB - Schneider Electric EZinstall3 2 gang 2x300W dimmer module (Router)
info  2020-02-29 18:46:19: 0x00124b00071ac152 (0x00124b00071ac152): U202DST600ZB - Schneider Electric EZinstall3 2 gang 2x300W dimmer module (Router)
info  2020-02-29 18:46:19: 0x00124b0018df4ea9 (0x00124b0018df4ea9): U202DST600ZB - Schneider Electric EZinstall3 2 gang 2x300W dimmer module (Router)
info  2020-02-29 18:46:19: Zigbee: disabling joining new devices.
info  2020-02-29 18:46:19: Configuring '0x00124b000e9071f5'
error 2020-02-29 18:46:19: Failed to configure '0x00124b000e9071f5', attempt 1 (TypeError: Cannot read property 'bind' of undefined
    at bind (/zigbee2mqtt-1.11.0/node_modules/zigbee-herdsman-converters/devices.js:21:24)
    at Object.configure (/zigbee2mqtt-1.11.0/node_modules/zigbee-herdsman-converters/devices.js:5965:19)
    at DeviceConfigure.configure (/zigbee2mqtt-1.11.0/lib/extension/deviceConfigure.js:95:32)
    at DeviceConfigure.onZigbeeStarted (/zigbee2mqtt-1.11.0/lib/extension/deviceConfigure.js:64:28)
    at Controller.callExtensionMethod (/zigbee2mqtt-1.11.0/lib/controller.js:329:44))
info  2020-02-29 18:46:19: Configuring '0x00124b000aa0dbaf'
error 2020-02-29 18:46:19: Failed to configure '0x00124b000aa0dbaf', attempt 1 (TypeError: Cannot read property 'bind' of undefined
    at bind (/zigbee2mqtt-1.11.0/node_modules/zigbee-herdsman-converters/devices.js:21:24)
    at Object.configure (/zigbee2mqtt-1.11.0/node_modules/zigbee-herdsman-converters/devices.js:5965:19)
    at DeviceConfigure.configure (/zigbee2mqtt-1.11.0/lib/extension/deviceConfigure.js:95:32)
    at DeviceConfigure.onZigbeeStarted (/zigbee2mqtt-1.11.0/lib/extension/deviceConfigure.js:64:28)
    at async Controller.callExtensionMethod (/zigbee2mqtt-1.11.0/lib/controller.js:329:21))
info  2020-02-29 18:46:19: Configuring '0x00124b0018df58fa'
error 2020-02-29 18:46:19: Failed to configure '0x00124b0018df58fa', attempt 1 (TypeError: Cannot read property 'bind' of undefined
    at bind (/zigbee2mqtt-1.11.0/node_modules/zigbee-herdsman-converters/devices.js:21:24)
    at Object.configure (/zigbee2mqtt-1.11.0/node_modules/zigbee-herdsman-converters/devices.js:5965:19)
    at DeviceConfigure.configure (/zigbee2mqtt-1.11.0/lib/extension/deviceConfigure.js:95:32)
    at DeviceConfigure.onZigbeeStarted (/zigbee2mqtt-1.11.0/lib/extension/deviceConfigure.js:64:28)
    at async Controller.callExtensionMethod (/zigbee2mqtt-1.11.0/lib/controller.js:329:21))
info  2020-02-29 18:46:19: Configuring '0x00124b00114a02d0'
error 2020-02-29 18:46:19: Failed to configure '0x00124b00114a02d0', attempt 1 (TypeError: Cannot read property 'bind' of undefined
    at bind (/zigbee2mqtt-1.11.0/node_modules/zigbee-herdsman-converters/devices.js:21:24)
    at Object.configure (/zigbee2mqtt-1.11.0/node_modules/zigbee-herdsman-converters/devices.js:5965:19)
    at DeviceConfigure.configure (/zigbee2mqtt-1.11.0/lib/extension/deviceConfigure.js:95:32)
    at DeviceConfigure.onZigbeeStarted (/zigbee2mqtt-1.11.0/lib/extension/deviceConfigure.js:64:28)
    at async Controller.callExtensionMethod (/zigbee2mqtt-1.11.0/lib/controller.js:329:21))
info  2020-02-29 18:46:19: Configuring '0x00124b0014ae066d'
error 2020-02-29 18:46:19: Failed to configure '0x00124b0014ae066d', attempt 1 (TypeError: Cannot read property 'bind' of undefined
[... cut ...]
info  2020-02-29 18:50:27: MQTT publish: topic 'zigbee2mqtt/0x00124b000aa0dbaf', payload '{"state":"OFF","brightness":255,"state_l2":"ON","state_l1":"ON"}'
debug 2020-02-29 18:50:34: Received MQTT message on 'zigbee2mqtt/0x00124b000e9071f5/l1/set' with data '{"state": "ON"}'
debug 2020-02-29 18:50:34: Publishing 'set' 'state' to '0x00124b000e9071f5'
info  2020-02-29 18:50:34: MQTT publish: topic 'zigbee2mqtt/0x00124b000e9071f5', payload '{"state":"ON","state_l1":"ON"}'
debug 2020-02-29 18:51:17: Saving state to file /share/zigbee2mqtt/state.json
`

Koenkk, 2020-02-29:
Sorry there was a mistake in the definition, please try with below and let me know if any messages appear in the debug logging when pressing the buttons.

```js
{
    zigbeeModel: ['U202DST600ZB'],
    model: 'U202DST600ZB',
    vendor: 'Schneider Electric',
    description: 'EZinstall3 2 gang 2x300W dimmer module',
    supports: 'on/off, brightness',
    fromZigbee: [fz.generic_state_multi_ep],
    toZigbee: [tz.light_onoff_brightness, tz.ignore_transition],
    meta: {configureKey: 2},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint1 = device.getEndpoint(10);
        await bind(endpoint1, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
        await configureReporting.onOff(endpoint1);
        await configureReporting.brightness(endpoint1);
        const endpoint2 = device.getEndpoint(11);
        await bind(endpoint2, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
        await configureReporting.onOff(endpoint2);
        await configureReporting.brightness(endpoint2);
    },
    endpoint: (device) => {
        return {'l1': 10, 'l2': 11};
    },
},
```

onewaylifestreet, 2020-02-29:
`info  2020-02-29 21:30:39: Logging to console and directory: '/share/zigbee2mqtt/log/2020-02-29.21-30-39' filename: log.txt
debug 2020-02-29 21:30:39: Removing old log directory '/share/zigbee2mqtt/log/2020-02-28.19-59-30'
debug 2020-02-29 21:30:40: Loaded state from file /share/zigbee2mqtt/state.json
info  2020-02-29 21:30:40: Starting zigbee2mqtt version 1.11.0 (commit #unknown)
info  2020-02-29 21:30:40: Starting zigbee-herdsman...
debug 2020-02-29 21:30:40: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/share/zigbee2mqtt/database.db","databaseBackupPath":"/share/zigbee2mqtt/database.db.backup","backupPath":"/share/zigbee2mqtt/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM0"}}'
info  2020-02-29 21:30:43: zigbee-herdsman started
info  2020-02-29 21:30:43: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20180815}}'
debug 2020-02-29 21:30:43: Zigbee network parameters: {"panID":6754,"extendedPanID":"0xdddddddddddddddd","channel":11}
info  2020-02-29 21:30:44: Currently 7 devices are joined:
info  2020-02-29 21:30:44: 0x00124b000e9071f5 (0x00124b000e9071f5): Not supported (Router)
info  2020-02-29 21:30:44: 0x00124b000aa0dbaf (0x00124b000aa0dbaf): U202DST600ZB - Schneider Electric EZinstall3 2 gang 2x300W dimmer module (Router)
info  2020-02-29 21:30:44: 0x00124b0018df58fa (0x00124b0018df58fa): Not supported (Router)
info  2020-02-29 21:30:44: 0x00124b00114a02d0 (0x00124b00114a02d0): U202DST600ZB - Schneider Electric EZinstall3 2 gang 2x300W dimmer module (Router)
info  2020-02-29 21:30:44: 0x00124b0014ae066d (0x00124b0014ae066d): Not supported (Router)
info  2020-02-29 21:30:44: 0x00124b00071ac152 (0x00124b00071ac152): Not supported (Router)
info  2020-02-29 21:30:44: 0x00124b0018df4ea9 (0x00124b0018df4ea9): Not supported (Router)
info  2020-02-29 21:30:44: Zigbee: disabling joining new devices.
info  2020-02-29 21:30:44: Configuring '0x00124b000aa0dbaf'
info  2020-02-29 21:30:44: Connecting to MQTT server at mqtt://core-mosquitto
info  2020-02-29 21:30:45: Connected to MQTT server
info  2020-02-29 21:30:45: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
info  2020-02-29 21:30:45: MQTT publish: topic 'zigbee2mqtt/0x00124b000e9071f5', payload '{"state":"ON","state_l1":"ON"}'
info  2020-02-29 21:30:45: MQTT publish: topic 'zigbee2mqtt/0x00124b000aa0dbaf', payload '{"state":"OFF","brightness":255,"state_l2":"OFF","state_l1":"OFF"}'
info  2020-02-29 21:30:45: MQTT publish: topic 'zigbee2mqtt/0x00124b0018df58fa', payload '{"state":"ON","brightness":12,"state_l1":"ON"}'
info  2020-02-29 21:30:45: MQTT publish: topic 'zigbee2mqtt/0x00124b00114a02d0', payload '{"state":"ON","state_l2":"OFF","state_l1":"OFF","brightness":229}'
info  2020-02-29 21:30:45: MQTT publish: topic 'zigbee2mqtt/0x00124b0014ae066d', payload '{"state":"ON","state_l1":"OFF"}'
info  2020-02-29 21:30:45: MQTT publish: topic 'zigbee2mqtt/0x00124b00071ac152', payload '{"state_l1":"OFF","state_l2":"OFF"}'
info  2020-02-29 21:30:45: MQTT publish: topic 'zigbee2mqtt/0x00124b0018df4ea9', payload '{"state_l2":"OFF","state_l1":"OFF"}'
info  2020-02-29 21:30:45: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.11.0","commit":"unknown","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20180815}},"log_level":"debug","permit_join":false}'
error 2020-02-29 21:30:54: Failed to configure '0x00124b000aa0dbaf', attempt 1 (Error: Bind 0x00124b000aa0dbaf/10 genOnOff from '0x00124b00193670d2/1' failed (Error: AREQ - ZDO - bindRsp after 10000ms)
    at Endpoint.<anonymous> (/zigbee2mqtt-1.11.0/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:235:23)
    at Generator.throw (<anonymous>)
    at rejected (/zigbee2mqtt-1.11.0/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65))
info  2020-02-29 21:30:54: Configuring '0x00124b00114a02d0'
error 2020-02-29 21:31:04: Failed to configure '0x00124b00114a02d0', attempt 1 (Error: Bind 0x00124b00114a02d0/10 genOnOff from '0x00124b00193670d2/1' failed (Error: AREQ - ZDO - bindRsp after 10000ms)
    at Endpoint.<anonymous> (/zigbee2mqtt-1.11.0/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:235:23)
    at Generator.throw (<anonymous>)
    at rejected (/zigbee2mqtt-1.11.0/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65))
debug 2020-02-29 21:31:27: Received MQTT message on 'zigbee2mqtt/0x00124b0018df4ea9/l2/set' with data '{"state": "ON"}'
error 2020-02-29 21:31:27: Failed to call 'EntityPublish' 'onMQTTMessage' (AssertionError [ERR_ASSERTION]: Postfix 'l2' is given but device is unsupported
    at Zigbee.resolveEntity (/zigbee2mqtt-1.11.0/lib/zigbee.js:179:25)
    at EntityPublish.onMQTTMessage (/zigbee2mqtt-1.11.0/lib/extension/entityPublish.js:59:36)
    at Controller.callExtensionMethod (/zigbee2mqtt-1.11.0/lib/controller.js:329:44)
    at Controller.onMQTTMessage (/zigbee2mqtt-1.11.0/lib/controller.js:241:14)
    at MQTT.emit (events.js:223:5)
    at MQTT.onMessage (/zigbee2mqtt-1.11.0/lib/mqtt.js:91:14)
    at MqttClient.emit (events.js:223:5)
    at MqttClient._handlePublish (/zigbee2mqtt-1.11.0/node_modules/mqtt/lib/client.js:1162:12)
[... cut ...]
debug 2020-02-29 21:33:55: No converter available for 'U202DST600ZB' with cluster 'genLevelCtrl' and type 'attributeReport' and data '{"currentLevel":189}'
debug 2020-02-29 21:33:56: Received Zigbee message from '0x00124b000aa0dbaf', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":177}' from endpoint 10 with groupID 0
debug 2020-02-29 21:33:56: No converter available for 'U202DST600ZB' with cluster 'genLevelCtrl' and type 'attributeReport' and data '{"currentLevel":177}'
`

onewaylifestreet, 2020-02-29:
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/4270414/log.txt)

Koenkk, 2020-03-01:
Good, now add the following converter to `fromZigbee.js` (in de converters directory next to `converters`:

```js
    generic_brightness_multi_ep: {
        cluster: 'genOnOff',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            const key = `brightness_${getKey(model.endpoint(msg.device), msg.endpoint.ID)}`;
            const payload = {};
            payload[key] = msg.data['currentLevel'];
            return payload;
        },
    },
```

and change the entry to 

```js
{
    zigbeeModel: ['U202DST600ZB'],
    model: 'U202DST600ZB',
    vendor: 'Schneider Electric',
    description: 'EZinstall3 2 gang 2x300W dimmer module',
    supports: 'on/off, brightness',
    fromZigbee: [fz.generic_state_multi_ep, fz.generic_brightness_multi_ep],
    toZigbee: [tz.light_onoff_brightness, tz.ignore_transition],
    meta: {configureKey: 2},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint1 = device.getEndpoint(10);
        await bind(endpoint1, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
        await configureReporting.onOff(endpoint1);
        await configureReporting.brightness(endpoint1);
        const endpoint2 = device.getEndpoint(11);
        await bind(endpoint2, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
        await configureReporting.onOff(endpoint2);
        await configureReporting.brightness(endpoint2);
    },
    endpoint: (device) => {
        return {'l1': 10, 'l2': 11};
    },
},
```

onewaylifestreet, 2020-03-01:
> Good, now add the following converter to `fromZigbee.js` (in de converters directory next to `converters`:
>
> ```js
>     generic_brightness_multi_ep: {
>         cluster: 'genOnOff',
>         type: [...]

Sorry I am asking this, but how to get access to converters directory?
I am using samba addon on home assistant and have a little knowledge of the linux.

Thanks

Koenkk, 2020-03-02:
I've added support for this device in the latest dev branch. Please try the latest **edge** addon in +- 3 hours from now.

onewaylifestreet, 2020-03-04:
> I've added support for this device in the latest dev branch. Please try the latest **edge** addon in +- 3 hours from now.

Hello, I have installed the **zigbee2mqtt-edge** and the physical switch still not supported.
here is the log when using the physical switches on U202DST600ZB.
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/4285741/log.txt)

If you don't mind please also add the following zigbeeModel to the supported device;
'U201DST600ZB' "EZinstall3" - 1 Gang 550W Dimmer Module.
'U201SRY2KWZB' "Ulti" - 240V 9.1 A 1 Gang relay switch impress switch module.
'U202SRY2KWZB' "Ulti" - 240V 9.1 A 2 Gang relay switch impress switch module.
These already worked well with the U202DST600ZB configuration except the physical switches.

Thank you so much.

Koenkk, 2020-03-04:
Can you try repairing the device?

onewaylifestreet, 2020-03-05:
> Can you try repairing the device?

I did and still the same result.
Here is the log.

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/4290380/log.txt)

Koenkk, 2020-03-05:
Do you see anything in the log now when manually controlling the switch?

onewaylifestreet, 2020-03-05:
> Do you see anything in the log now when manually controlling the switch?

Yes I did

Koenkk, 2020-03-05:
Sorry, I'm not sure what changed but with the same configuration it worked before??: https://github.com/Koenkk/zigbee2mqtt/issues/3003#issuecomment-592951569

onewaylifestreet, 2020-03-05:
> Sorry, I'm not sure what changed but with the same configuration it worked before??: [#3003 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/3003#issuecomment-592951569)

I mean it works except the physical switch not showing state on home assistant button.

Koenkk, 2020-03-05:
Yes but before it showed state changes in the log and now not.

onewaylifestreet, 2020-03-05:
> Yes but before it showed state changes in the log and now not.

Where do I get the log of zigbee2mqtt edge?
Is it the folder as zigbee2mqtt?

onewaylifestreet, 2020-03-05:
Now I switched back to zigbee2mqtt not edge, can I IM you somehow?

onewaylifestreet, 2020-03-05:
I will reinstall homeassistant and zigbee2mqtt-edge. start fresh and will provide the log to you as soon as possible.

onewaylifestreet, 2020-03-08:
> Yes but before it showed state changes in the log and now not.

Hello, I have formatted my homeassistant and reinstall.
Here is log.

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/4303460/log.txt)

I can see the state and brightness on l1 transmit thru zigbee on the log but for l2 I can see the state only with no brightness on the log.

Koenkk, 2020-03-09:
That configuration is not correct, please use discovery (https://www.zigbee2mqtt.io/integration/home_assistant.html#mqtt-discovery)

onewaylifestreet, 2020-03-09:
> That configuration is not correct, please use discovery (https://www.zigbee2mqtt.io/integration/home_assistant.html#mqtt-discovery)

The discovery shows only 1 switch and still not working with physical switcch
[image]

Koenkk, 2020-03-09:
Can you provide the log when now controlling it and provide a screenshot after this?

onewaylifestreet, 2020-03-10:
This is the log when controlling it thru home assistant and thru physical buttons.
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/4309865/log.txt)

[image]
This is when toggle the home assistant switch to on and it went back to off.

[image]
This is when pressing both physical switches to on.

Koenkk, 2020-03-10:
Pushed a possible fix, please try again with the latest dev branch and provide the same screenshots and logging again.

onewaylifestreet, 2020-03-11:
> Pushed a possible fix, please try again with the latest dev branch and provide the same screenshots and logging again.

I removed the edge addon but now I cannot find it in https://github.com/danielwelch/hassio-zigbee2mqtt Respiratory anymore.
Any clue?

onewaylifestreet, 2020-03-14:
> Pushed a possible fix, please try again with the latest dev branch and provide the same screenshots and logging again.

Hello again, now that the edge is back on the addon store.

First issue:
[image]
there should 2 entity buttons for each friendly name (ex. l1 and l2) but there is only one show up and it is for l2 only.

Here is the log:
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/4332177/log.txt)

[image]
The state l2 went to ON when pressing button on home assistant but the button not remain on position; no button for l1;

[image]
Here is the screen shot when pressing both physical buttons to on; The state of l1 and l2 is on.

Koenkk, 2020-03-15:
I've managed to reproduce it, unfortunately the MQTT template light cannot be used for this.

I've switched this device to use the MQTT JSON light, this has now been implemented in the latest dev branch.

After updating to the latest dev branch, you have to remove the device first trough https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigremove and pair the device again in order for Home Assistant to load the new configuration.

onewaylifestreet, 2020-03-15:
> I've managed to reproduce it, unfortunately the MQTT template light cannot be used for this.
>
> I've switched this device to use the MQTT JSON light, this has now been implemented in the latest dev [...]

Did as you said; how to I add light entity? 
```
light:
  - platform: mqtt
    schema: json
    name: Bedroom Wardrobe
    state_topic: "zigbee2mqtt/0x00124b000aa0dbaf/l1/set"
    command_topic: "zigbee2mqtt/0x00124b000aa0dbaf/l1/set"
    brightness: true
    brightness_scale: 255
  - platform: mqtt
    schema: json
    name: Bedroom Light
    state_topic: "zigbee2mqtt/0x00124b000aa0dbaf/l2/set"
    command_topic: "zigbee2mqtt/0x00124b000aa0dbaf/l2/set"
    brightness: true
    brightness_scale: 255
```
By adding this to configuration.yaml the lights show up but still same result when pressing physical button(no state change in homeassistant);

Koenkk, 2020-03-15:
You can use MQTT discovery https://www.zigbee2mqtt.io/integration/home_assistant.html#mqtt-discovery

onewaylifestreet, 2020-03-15:
> You can use MQTT discovery https://www.zigbee2mqtt.io/integration/home_assistant.html#mqtt-discovery

Still only 1 switch shows
[image]
[image]

Koenkk, 2020-03-15:
Are you sure you are on the latest zigbee2mqtt edge? (based on the name it doesn't seem to be). The latest build succeeded 2 hours ago so you probably arent.

onewaylifestreet, 2020-03-15:
> Are you sure you are on the latest zigbee2mqtt edge? (based on the name it doesn't seem to be). The latest build succeeded 2 hours ago so you probably arent.

I will try again,
Any way can you add these zigbeeModel for all of switches;
'U202DST600ZB' 2 GANG DIMABLE
'U201DST600ZB' 1 GANG DIMABLE
'U201SRY2KWZB' 1 GANG NO DIM
'U202SRY2KWZB' 2 GANG NO DIM

onewaylifestreet, 2020-03-15:
> I will try again,
> Any way can you add these zigbeeModel for all of switches;
> 'U202DST600ZB' 2 GANG DIMABLE
> 'U201DST600ZB' 1 GANG DIMABLE
> 'U201SRY2KWZB' 1 GANG NO DIM
> 'U202SRY2KWZB' 2 GANG NO DIM

Koenkk, 2020-03-15:
Can you provide the database.db entries for all those devices?

onewaylifestreet, 2020-03-15:
> Can you provide the database.db entries for all those devices?

Here it is sir
[database.zip](https://github.com/Koenkk/zigbee2mqtt/files/4334371/database.zip)

Koenkk, 2020-03-15:
All supported in the latest dev branch now. Update to the latest zigbee2mqtt edge hassio addon after +- 4 hours from now.

Please close if it works as expected :smile:

onewaylifestreet, 2020-03-16:
> All supported in the latest dev branch now. Update to the latest zigbee2mqtt edge hassio addon after +- 4 hours from now.
> 
> Please close if it works as expected 😄

Hello Koenkk,

It is all working perfectly now. I am closing this issue and I would like to thank you for your support from the beginning. Stay safe.

Best regards,
Kip

closed 2020-03-16 by onewaylifestreet

onewaylifestreet, 2020-03-16:
Hello, again

I found an issue on 2 gang dimmable, the brightness for l2 is not reporting to homeassistant. Maybe the name is brightness not l2 brightnesa? Please take a look. Thank you.

Koenkk, 2020-03-16:
Did you already set the brightness for l2?

onewaylifestreet, 2020-03-16:
> Did you already set the brightness for l2?

It is working with homeassistant switch on brightness on both l1 and l2
But the feedback from physical switch works with l1 only not l2

Koenkk, 2020-03-16:
Please provide the debug logging when setting brightness of l2 manually.

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

onewaylifestreet, 2020-03-16:
> ```yaml
>   log_level: debug
> ```

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/4339339/log.txt)
Here you go

Koenkk, 2020-03-16:
The configure fails which is probably the reason for this

```
error 2020-03-16 23:55:15: Failed to configure '0x00124b0018df4ea9', attempt 1 (Error: ConfigureReporting 0x00124b0018df4ea9/11 genOnOff([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":0}], {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":true}) failed (Error: Timeout - 56960 - 11 - 8 - 6 - 7 after 6000ms)
```

Can you try commenting this line and try again ? https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices.js#L8209

onewaylifestreet, 2020-03-16:
> The configure fails which is probably the reason for this
>
> ```
> error 2020-03-16 23:55:15: Failed to configure '0x00124b0018df4ea9', attempt 1 (Error: ConfigureReporting 0x00124b0018df4ea9/11 [...]

the 0x00124b0018df4ea9 is not the switch that have a problem. Would you like me to try comment the line anyway?

Koenkk, 2020-03-16:
Yes, I see the same error for `0x00124b000aa0dbaf`.

onewaylifestreet, 2020-03-16:
> Yes, I see the same error for `0x00124b000aa0dbaf`.

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/4339599/log.txt)
same result

Koenkk, 2020-03-16:
Are you sure you commented https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices.js#L8209 ? Because that is the line causing this error.

onewaylifestreet, 2020-03-17:
> Are you sure you commented https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices.js#L8209 ? Because that is the line causing this error.

The line just moved! Its not the same as last night. And do I need to move back to normal branch?

Koenkk, 2020-03-17:
Sorry, its this line: https://github.com/Koenkk/zigbee-herdsman-converters/blob/9fdd87f64b880c894beb8dde466b33c73dab820f/devices.js#L8212 you have to change it to `// await configureReporting.onOff(endpoint2);`