Koenkk/zigbee2mqtt #3005: Switch from dev-latest to latest on Docker (Synology)

JHurk, 2020-02-26:
I installed 'dev-latest' when did my initial setup a few weeks ago.
The first zigbee devices I purchased were not supported and Koen made a quick change in dev for me to try. Therefor I installed that version from the start.
I have been running Zigbee2MQTT for some weeks now and it works great.

I want to change back to 'latest' again to keep away from possible issues with dev-versions, but can not find a way to do this without creating a second container in Docker. 
Is there some way I can change my current setup from **latest-dev **to** latest** without losing my working setup?

I am on Docker for Synology and work mostly with the GUI settings but am able to SSH into the Synology (but no expert :)).

Hopefully somebody can help me with this so I can change the version.

Koenkk, 2020-02-26:
Remove the old container and start the new one, just make sure you point then to the same data directory.

JHurk, 2020-02-27:
I thought I did this the previous time and it didn't work because my docker folder is pointing to dev version of app/data now. And when I start a new container for latest and point the docker/zigbee2mqtt/data to app/data, it will point to the app/data of the latest container. While all settings are in app/data of the dev container.
Or did I make a mistake last time I did this, which let to the latest container not working?

closed 2020-02-27 by JHurk